MichaIng/DietPi #4192: Since last update First boot installation is not completed successfuly anymore

RonPimplebut, 2021-03-12:
### Creating a bug report/issue

#### Required Information
- DietPi version |
      G_DIETPI_VERSION_CORE=7
      G_DIETPI_VERSION_SUB=0                                                                                                  
      G_DIETPI_VERSION_RC=2
      G_GITBRANCH='master'                                                                                                    
      G_GITOWNER='MichaIng'

- Distro version |
      Buster 
- Kernel version |
      Linux DietPi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- SBC model |
      RPi 4 Model B (aarch64)  
- Power supply used |
      Official Raspberry Pi 4 Power supply (North America)
- SDcard used | NexTech 32 Gb class 10

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Before last update everything was woring like a charm 
2. To reproduse download dietpi and then edit the dietpi.txt with the following setup

```
# IMPORTANT:
# - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
# - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
#------------------------------------------------------------------------------------------------------
# By setting this to "1" you accept the DietPi GPLv2 license and skip the related interactive dialog.
# - Full license text: /boot/dietpi-LICENSE.txt
AUTO_SETUP_ACCEPT_LICENSE=1

##### Language/Regional Options #####
# Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
AUTO_SETUP_LOCALE=C.UTF-8

# Keyboard Layout eg: "gb" / "us" / "de" / "fr"
AUTO_SETUP_KEYBOARD_LAYOUT=us

# Timezone eg: "Europe/London" / "America/New_York" | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
AUTO_SETUP_TIMEZONE=America/New_York

##### Networking Options #####
# Enable Ethernet or WiFi adapter: 1=enable | 0=disable
# - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.
# - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1

# Force ethernet speeds: 0=automatic speed | 10 = 10Mbit | 100 = 100Mbit etc
# - This is mainly aimed at Pine A64 which may have an HW issue that causes unstable 1Gbit link.
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

# WiFi country code: 2 uppercase character value (e.g. GB US DE JP): https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=CA

# Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
[... 238 lines cut ...]

Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
		to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
		(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar 12 02:19:50 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"                                                         
(EE)Fatal server error:
(EE) no screens found
(EE)
(EE)Please consult the The X.Org Foundation support at http://wiki.x.org for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

#### Extra details
When typing command 'startx' i get the same error message.
```

Joulinar, 2021-03-12:
Hi,

this is known issue and will be fixed on next version. At the moment `xserver` is not able to start but there is a workaround available

`/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d-pi4`

fixed with https://github.com/MichaIng/DietPi/issues/4175#issuecomment-794411691

RonPimplebut, 2021-03-12:
it works like a charm again now 👍  thank you very much!

while waiting for the next update. Is there a way to avoid the dietpi update to 7.0.2? I have tried to set "CONFIG_CHECK_DIETPI_UPDATES=0" parameter to 0 but I notice that dietpi 6.33 or 6.34 still update to 7.0.2 on first boot

Thank you

Joulinar, 2021-03-12:
On first Initial boot, DietPi will perform an update to latest version. This is as expected 😃

MichaIng, 2021-03-12:
I hacked in a patch via MOTD, so every freshly flashed DietPi RPi image will have it fixed and all that did not disable MOTD in the DietPi banner on first login after daily cron job (MOTD refresh).

closed 2021-04-24 by MichaIng