MichaIng/DietPi #4194: Cannot install Kodi

mmnpkf, 2021-03-13:
#### Details:
- Date           | Sat Mar 13 15:05:10 AEST 2021
- Bug report     | 34ef5902-0f6e-4253-af28-c5fe52af824e
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq --allow-change-held-packages install libcec4 libnfs12 kodi fbset
- Exit code      | 100
- Software title | DietPi-Software
- 
#### Steps to reproduce:
1. Choose Kodi from list of optimised for dietpi software.
2. Fails to install

#### Expected behaviour:
Should install

#### Actual behaviour:
Failing with error

#### Extra details:
Sorry, I know very little about Linux which is what I like about dietpi - very little knowdege requried to get it up and running and install/remove software.

#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2021-03-13:
Hi,

The problem might be the 64bit environment. But let's see. Can you post full output of `apt update && apt upgrade`

MichaIng, 2021-03-13:
Yes it is, still no Kodi package for RPi 64-bit: https://github.com/MichaIng/DietPi/issues/3570#issue-626586809
Last update is 9 months ago, I hope there are still active maintainers 🤔.

MichaIng, 2021-03-15:
Even that the RPi repo does not ship a Kodi package, the one from the Debian repository should be used as fallback automatically. So would be good to know why it fails by checking `apt update && apt upgrade` output.

Since I aim to make the 64-bit image stable with upcoming release, that one should be sorted, similarly like with the Chromium install, where we need to change the installed package name on RPi 64-bit.

MichaIng, 2021-04-22:
The problem is that the `kodi` package does not contain any binary, but meta files, service files, configs etc, and is hence marked for architecture `all`. As such, having two packages with architecture `all` available, one from RPi repo and one from Debian, makes APT choose the newer one from the RPi repo. But that depends on the `kodi-bin` package with same version or newer, which is only available in the RPi repo as `armhf` package. As APT prefers the `arm64` package, matching the primary system architecture, it wants to use the package from Debian, which does not fulfil the dependency.

`apt install kodi kodi-bin:armhf`, would pull a lot of `armhf` libraries from the repo, and mixing architecture is not what we want and even having the foreign architecture in place is not required anymore: https://github.com/MichaIng/DietPi/commit/a544757c2bc75302128b858748eeade4e9f5519d

I see no other solution than blacklisting all kodi* packages from the RPi distro. Better would be some sort of check whether it has an arm64 kodi-bin available or not, but I have no idea how to do that. A blacklist/bin is still required to prevent `apt upgrade` from attempting to pull the wrong package. Once the arm64 package is available, we need to remove that pin manually with following DietPi update.

MichaIng, 2021-04-22:
Workaround applied with: https://github.com/MichaIng/DietPi/commit/7d1fbcdfd459dc548333c53782cd45cc6a941cd8
Changelog: https://github.com/MichaIng/DietPi/commit/41a6ceea3e523ed73f60969a489292cca6f98a00

closed 2021-04-24 by MichaIng