MichaIng/DietPi #4195: General | Migrate vom p7zip to new 7-Zip for Linux

MichaIng, 2021-03-13:
https://sourceforge.net/p/sevenzip/discussion/45797/thread/cec5e63147/

Awesome, Igor provides official 7-Zip console version for Linux now!! Since p7zip did't show development for five years, this is highly welcome to benefit from all the enhancements that 7-Zip received since v16.02.

The community will need some time to create/merge the build scripts/instructions, but I'll start testing the binary on VM. Debian won't ship packages anymore (with Bullseye) I think, so will ship an own one.

MichaIng, 2021-04-08:
First test as of an issue with p7zip (#4251) and the 7zz x64_64 binary works perfectly fine, being fully compatible regarding command options/arguments. They're available for all architectures we support: https://sourceforge.net/p/sevenzip/discussion/45797/thread/cec5e63147/
Although RPi ARMv6 needs to be tested. I'm thinking to create own DEB packages to use them as replacement for `p7zip`.

Compression ratio comparison is outstanding.

Joulinar, 2021-04-08:
time to get an own apt server as well 🤣

Joulinar, 2021-04-08:
If I understood correctly, something to be done on server side. Means nothing we could support I guess 🙄

MichaIng, 2021-04-08:
> If I understood correctly, something to be done on server side. Means nothing we could support I guess 🙄

Not a big deal to host an APT server on dietpi.com. We already host the mirror for Meveric's Odroid repo at https://dietpi.com/meveric/.
I plan `https://dietpi.com/apt/` to contain our DEB packages. I already converted quite a few ones which were simple archives at first.