MichaIng/DietPi #4196: Error upgrading from v6.34.3 to v7.0.2

lcranf, 2021-03-14:
#### Details:
- Date           | Sat Mar 13 19:58:45 EST 2021
- Bug report     | b8943f4b-3fd7-40a9-8af7-92b112a652fe
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux <<machine_name>> 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ran `sudo dietpi-update`
#### Expected behaviour:
Upgrade `v6.34.3` to `v7.0.2`
- ...
#### Actual behaviour:
Error as outlined above
- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

Joulinar, 2021-03-14:
Hi,

Pls can you post full output of `apt update && apt upgrade`

lcranf, 2021-03-14:
Something going on with `wireguard-tools`?

```
~$ sudo apt update && sudo apt upgrade
Hit:1 https://download.docker.com/linux/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease
Hit:5 https://deb.debian.org/debian buster-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be DOWNGRADED:
  wireguard-tools
0 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Need to get 90.7 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 https://archive.raspberrypi.org/debian buster/main arm64 wireguard-tools arm64 1.0.20200827-1~bpo10+1 [90.7 kB]
Fetched 90.7 kB in 1s (114 kB/s)
(Reading database ... 66238 files and directories currently installed.)
Preparing to unpack .../wireguard-tools_1.0.20200827-1~bpo10+1_arm64.deb ...
Unpacking wireguard-tools (1.0.20200827-1~bpo10+1) over (1.0.20200827-1~bpo10+1) ...
Setting up wireguard-tools (1.0.20200827-1~bpo10+1) ...
wg-quick.target is a disabled or a static unit not running, not starting it.
```

Joulinar, 2021-03-14:
looks like `apt` identified package `wireguard-tools` as to be downgrade. Which is strange because old and new version are same `Unpacking wireguard-tools (1.0.20200827-1~bpo10+1) over (1.0.20200827-1~bpo10+1)`

Does it finished without issues?

lcranf, 2021-03-14:
Good point.  It does...above is the full output.

Joulinar, 2021-03-14:
can you reboot and check Wireguard, as well run again `apt update && apt upgrade`. If all is working and no new packages detected, you can update DietPi

lcranf, 2021-03-14:
#### Ran the following...

1. `sudo reboot --reboot`
2. `sudo wg`
    - saw all client peers previously setup
3.  `sudo apt update && sudo apt upgrade`
    - apt still detects `wireguard-tools` as a downgrade
4. `sudo dietpi-update`
    - get same error as above

Joulinar, 2021-03-14:
DietPi update will not work as long as `apt` is detecting `wireguard-tools` as to be downgrade

lcranf, 2021-03-14:
Okay.   Sounds like I need to remove `wireguard-tools`, upgrade and re-installing `wireguard-tools` is the way forward?  Let me try that

lcranf, 2021-03-14:
I was able to resolve by doing the following....

1. `sudo apt-get remove wireguard-tools`
    - **DO NOT** run `purge` so it keeps current `wg` configs
2. `sudo reboot --reboot`
3. `sudo apt update && sudo apt upgrade`
    - To verify that no packages need upgrading **OR** downgrading
4. `sudo dietpi-update`
5. `sudo apt-get install wireguard-tools`
6. I'm using `pivpn` so ran `pivpn -d` to start wireguard
7. verify everything works by running `sudo wg` or `pivpn -c`

Joulinar, 2021-03-14:
Ok I'm going to close the issue for now. Feel free to reopen if needed.

closed 2021-03-14 by Joulinar

MichaIng, 2021-03-14:
Strange as the version strings are identical. I saw a similar report on Twitter, which was due to the Sid, then Bullseye repository we used to install WireGuard, while now it's available via Buster backports. The version string however has not changed, only the `~bpo10+1` suffix for Buster backports, no actually changed files. But in your case it was already a backports package.

However, can you check whether there are obsolete WireGuard-related APT repos/configs in place?
```
ls -l /etc/apt/{sources.list,preferences}.d
```

lcranf, 2021-03-14:
@MichaIng 

```
~$ ls -l /etc/apt/{sources.list,preferences}.d
/etc/apt/preferences.d:
total 0

/etc/apt/sources.list.d:
total 12
-rw-r--r-- 1 root root 72 Jul 19  2020 docker.list
-rw-r--r-- 1 root root 56 Jun 23  2020 raspi.list
-rw-r--r-- 1 root root 41 Mar 13 08:24 vscode.list
```

MichaIng, 2021-03-14:
Okay that is all fine. Then I have no idea why APT interprets it as downgrade or even sees it as a changed version at all, as it's identical 🤔. However, as the removal an reinstall fixed it, nothing to worry.