MichaIng/DietPi #4197: Unable update and/or Reinstall on 64bit kernel #arch64

cbeinstein, 2021-03-14:
#### Details:
- Date           | Sun Mar 14 13:01:01 IST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install ./emby-server-deb_4.5.4.0_arm64.deb
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2021-03-14:
Hi,

Pls can you post full output of `apt update && apt upgrade`

Joulinar, 2021-05-04:
I'm going to close the issue, feel free to reopen if needed.

closed 2021-05-04 by Joulinar