MichaIng/DietPi #4199: Error on fresh dietpi install running dietpi-software to install nextcloud

maxmoonshark, 2021-03-14:
#### Details:
- Date           | Sun Mar 14 09:18:39 GMT 2021
- Bug report     | cc6bfdb9-0a47-44a2-9018-08e96fd0ca8d
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ncc maintenance:install --no-interaction --database mysql --database-name nextcloud --database-user tmp_root --database-pass <omitted> --admin-user admin --admin-pass <omitted> --data-dir /mnt/dietpi_userdata/nextcloud_data
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Run standard setup procedure
2. dietpi-software to install 114 nextcloud
#### Expected behaviour:
Nextcloud installation
#### Actual behaviour:
Error while executing command.
- ...
#### Extra details:
When running the command manually with the devs only option the following error message is shown:

Error while trying to create admin user: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [1045] Access denied for user 'tmp_root'@'localhost' (using password: YES)
Trace: #0 /var/www/nextcloud/lib/private/DB/ConnectionAdapter.php(69): OC\DB\Exceptions\DbalException::wrap(Object(Doctrine\DBAL\Exception))
#1 /var/www/nextcloud/lib/private/DB/MySqlTools.php(60): OC\DB\ConnectionAdapter->executeQuery('SELECT VERSION(...')
#2 /var/www/nextcloud/lib/private/DB/MySqlTools.php(40): OC\DB\MySqlTools->isMariaDBWithLargePrefix(Object(OC\DB\ConnectionAdapter))
#3 /var/www/nextcloud/lib/private/Setup/MySQL.php(50): OC\DB\MySqlTools->supports4ByteCharset(Object(OC\DB\ConnectionAdapter))
#4 /var/www/nextcloud/lib/private/Setup.php(360): OC\Setup\MySQL->setupDatabase('admin')
#5 /var/www/nextcloud/core/Command/Maintenance/Install.php(108): OC\Setup->install(Array)
#6 /var/www/nextcloud/3rdparty/symfony/console/Command/Command.php(255): OC\Core\Command\Maintenance\Install->execute(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#7 /var/www/nextcloud/3rdparty/symfony/console/Application.php(1009): Symfony\Component\Console\Command\Command->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#8 /var/www/nextcloud/3rdparty/symfony/console/Application.php(273): Symfony\Component\Console\Application->doRunCommand(Object(OC\Core\Command\Maintenance\Install), Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#9 /var/www/nextcloud/3rdparty/symfony/console/Application.php(149): Symfony\Component\Console\Application->doRun(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#10 /var/www/nextcloud/lib/private/Console/Application.php(215): Symfony\Component\Console\Application->run(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#11 /var/www/nextcloud/console.php(100): OC\Console\Application->run()
#12 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#13 {main}
#### Additional logs:
```

  The "-J" option does not exist.

maintenance:install [--database DATABASE] [--database-name DATABASE-NAME] [--database-host DATABASE-HOST] [--database-port DATABASE-PORT] [--database-user DATABASE-USER] [--database-pass [DATABASE-PASS]] [--database-table-space [DATABASE-TABLE-SPACE]] [--admin-user ADMIN-USER]
 [--admin-pass ADMIN-PASS] [--admin-email [ADMIN-EMAIL]] [--data-dir DATA-DIR]

Joulinar, 2021-03-14:
Hi,

I did a test installation of NextCloud and it was working without issues. Let's check why it is failing on your system. Can you try to run installation again. Once hit by the issue, can you scroll up a little bit on your SSH terminal and post last lines from the output.

MichaIng, 2021-03-14:
> The "-J" option does not exist.

Does your global software password start with a dash?

Nextcloud requires a database root user with password authentication to create its own database user to install everything with. MariaDB by default sets up the root user with unix_socket authentication which means that only the UNIX root user can login as database root user without any password. Since we don't want to force password authentication for the database root user, we create a temporary second user with root permissions and password to be used by Nextcloud for the install. And this user gets the global software password and is given Nextcloud to do its install:
```
mysql -e "grant all privileges on *.* to 'tmp_root'@'localhost' identified by '$GLOBAL_PW' with grant option"
G_EXEC ncc ... --database-pass '<omitted>' # Is applied internally to avoid printing it to console
```
And now I can imagine that `ncc` (actually the Nextcloud's `occ` command) interprets a leading dash as a new option, even if it follows a previous option that takes a following value. If that is so, I'll open a bug report at Nextcloud since `--database-pass '-myPasswordStartsWithDash'` should be allowed as the second argument can be safely interpreted as value for the password option.

But we can enhance something on our install as well:
- There is no reason to use the global software password here. It's a temporary user so it should get a temporary password.
- That way we can control the characters, e.g. use a random hex or alphanumerical string to avoid special characters.
- And it's also safer to not use the kinda sensitive global software password when not required. E.g. when the ncc install fails, the user is left in place, so there is a user with full database root privileges that uses a password that is in some cases stored by default in plain text config files elsewhere. I mean it should be changed in every case ASAP, but I'm pretty sure most users do not do that.

maxmoonshark, 2021-03-14:
Yes... it does! "-J" are the first to characters. I didn't think much of it since I have used an alteration as the global password on dietpi before, which I probably set after I installed nextcloud.

Everything you said @MichaIng makes sense...even to me :)

I'll try again with a different password and let you know.

MichaIng, 2021-03-14:
Okay, so we found the culprit. I'll report to Nextcloud devs, as this should be possible to fix.

While I'm implementing a random password, I recognised that while we can use a random password for the random database user, of course for the Nextcloud login user the global software password must still be used (that's the whole purpose of it). So while the change is a good idea, a leading dash would currently still cause an issue.

MichaIng, 2021-03-14:
**Issue reported: https://github.com/nextcloud/server/issues/26109**
While checking the code, I'm not sure whether it can be achieved easily, or whether for some reason it is wanted that the database user password can be not set, so that the next input argument must have the chance to be treated as new option key. Let's see.

maxmoonshark, 2021-03-14:
I or rather my KeePass seems to make a habit of picking random passwords that lead to random errors.

console output:
ERROR 1064 (42000) at line 1: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '3X!N^yDDj' with grant o                                                                                                                            ption' at line 1

and the subsequent dietpi-software error message:

Nextcloud ncc install
                                                                                                 │  - Command: ncc maintenance:install --no-interaction --database mysql --database-name nextcloud --database-user
                                                                                                 │ tmp_root --database-pass <omitted> --admin-user admin --admin-pass <omitted> --data-dir
                                                                                                 │ /mnt/dietpi_userdata/nextcloud_data
                                                                                                 │  - Exit code: 1
                                                                                                 │  - DietPi version: v7.0.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 2 | DISTRO: 5
                                                                                                 │  - Image creator: DietPi Core Team
                                                                                                 │  - Pre-image: Raspbian Lite
                                                                                                 │  - Error log:
                                                                                                 │ Error while trying to create admin user: Failed to connect to the database: An exception occurred in the driver:
                                                                                                 │ SQLSTATE[HY000] [1698] Access denied for user 'tmp_root'@'localhost'
                                                                                                 │ Trace: #0 /var/www/nextcloud/lib/private/DB/ConnectionAdapter.php(69):
                                                                                                 │ OC\DB\Exceptions\DbalException::wrap(Object(Doctrine\DBAL\Exception))
                                                                                                 │ #1 /var/www/nextcloud/lib/private/DB/MySqlTools.php(60): OC\DB\ConnectionAdapter->executeQuery('SELECT VERSION(...')
                                                                                                 │ #2 /var/www/nextcloud/lib/private/DB/MySqlTools.php(40):
                                                                                                 │ OC\DB\MySqlTools->isMariaDBWithLargePrefix(Object(OC\DB\ConnectionAdapter))
                                                                                                 │ #3 /var/www/nextcloud/lib/private/Setup/MySQL.php(50):
                                                                                                 │ OC\DB\MySqlTools->supports4ByteCharset(Object(OC\DB\ConnectionAdapter))
                                                                                                 │ #4 /var/www/nextcloud/lib/private/Setup.php(360): OC\Setup\MySQL->setupDatabase('admin')
                                                                                                 │ #5 /var/www/nextcloud/core/Command/Maintenance/Install.php(108): OC\Setup->install(Array)
                                                                                                 │ #6 /var/www/nextcloud/3rdparty/symfony/console/Command/Command.php(255):
                                                                                                 │ OC\Core\Command\Maintenance\Install->execute(Object(Symfony\Component\Console\Input\ArgvInput),
                                                                                                 │ Object(Symfony\Component\Console\Output\ConsoleOutput))
                                                                                                 │ #7 /var/www/nextcloud/3rdparty/symfony/console/Application.php(1009):
                                                                                                 │ Symfony\Component\Console\Command\Command->run(Object(Symfony\Component\Console\Input\ArgvInput),
                                                                                                 │ Object(Symfony\Component\Console\Output\ConsoleOutput))
                                                                                                 │ #8 /var/www/nextcloud/3rdparty/symfony/console/Application.php(273):
                                                                                                 │ Symfony\Component\Console\Application->doRunCommand(Object(OC\Core\Command\Maintenance\Install),
                                                                                                 │ Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
                                                                                                 │ #9 /var/www/nextcloud/3rdparty/symfony/console/Application.php(149):
[... cut ...]
                                                                                                 │ Object(Symfony\Component\Console\Output\ConsoleOutput))
                                                                                                 │ #11 /var/www/nextcloud/console.php(100): OC\Console\Application->run()
                                                                                                 │ #12 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
                                                                                                 │ #13 {main}

I'll keep you posted...seems like I am on a roll :)

MichaIng, 2021-03-14:
You should start using alphanumerical passwords 😄, at least for the global software password, as Nextcloud is pretty sure not the only one where special characters cause issues.

You can and should always change passwords afterwards, it's anyway not good to have the same in all places. So that default password is only a temporary one to not have fresh installs completely unprotected and a known defined password for it.

The single quote characters definitely breaks the MariaDB/MySQL syntax which uses single quotes around the password to escape other special characters: https://mariadb.com/kb/en/grant/#identified-by-password
We could use the hashed password, but it's not nice to parse it from the output of a MariaDB CLI call. They say it's based on SHA-1 algorithm but `sha1sum` output does not match.

maxmoonshark, 2021-03-14:
Nextcloud installation worked like a charm with alphanumerical password.
Thanks for your help and explanations. I learned a lot today!

MichaIng, 2021-03-14:
Me too 🙂. Good find about the leading dash issue. You can change the password freely afterwards of course.

MichaIng, 2023-01-20:
Sorry for forgetting about this. The solution was actually there already last year: https://github.com/nextcloud/server/issues/26109#issuecomment-1099870443

Solved with: https://github.com/MichaIng/DietPi/commit/1e26d00

closed 2023-01-23 by MichaIng