MichaIng/DietPi #4200: Dropbear Keybased Authentication

manomaya90, 2021-03-15:
This is just a question on what needs to be modified to allow keybased authentication in Dropbear. Beside security, it's just nice to use one password for all ssh.

1. I have ssh-copy-id
2. My key is already inside /etc/dropbear/authorized_keys
3. I saw a config file in /etc/default/dropbear but there isn't anything to change there
4. When I re-ssh, it still only ask for password

Is Dropbear incapable of keybased authentication? But in openwrt I can do that. Should I make a ~/.ssh/authorized_keys for dietpi user?

ravenclaw900, 2021-03-15:
Hello,
Could you give the permissions for the files?
```sh
stat -c %a /etc/dropbear /etc/dropbear/authorized_keys
```

manomaya90, 2021-03-15:
Here it is:
```
root@DietPi:~# stat -c %a /etc/dropbear /etc/dropbear/authorized_keys
755
600
```
```
root@DietPi:~# ls -la /etc/dropbear/
total 32
drwxr-xr-x  3 root root 4096 Mar 15 08:32 .
drwxr-xr-x 90 root root 4096 Mar 15 09:01 ..
-rw-------  1 root root  743 Mar 14 10:31 authorized_keys
-rw-------  1 root root  459 Jan 22 04:58 dropbear_dss_host_key
-rw-------  1 root root  140 Jan 22 04:58 dropbear_ecdsa_host_key
-rw-------  1 root root  805 Jan 22 04:58 dropbear_rsa_host_key
drwxr-xr-x  2 root root 4096 Jan 22 04:49 log
-rwxr-xr-x  1 root root  100 Feb 12  2019 run
```

ravenclaw900, 2021-03-15:
Ah, wait.  The `/etc/dropbear/authorized_keys` path is specifically an OpenWRT thing.  Normally Dropbear uses the `.ssh/authorized_keys` approach.  So yes, you should put your key there.

manomaya90, 2021-03-15:
Ok it works when I manually put it in dietpi and root folder. Perhaps in DietPi the ssh-copy-id should be directed to it's own user folder rather than to dropbear.

Is there anywhere I can specify not to use password at all for all users and disable root login? The settings in /etc/default/dropbear only has option to change port, well I'm not sure what to put in there for other settings.

Off topic: Usually when using sudo I need to input the user password, but I wonder why the dietpi user doesn't need to do that? Even before I add it to group sudo, as I recall it can directly do sudo.

ravenclaw900, 2021-03-15:
Running `dropbear -wsg` should do the trick.  The `dietpi` user doesn't need a password for sudo because of the entry in `/etc/sudoers.d/dietpi`:
```
dietpi ALL=NOPASSWD: ALL
```

manomaya90, 2021-03-15:
Nice! Thanks 🙏🏻

closed 2021-03-15 by manomaya90

MichaIng, 2021-03-15:
For completeness:
- `/etc/default/dropbear` allows to add custom command line options like those to disable password login or root login in general, like `DROPBEAR_EXTRA_ARGS="-ws"`.
- `-g` (disable root password login) has no purpose when `-w` (disable root login) is used: https://manpages.debian.org/dropbear
- Key authentication is enabled automatically, using the regular `~/.ssh/authorized_keys` files one would use for OpenSSH as well.  `/etc/dropbear/authorized_keys` can only make sense for single-user systems, as obviously this breaks the ability to have multiple users with multiple keys 😉.