MichaIng/DietPi #4201: Invalid argument encountered with DietPi Set hardware when installing Kodi

vinhtq115, 2021-03-15:
#### Details:
- Date           | Mon Mar 15 10:40:02 +07 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | alsactl -g store
- Exit code      | 22
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:
It failed at `[FAILED] DietPi-Set_hardware | alsactl -g store`
- ...
#### Additional logs:
```
alsactl: get_control:256: Cannot read control '3,0,0,Playback Channel Map,0': Invalid argument
```

MichaIng, 2021-03-15:
Many thanks for your report.

I recognised this as well when running some tests on RPi. It seems to be the case when, at least since the 5.10 kernel upgrade, the full KMS driver is enabled but no other sound card yet. The full KMS driver adds own HDMI sound card identifiers, even if onboard audio capabilities are not enabled. I still didn't fully understand the purpose of those. Enabling onboard audio requires a reboot on RPi, but `alsactl` finds those KMS HDMI sound cards and tries to store controls for them, which cannot be read, for some reason. AFAIK this did not cause an error until recently, so I suspect a change in kernel/firmware being the culprit.

It can be basically ignored, storing the ALSA controls is more a failsafe step to prevent them being lost when the system crashes before the next regular shutdown. So you can hit exit or replace the command with `:` from the error prompt, all relevant steps have been done already. But probably it's something to forward to RPi developers.

Did you have onboard audio enabled before doing this Kodi install?
Do you have the full KMS driver enabled, e.g. by selecting `vc4-kms-v3d-pi4` in display resolution settings, or adding `dtoverlay=vc4-kms-v3d-pi4` manually to `/boot/config.txt`?

vinhtq115, 2021-03-15:
> Did you have onboard audio enabled before doing this Kodi install?

I don't remember since I only used my Pi as a headless server. Since the dietpi installation, I haven't touched the audio settings until today.

> Do you have the full KMS driver enabled, e.g. by selecting `vc4-kms-v3d-pi4` in display resolution settings, or adding `dtoverlay=vc4-kms-v3d-pi4` manually to `/boot/config.txt`?

My `/boot/config.txt` is currently set to `dtoverlay=vc4-fkms-v3d`.

MichaIng, 2021-03-15:
Okay so audio was disabled first and KMS driver not sure. Since nothing is applied without a reboot, AFAIK it must have been enabled, else it would have been a "no sound card found" error, expected and omitted by the alsactl `-g` option 🤔. I'll run further tests on this to be sure.

Indeed full KMS driver (vc4-kms-v3d-pi4) is not really supported by Kodi on RPi 4, hence on Kodi install we apply the fake KMS driver (vc4-fkms-v3d) automatically, which also should not create those additional HDMI sound cards anymore.

Could you verify after a reboot: `aplay -l`

vinhtq115, 2021-03-15:
Here is the output:

```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
  ```

MichaIng, 2021-03-15:
Jep, that's all fine now, only the 3.5mm jack (headphones) is shown as it should be, unless you attached a HDMI monitor.