MichaIng/DietPi #4204: Kodi restart with kodi.service

elitearmedforce, 2021-03-15:
### Creating a help/question
I was not sure which option it should go under so put it in bug report. I need help, please. I know it issue is from Kodi side.
Kodi crashes in the middle of playing a video or using tv remote(hdmi cec). So I thought of doing something that restarts KODI automatically after it crashes. Stumbled upon this website from [kodi forums](https://forum.kodi.tv/showthread.php?tid=247352), which restarts it. This solution works by adding Kodi to systemd service. If any better alternative exit please mention it.

Screenshot from Kodi:
[image]

I need help, what should I enter here:
```
[Unit]
Description = start Kodi
Requires = ???
PartOf = ???

[Service]
User = xbmc
Group = xbmc
Environment = DISPLAY=:0.0
ExecStartPre = /bin/sleep 1
ExecStart = /usr/bin/kodi
ExecStop = /usr/bin/pkill -9 kodi.bin
Restart = on-failure

[Install]
WantedBy = multi-user.target
```

#### Required Information
- DietPi version | 7.0
- Distro version | buster_10.8
- Kernel version | 5.10.17-v7l+
- SBC model | RPi 4 Model B
- Power supply used | Samsung adapter
- SDcard used | Samsung evo 32gb

#### Additional Information (if applicable)
- Software title | Kodi
- Installed through dietpi-software

#### Steps to reproduce

I know it is Kodi issue not dietpi
1. Sometimes crashes either in the middle of watching a video or clicking next video with tv remote.
2. Often crashes playing HEVC videos, like 2 times in a row just by clicking to play HEVC video

#### Expected behaviour

- Not crashing, when I use the tv remote. (issue mentioned below)

#### Actual behaviour

- Kodi sometimes crashes, either it is due to HDMI-CEC or rarely during HEVC videos. After checking I found these issues.

elitearmedforce, 2021-03-16:
@ravenclaw900 it is a question not requesting for a feature. I wanted to know which desktop management is used for kodi in dietpi. As I wanted to **create a service myself**, I need to know which desktop manager is used.

MichaIng, 2021-03-16:
When using the autostart option, Kodi is started as standalone application, either via `xinit` or as DRM/KMS on RPi, Odroid N2+C4, which do not support X. So desktop manager is invoked to start it.

However, when choosing not root as autostart user, I plan to change this so that LightDM is used, as a step to reliably solve permission issues that non-root users often have to start an X server.

On RPi 4, first assure that the fake KMS driver is enabled, although this should be done on Kodi install automatically:
```
/boot/dietpi/func/dietpi-set_software rpi-opengl vc4-fkms-v3d
```

For the systemd unit:
- Skip `Requires` and `PartOf`, so we can start Kodi in standalone mode instead of overriding an existing X session screen.
- Instead add `After=dietpi-postboot.service` there to have it started at the end of the boot process.
- The `xmbc` user does not exist, so use any other user or create a new one. E.g.:
    ```
    useradd -rMU -G audio,video,render -d /nonexistent -s /usr/sbin/nologin kodi
    ```
    A pure system user with permissions to directly communicate with audio, video and DRM devices.
    Then you can use `User=kodi` and skip `Group`, which will be the `kodi` group automatically. Setting `Group` explicitly does sometimes override supplementary groups, so audio/video/render permissions would be lost.
- Remove `Environment`.
- Use `ExecStart=/usr/bin/kodi --standalone` to start Kodi in standalone mode.
- The `ExecStop` should not be required.

elitearmedforce, 2021-03-17:
Sorry, for later reply. I will try this out asap and let u know. Thank for taking out some time in replying on this. Thanks   to u and team, for DietPi updates.

elitearmedforce, 2021-03-18:
Apologies for the late update, I did exactly what you mentioned. Just some extra commands used and here are the updates.
to start and enable the kodi.service (`systemctl start kodi.service `and` systemctl enable kodi.service`)
Output:
```
root@DietPi:~# systemctl start kodi.service 
root@DietPi:~# systemctl status kodi.service 
● kodi.service - start Kodi
   Loaded: loaded (/lib/systemd/system/kodi.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-03-18 21:31:30 +04; 375ms ago
  Process: 22949 ExecStartPre=/bin/sleep 1 (code=exited, status=0/SUCCESS)
 Main PID: 22951 (kodi)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/kodi.service
           ├─22951 /bin/sh /usr/bin/kodi --standalone
           └─22980 /usr/lib/arm-linux-gnueabihf/kodi/kodi.bin_v8 --lircdev /var/run/lirc/lircd --standalone

Mar 18 21:31:29 DietPi systemd[1]: Starting start Kodi...
Mar 18 21:31:30 DietPi systemd[1]: Started start Kodi.
```
```
root@DietPi:~# systemctl enable kodi.service 
Created symlink /etc/systemd/system/multi-user.target.wants/kodi.service → /lib/systemd/system/kodi.service.
root@DietPi:~# reboot
```

After reboot output: (works fine, no problem yet)
```
root@DietPi:~# systemctl status kodi.service 
● kodi.service - start Kodi
   Loaded: loaded (/lib/systemd/system/kodi.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-03-18 21:42:31 +04; 1s ago
  Process: 11850 ExecStartPre=/bin/sleep 1 (code=exited, status=0/SUCCESS)
 Main PID: 11851 (kodi)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/kodi.service
           ├─11851 /bin/sh /usr/bin/kodi --standalone
           └─11900 sleep 1

Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 237: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 243: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 244: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 245: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 247: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 248: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 249: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: /usr/bin/kodi: 250: /usr/bin/kodi: cannot create /nonexistent/kodi_crashlog-20210318_214232.log: Directory nonexistent
Mar 18 21:42:32 DietPi kodi[11851]: Crash report available at /nonexistent/kodi_crashlog-20210318_214232.log
Mar 18 21:42:32 DietPi kodi[11851]: [E] Invalid arguments ' '

```
Still pending to test does it restart after a crash, as it happens randomly. 
The reason i pasted the output, so that you know what is going and can consider this in a future update and this as a test(just a suggestion).

Thank you,

MichaIng, 2021-03-19:
Ah it seems to log to the working directory and that is the non-existent home directory.

Can you try:
```
mkdir /var/lib/kodi # I think that does already exist
usermod -d /var/lib/kodi kodi
```
At least you get the crash log with this.

elitearmedforce, 2021-03-21:
haha, thanks, honestly it never bothered me. I have added these commands. 
Updates on service file, day before yesterday it crashed several times and it didn't restart automatically. So I modified by adding first `RestartSec = 30s` then after another crash I modified this to `Restart = always`. Still w8ing for this work once it crashes

elitearmedforce, 2021-03-21:
Update, so I noticed it doesn't restart after the crash even after the modification mentioned above. So, I tried running manually with this command `systemctl restart kodi.service` still nothing until i run command 'kodi'. To confirm this I used [bpytop](https://github.com/aristocratos/bpytop) to confirm kodi process running, but found none. So I believe that as per my understanding, "ExecStart" never runs or executed. This could be that command u provided is wrong for "ExecStart"(as i found that `/usr/bin/kodi-standalone` exist not `/usr/bin/kodi --standalone', previously i thought it was a command to run kodi(from loc /usr/bin/kodi) in a standalone as u mentioned above)
This is the present file of kodi.service
```
[Unit]
Description = start Kodi
After = dietpi-postboot.service

[Service]
User = kodi
Group = kodi
ExecStartPre = /bin/sleep 1
ExecStart = /usr/bin/kodi --standalone
Restart = always
RestartSec = 59s

[Install]
WantedBy = multi-user.target
```

MichaIng, 2021-03-23:
`/usr/bin/kodi-standalone` is only a wrapper for `/usr/bin/kodi --standalone` (calling `/usr/bin/kodi` with the `--standalone` argument) that starts PulseAudio (we use ALSA) and wraps Kodi into an own crash loop, not required when the systemd unit does this already: https://salsa.debian.org/multimedia-team/kodi-media-center/kodi/-/blob/master/tools/Linux/kodi-standalone.sh.in#L26

Could you run and paste the output of `journalctl -u kodi` again when the service fails? I see the "Invalid arguments" in the above output, but actually this cannot be. I just checked the `kodi-standalone` of the package from the RPi archive and it does exactly that (like the one from the Debian repo linked above), nothing else.
It should not be an issue, but please also try to remove the spaces before and after `=` in the systemd unit. I see it here and there, but it's actually not part of the syntax that systemd intends and no Debian package ships with these spaces. Probably it has a non-obvious subtil effect.

elitearmedforce, 2021-03-30:
@MichaIng Apologies for the late reply, was w8ing for Kodi to crash in my presence.
Updates, as u said I reverted the changes i made (`/usr/bin/kodi --standalone` thingy) back to what u said. In addition, removed unnecessary space.

Present kodi.service file:
```
[Unit]
Description=start Kodi
After=dietpi-postboot.service

[Service]
User=kodi
ExecStartPre=/bin/sleep 1
ExecStart=/usr/bin/kodi --standalone
Restart=on-failure
RestartSec=30s

[Install]
WantedBy=multi-user.target
```
After few days
Then typed this command `journalctl -u kod`
[Output in pastebin](https://pastebin.com/MyK6GZjw)
After crash found this using command `journalctl -u kod`
[Output in pastebin](https://pastebin.com/q6duCRsH)
I noticed issue with permission denied. so i changed the user of it from root to kodi `chown kodi /var/lib/kodi/`(group unchanged, default root). Then rebooted it. 
Output:
```
root@DietPi:~# journalctl -u kodi
-- Logs begin at Tue 2021-03-30 13:18:47 +04, end at Tue 2021-03-30 13:22:28 +04. --
Mar 30 13:19:46 DietPi systemd[1]: Starting start Kodi...
Mar 30 13:19:47 DietPi systemd[1]: Started start Kodi.
```

elitearmedforce, 2021-03-30:
After all of that, I tried experimenting to check that **kodi.service** works or not:
This is how my kodi looks(this starts without the kodi.service, default one)
[image]
So I tried exiting kodi:
[image]
This the output from tv screen after exiting kodi:
[image]
then used command systemctl restart kodi.service and it worked surprisingly(coz last time it didn't work)
[image]
But, now if u notice it starts different version of kodi(like clean slate, this kodi looks more of fresh install).I started suspecting that another instance of kodi is executed. To confirm this, restarted my device and screen displays new fresh kodi, not the previous one. To further confirm, checked the processes running(screenshot attached below).
[image]
From this image it is clear that it runs two instance of kodi with different user. I believe this is because the system kodi is executed first with username kodi and then root kodi is executed later. 

Here is command output from `journalctl -u kodi` after all of that:
```
root@DietPi:~# journalctl -u kodi
-- Logs begin at Tue 2021-03-30 13:18:47 +04, end at Tue 2021-03-30 13:37:30 +04. --
Mar 30 13:19:46 DietPi systemd[1]: Starting start Kodi...
Mar 30 13:19:47 DietPi systemd[1]: Started start Kodi.
Mar 30 13:29:47 DietPi systemd[1]: Stopping start Kodi...
Mar 30 13:29:47 DietPi systemd[1]: kodi.service: Main process exited, code=killed, status=15/TERM
Mar 30 13:29:49 DietPi systemd[1]: kodi.service: Succeeded.
Mar 30 13:29:49 DietPi systemd[1]: Stopped start Kodi.
Mar 30 13:29:49 DietPi systemd[1]: Starting start Kodi...
Mar 30 13:29:50 DietPi systemd[1]: Started start Kodi.
Mar 30 13:30:47 DietPi systemd[1]: Stopping start Kodi...
Mar 30 13:30:47 DietPi systemd[1]: kodi.service: Main process exited, code=killed, status=15/TERM
Mar 30 13:31:06 DietPi systemd[1]: kodi.service: Succeeded.
Mar 30 13:31:06 DietPi systemd[1]: Stopped start Kodi.
Mar 30 13:31:06 DietPi systemd[1]: Starting start Kodi...
Mar 30 13:31:07 DietPi systemd[1]: Started start Kodi.
Mar 30 13:31:26 DietPi kodi[3832]: [E] Invalid arguments ' '
Mar 30 13:31:27 DietPi systemd[1]: kodi.service: Succeeded.
Mar 30 13:36:48 DietPi systemd[1]: Starting start Kodi...
Mar 30 13:36:49 DietPi systemd[1]: Started start Kodi.
```
Now i disabled the service using command `systemctl disable kodi.service`. Then reboot, and it is back to previous kodi. Only one process of kodi is running.

elitearmedforce, 2021-03-31:
> You run Kodi with two different users ("root" from console and "kodi" from the service), which store their own settings etc in their home directories, so it is expected that you have the impression it's two Kodi instances. It's only one, but two settings/data instances. Does this make sense?

Yes, I absolutely understand that. Now I understand the full picture. Is there a way to disable running Kodi with user "root", so that I can run it with user kodi by enabling kodi.service, presently it is disabled coz I don't want two process(CPU consumption). Can I know which file runs kodi(as root) to disable that, so that it can be executed automatically at startup with systemd.

MichaIng, 2021-03-31:
> Is there a way to disable running Kodi with user "root"

Probably `dietpi-autostart` is still set to Kodi? In case switch back to manual console login. Otherwise nothing is starting Kodi as root 🤔.

elitearmedforce, 2021-04-06:
@MichaIng First of all thanks for helping me out, I learned new things in the process. Finally, Kodi crashed and it restarted automatically after 30sec. 
Updates:
First disabled auto boot to kodi from `dietpi-autostart` to manual login.
Then enabled the kodi service. Reboot.
Kodi.service file:
```
[Unit]
Description=start Kodi
After=dietpi-postboot.service

[Service]
User=kodi
ExecStartPre=/bin/sleep 1
ExecStart=/usr/bin/kodi --standalone
Restart=on-failure
RestartSec=30s

[Install]
WantedBy=multi-user.target
```
But I made changes to kodi.service coz I didn't t want to go through kodi setup with emby addon process again until I format the device, so I changed the User to `User=root`. Works absolutely fine, even restarts after a crash.
U can consider this thing in a new dietpi update maybe.
Thanks,

closed 2021-04-06 by elitearmedforce

MichaIng, 2021-04-06:
Great that it works now as intended.

> coz I didn't t want to go through kodi setup with emby addon process again

You mean you did that setup with root user already and don't want to repeat that for the "kodi" user, which does not see the previous root user setup? Probably the related configs/data can be copied from within `/root/` to `/var/lib/kodi/`, but not sure currently how/where exactly those are stored.

elitearmedforce, 2021-04-06:
>You mean you did that setup with root user already and don't want to repeat that for the "kodi" user, which does not see the previous root user setup?

Yes. Presently settings are manageable. I don't think so that is a huge issue, so I didn't bother much in copying it.

>but not sure currently how/where exactly those are stored.

In future, when I get a fresh install of Dietpi, then I can google it. Not a huge issue presently, so no worries, will figure it out.
If you want me to test it, I can do that, just saying.

Honestly, thank you again,