MichaIng/DietPi #4206: pytz module ImportError

githlp, 2021-03-16:
G_DIETPI_VERSION_CORE=7, SUB=0, RC=2, master, MichaIng
buster
Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
RPi 4 Model B (armv7l)
5V 3A
SanDisk ultra 32GB

#### Steps to reproduce
in case you activate this

local_time_file /etc/localtime

in /etc/motioneye/motioneye.conf

then the motioneye.service on DietPi is broken:

sudo journalctl -u motioneye
...
DietPi meyectl[507]: ImportError: no module named pytz

it cannot load, although i installed pytz as user with pip/pip2. In python shell the pytz module can be loaded though..

MichaIng, 2021-03-16:
Many thanks for your report.

Please try to install the module globally by running pip2 as root: `sudo pip2 install -U pytz`

githlp, 2021-03-17:
seems to work, thx!

Now i have an issue with dietpi user permantly logged in on dietpi-ip:8765 and no cam shown though i see the actual stream on dietpi-ip:8081..

i restartet motion.service, motioneye.service, the raspi .. nothing helps. When accessing :8765 the user DietPi is logged in (even on other browsers) and i have no clue how to logout and login as admin or as user again..

MichaIng, 2021-03-17:
Did it work before you added/uncommented the local_time_file setting? Unlikely, but at least worth to give it a try when this is all you changed. I tested here and it works, also the switch user button. But I see the same 403 warning in the logs.

I can only change the admin username and add a surveillance user. Is "dietpi" your surveillance user?

I also see an error about missing `lsb_release` command which can be fixed by `apt install lsb-release`, but I cannot see what changed, i.e. what it does. Will further investigate tomorrow: https://github.com/ccrisan/motioneye

githlp, 2021-03-18:
The user is not "dietpi" , i figured out the DietPi next to the "switch user" button represents the computer (OS) name.

So it seems i am not logged in at all, therefor no cam is found/available.

It worked before for "admin" as well as for "user", and as far as i can remember i logged in mostly as "admin" to adjust settings.

To me it seems there is some problem with changed camera resolution but why should that have an effect on the login popup mask?

githlp, 2021-03-18:
now it worked from another Computer, though. But i checked firewall and fail2ban and the first computer is not blocked or banned in any way.. Maybe DietPi or motioneye itself is blocking the IP somehow so the login popup mask won't show up?

Joulinar, 2021-03-18:
Did you tried a different browsers? Did you cleared cookies and cache on the browser?

githlp, 2021-03-18:
Yes, of course. That is why i think the IP gets some restriction. I will examine more logs, though most of them are empty or on some different place within DietPi..

MichaIng, 2021-03-19:
DietPi does not block any IP by itself and does not configure motioneye to do so. By default there are indeed two users, "admin" to setup everything and indeed "user" to only view the camera. Yes the settings slide shows the hostname, hence "DietPi" by default.

Are the DietPi system and both PCs in the same network, i.e. do their IP only differ in the last triple?

githlp, 2021-03-20:
No, they DietPi is in different subnet, but as i said, it used to work and after some while the login page just "hangs" somehow.

I can access the login page from another laptop that is also in another subnet of my home network - so that is not the problem..

When the motioneye service is not reachable/available i do not see anything in the browser except an error message. Thats why i was astonished to see the motioneye web interface just omitting the login dialog to me..

There must be some function within the login procedure that for any reason does not work for me anymore on my main computer.

MichaIng, 2021-03-23:
Please try to update MotionEye: `sudo pip2 install -U motioneye`

MichaIng, 2021-04-16:
I mark this as closed. If the issue persists, feel free to reopen.

closed 2021-04-16 by MichaIng