MichaIng/DietPi #4207: DietPi update failed

omavoss, 2021-03-16:
dietpi update failed - please help

#### Details:
- Date           | Tue Mar 16 19:51:36 CET 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://mirror.netcologne.de/raspbian/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Err:2 https://archive.raspberrypi.org/debian buster InRelease
  Splitting up /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_InRelease into data and signature failed
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://archive.raspberrypi.org/debian buster InRelease: Splitting up /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_InRelease into data and signature failed
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Splitting up /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_InRelease into data and signature failed
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: Splitting of clearsigned file /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_buster_InRelease failed as it doesn't contain all expected parts
E: The package lists or status file could not be parsed or opened.

Thank you.

MichaIng, 2021-03-16:
Many thanks for your report.

Please try to reset APT cache and do the list update manually:
```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

Joulinar, 2021-03-16:
for completeness, issue was opened on forum as well https://dietpi.com/phpbb/viewtopic.php?f=11&t=8811

omavoss, 2021-03-16:
Many thanks for your quickly support, it was very helpful.

root@DietPi:~# apt update
Get:1 https://mirror.netcologne.de/raspbian/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:3 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf Packages [13.0 MB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf Packages [367 kB]
Get:5 https://mirror.netcologne.de/raspbian/raspbian buster/contrib armhf Packages [58.7 kB]
Get:6 https://mirror.netcologne.de/raspbian/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.6 MB in 1min 12s (188 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@DietPi:~#

Many friendly greetings.

Joulinar, 2021-03-16:
is `dietpi-update` working now?

omavoss, 2021-03-16:
Wait a moment please . . . I try . . .

closed 2021-03-16 by omavoss