MichaIng/DietPi #4210: during fresh install...

deckspace, 2021-03-17:
Hi all, just going through clean install process, adding first set of software having just changed host name, enabled wireless and enabled avahi.

#### Details:
- Date           | Wed Mar 17 20:09:47 GMT 2021
- Bug report     | 511ff928-cbed-4249-b9f9-b941511965d8
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux BabarLuck 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | alsactl -g store
- Exit code      | 22
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
alsactl: get_control:256: Cannot read control '3,0,0,Playback Channel Map,0': Invalid argument

MichaIng, 2021-03-19:
Many thanks for your report.

You can basically ignore it and hit exit on the error prompt. All relevant steps have steady been done. Is a change with the latest kernel that breaks saving ALSA configs on RPi when onboard sound has not yet been disabled. Will be worked around with next release or fixed with next kernel package, if I can narrow down the issue more next week.

MichaIng, 2021-04-16:
I mark this as closed, feel free to reopen if required.

closed 2021-04-16 by MichaIng