MichaIng/DietPi #4211: WAN IP in dietpi-banner has a tendency to not resolve

cheqpat, 2021-03-19:
#### Required Information
- DietPi version | v7.0.2
- Distro version | buster
- Kernel version | 5.10.17-v8+
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | random usb c phone charger (5V/3A)
- SDcard used | SanDisk Plus 16gb (iirc)

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Enable WAN IP in bietpi-banner
2. Either reload it several times or just wait for it to fail naturally.
(Reloading rapidly probably hits some ratelimiter, to be fair)

#### Expected behaviour

- WAN IP should display

#### Actual behaviour

- WAN IP has a tendency not to display, or hangs for a notable amount of time (~1/2s)

#### Extra details

- Could be resolved with storing the output of the last successful WAN IP attempt, then using that value if no WAN IP is returned from ifconfig.co. A date displaying the time this IP was stored could be appended to prevent confusion as to why it may be outdated.

MichaIng, 2021-03-19:
Many thanks for your suggestion.

There is already a feature request to optionally allow obtaining the WAN IP and other info that require a network request via cron job. Probably this can be combined by using the same file to store the info on any case and use it as fallback if the request fails.

A timestamp behind the IP + location is probably too long for small screens. Probably on a second line with a failure indicator?

cheqpat, 2021-03-19:
Yeah that'd be fine, whatever works!
I removed the WAN IP geographical location from the banner a while back so I'd forgotten how many columns it usually takes up.

MichaIng, 2021-03-20:
Yes exactly.

MichaIng, 2021-10-04:
We switched to a more reliable WAN IP provider and with next release use one on our own domain. Probably this solves the issue of long response time. Ours is served via Cloudflare worker, so should be pretty fast served all over the world. Probably you can test it with current DietPi version and then again with the changes done here: https://github.com/MichaIng/DietPi/pull/4798/files#diff-c793d9b8219c44005903fe22fd5cfb1096b7789e1351ee0f9ed7c5c148c52c1b

closed 2021-10-09 by MichaIng