MichaIng/DietPi #4212: Many (minor) issues on a fresh install

pitsi, 2021-03-20:
### Creating a bug report/issue

#### Required Information
- DietPi version: 7.0.2
- Distro version: 10.8
- Kernel version: 5.10.17-v8+ (aarch64)
- SBC model: RPi 4 Model B
- Power supply used: Generic 5v 3a power supply (xbs-0530)
- SDcard used: Integral 16gb u1

#### Additional Information (if applicable)
- Software title: Xfce mostly, but some are unrelated to it and have to do with the os itself
- Was the software title installed freshly or updated/migrated? Installed fresh from the 64bit image.
- Can this issue be replicated on a fresh installation of DietPi? Yes.
- Bug report ID: 04772d13-06cd-4047-aedd-bc8cdd40ffc7 (although I did not send a bug report, this is what the echo command outputs)

#### Steps to reproduce
1. Apt is VERY slow on some occasions. Apt-cache search whatever and apt-get Install whatever. E.g. I just searched for chromium and apt-cache search took 15+ seconds to show any results! Apt-get install chromium took a bit less than 10 to show the "Reading package lists" part, let alone the rest of the output which was like 10-15 lines. All that was done via ssh through a wired connection
2. There is no shutdown or restart on xfce's logout menu. In fact, all it does is logout and then I have to do the rest via lightdm's menu. Dbus(-x11) and packagekit are both installed, in case it helps
3. Before installing lightdm, I tried the autologin option in dietpi-config > autostart options > automatic login. It failed to start xorg, showing "no screens found(EE)" in /home/dietpi/.local/share/xorg/Xorg.0.log. Same thing happened when I logged in as the dietpi user and ran startx. 
4. There is no recycle bin on xfce's desktop and in the file manager. Installing gvfs solves that issue, so I think gvfs should be a part of the xfce "set" of packages (dietpi-software > software optimised > 25 xfce).
5. There is a desktop shortcut of htop that just won't go away. I delete it on every login and on the next login it is back!
6. Htop and free -m report 3.7-3.8GB of usable memory. I thought an 64bit os would report the full 4GB of the sbc. Are the remaining 256MB reserved for the gpu? If so, I remember dietpi-config asking me to raise the gpu memory to 64 or so when I installed xfce, so where are the rest?

#### Expected behaviour
1. Apt, by default, is reasonably fast with the forementioned operations. In fact, it is faster on my old 32bit pc that runs on an hdd than on the rpi4.
2. The logout option on xfce pops up a window that has options for logout, shutdown, reboot, hibernate etc.
3. Autologin should start xorg, and so should startx when run manually.
4. Read above.
5. The shortcut should not be recreated.
6. Read above.

#### Actual behaviour
1. Read above.
2. Read above.
3. Startx works when it is run as a root.
4. Read above.
5. Read above.
6. Read above.

#### Extra details
This is my first attempt at a dietpi install from scratch and on an rpi4 (thus the demand for aarch64). The rpi4 itself will be plugged into a tv and it will show a couple of web pages throughout the day, thus the need for xorg and a simple desktop enviroment. Firefox-esr was chosen to be the browser for the work because the downloads page mentions there is no kodi and chromium for the 64bit image.
Except for xfce and firefox esr, avahi-daemon, ceni and the numix theme were installd too.
Before dietpi I tried the official images of raspbian os, ubuntu server 20.04 (arm64) and ubuntu mate 20.04 (arm64). I want to stick to a debian based distro because I run debian (testing and unstable) for the last 13 years, so I know its ins and outs almost perfectly. None of the forementioned distros pleased me, mainly because of their bloat, so I turned to dietpi.
On top of the above, I set up my desktop installation starting from command line and installing everything I need afterwards, so I do know what depends on what, e.g. thunar and gvfs for the recycle bin.

p.s. I am facing a neurological issue these last 2 weeks and it makes my left arm go numb from the elbow to the fingertips. So, please forgive any delays in answering. Plus, I have probably missed 2-3 more issues I have. Thank you for your understanding :)

MichaIng, 2021-03-20:
Many thanks for your report.

APT cache is indeed slow since we disable the cache by default and keep the lists compressed to reduce disk I/O on APT updates. The cache can be re-enabled via dietpi-config > Advanced Options, a switch for the lists compression will be added with next release.

The upower package sound add restart and shutdown buttons to the main menu. It's it installed as expected?

Autologin with non-root users is indeed currently not reliable as every SBC/GPU works different so that it's not trivial to grant sufficient permissions reliably.  With next release we'll switch to a LightDM-based autologin method for all X applications, which works reliable while keeping security up.

Not sure about the recycle bin. AFAIK we actively disable it for sure desktops, but not sure why. Makes sense to have it enabled by default which is what desktop users are used to from other OS'es as well.

I'll have a look into the htop issue. We have an autostart task (for the desktop) with initially creates the desktop icons, but that should disable itself. I'll have a look of that for some reason fails on Xfce as non-root.

pitsi, 2021-03-21:
Thank you for your support.

First things first, the issue with xfce's shutdown and restart buttons was my fault. Instead of disabling xfce's ability to restore session, I disabled the one that offered the forementioned options. As a friend of mine says "what kind of a pebcak user did that!?"... This may also be the problem with htop's shortcut on the desktop, but I will check it once I reconnect it to a tv.

About the autologin thing. Provided it is not a permissions issue for the user, i.e. the user can run startx and have xorg show up with no errors, how about using nodm? I used it ~7 years ago that I set up a pc that was just a picture viewer (as in digital frame) in the front and an ftp and samba server at the back.

As for the memory used by the gpu, dietpi-config > display options > gpu/ram memory split says it is set to 64MB. 
```
│ Hardware : RPi 4 Model B (aarch64)                                           │
│ Current: 64 MiB GPU | 4032 MiB RAM                                           │
```
And all the options under gpu memory splits in /boot/config.txt report 64
```
#-------GPU memory splits-------
gpu_mem_256=64
gpu_mem_512=64
gpu_mem_1024=64
```
I left it as it is.

One more issue and a question.
On boot, there is this message that repeats ~10 times
`DietPi-Boot[414]: [ INFO ] DietPi-Boot | Waiting for valid network connection before continuing boot | Mode=1`
Obviously it happens because it is not connected to any network when it is put on the tv. I can't see it when i connect it via ethernet, but I assume it does not happen. Shall I be worried about it? If there is a way to hide all boot messages, e.g. by setting the quiet kernel parameter, I will do it. But how do I do that? Is it the same as in desktop debian that I set the grub parameters in /etc/default/grub and run update-grub?

Speaking of networking, is there a way to set the wireless stuff (ssid, key etc) via dietpi-config or something? If not, I will resort to ceni. But if there is a gui way without using something extra (network manager, wicd or connman) I would prefer it.

Joulinar, 2021-03-21:
You can configure WiFi using `dietpi-config`. There is a network section. First you would enable onboard WiFi and Dona reboot. Once done you can configure your WiFi network. By default DiePi will wait 10 seconds on a valid network connection. If not needed this can be configured to be skipped.

pitsi, 2021-03-21:
Thank you. I used that submenu to only enable wifi and I never went back after that. I am keeping ceni just in case.
I also removed the 10 second delay and now things are smooth. I will make a new user later on to check that startx issue.

pitsi, 2021-03-22:
So, I made a new user, disabled lightdm, logged in through tty and tried to start xorg with startx. And it failed to start, popping the same error message I mention above. Moving on...

Is there a way to show the pi's temperature on the (xfce4-) panel? In raspbian and its lxde based desktop, all I had to do is to add the temperature plugin on lxde panel. If it is not possible, I will install conky and configure it accordingly. I already know about `vcgencmd measure_temp`.

Is there some service or something that takes care of apt's... maintenance? Every time I run apt-get update I see apt download nothing at all, plus apt's cache in /var/cache/apt/archives/ is always empty after a fresh start.

MichaIng, 2021-03-23:
> "what kind of a pebcak user did that!?"

What exactly did you disabled? The htop shortcut (DietPi-Desktop_setup) and the power buttons should not be related. I don't like the DietPi-Desktop_setup method either, if that was meant, but we couldn't find any other way to create shortcuts for new desktop users automatically, sadly. If someone has a better idea, I'd love to hear it 😃.

If `startx` form console works, the autostart option 2 must as well, since it does nothing else 🤔: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L46

> Current: 64 MiB GPU | 4032 MiB RAM

Okay, all fine. Btw the `4032 MiB` here are not precise as this is calculated via 4096 MiB (4 GiB) minus 64 MiB while the actual RAM size is always a little lower. I need to fix that...
I misunderstood the related sentence in the OT: "3.7-3.8GB" sound pretty correct for what the 4 GiB RPi 4 really comes with minus 64 MiB. E.g. here my 1 GiB model shows `Mem:             973` ("total" column) which means with 16 MiB GPU memory actual 989 MiB RAM while a full 1 GiB would be 1024 MiB. I never understood the real reason why manufacturer never ship those with the full GiB sizes, similarly why an e.g. 4 TiB drive does not ship with 4 TiB, not even with 4 TB (terabyte vs tebibyte) but always less.

About APT updates. By default those are checked daily and available updates shown in the login banner. In /boot/dietpi.txt there is a `CONFIG_CHECK_APT_UPDATES` setting which can be set to `2` to have daily APT updates not only checked but applied. Update checks do not download archives, but update the list files only. And when applying upgrades, `/var/cache/apt/archives/` should be used only temporarily until all installs have finished, then cleared. That is the reasonable APT default.

> Cross out the question about temperature display. Xfce4-sensors-plugin picks up the relevant sensor automatically and displays the temp on the panel just fine.

Yep, luckily the RPi kernel provides the temperature as well via `/sys/class/thermal/thermal_zone0/temp` and `/sys/class/hwmon/hwmon0/temp1_input` so that tools can read it from generic locations instead of needing to use an SBC+firmware specific tool like `vcgencmd`, which nobody would be keen to implement and maintain 😉.

pitsi, 2021-03-23:
I did not mean to offend anyone with my friend's words. They refer to me and only me, and this is what he usually says when I do mistakes that are too big for my experience in linux.
The mistake I did was to untick the option for "prompt on logout" instead of the one for "automatically save session on logout".
https://docs.xfce.org/xfce/xfce4-session/preferences

Startx from the console does not work, unless the user is root. It may be some permissions issue, e.g. dietpi not being part of a specific group (video). On my desktop, where my user is in the video group, i can use startx as usual with no issues, although I also use lightdm. But that is a lot different than the rpi.

About apt. Do I have to run apt-get (dist-)upgrade by hand then?

MichaIng, 2021-03-23:
> I did not mean to offend anyone with my friend's words. They refer to me and only me, and this is what he usually says when I do mistakes that are too big for my experience in linux.

All fine, no worries 😄. If that script was meant, I agree that it is not great. If anyone knows a better way to have the contained desktop shortcuts and settings (PCManFM bookmarks, ...) created automatically on first desktop login, that would be awesome. Until then, it just works.

> The mistake I did was to untick the option for "prompt on logout" instead of the one for "automatically save session on logout".

Ah `automatically save session on logout` likely is required to have Xfce settings changes saved. Removed desktop icons should actually not be affected since those are deleted immediately, not as part of a session info.

> dietpi not being part of a specific group (video).

Makes sense: `usermod -aG video,render dietpi`
I wonder why I didn't add this do be done automatically when non-root is chosen for desktop autologin, like we do for Kodi 🤔. However, the switch to LightDM autologin is required anyway, since even the above group memberships are not sufficient on some SBCs.

> About apt. Do I have to run apt-get (dist-)upgrade by hand then?

By default (`CONFIG_CHECK_APT_UPDATES=1`) yes, the banner then only shows you the info that updates are available, but does not run them. When switching to `CONFIG_CHECK_APT_UPDATES=2` it's all done automatically via daily cron job.

pitsi, 2021-03-25:
I forgot to switch that parameter to 2 last time and today apt prompted me about 2 packages that need updating (firefox-esr and its greek translation), so I missed the chance to see how it works. 

One other thing that I noticed with almost all the distros I have tried is that the lan leds stay on even after shutdown. The same applies to a small fan that I have put in the 4th and 6th pins (5v and ground respectively), it just keeps spinning. Is this a "feature" of the board, i.e. not to shutdown completely?

MichaIng, 2021-03-25:
Check `cat /var/tmp/dietpi/logs/dietpi-upgrade_apt.log` for logs about the daily automated upgrades.

> Is this a "feature" of the board, i.e. not to shutdown completely?

Yes, the RPi has no power switch/button, so when it's plugged, it's powered. That's the case for many SBCs btw, but I'd like to have a power button as well, to be true.

pitsi, 2021-03-27:
I am doing something wrong here. I set CONFIG_CHECK_APT_UPDATES to 2, 
```
# cat /boot/dietpi.txt | grep APT_UPDATES
CONFIG_CHECK_APT_UPDATES=2
```
yet dietpi prompted me to do an upgrade in the motd
`apt upgrade     : Run now to apply 2 available APT package upgrades.`
The packages in question today were openssl and libssl1.1.

Thank you for explaining things that seem obvious to you but not to me :)

MichaIng, 2021-03-27:
I just tested it and it works pretty well here:
```
2021-03-27 12:49:13 root@VM-Buster:~# /etc/cron.daily/dietpi

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/dev/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v7.1.0
[ INFO ] DietPi-Update | Latest version  : v7.1.0
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | Storing number of available APT upgrades to file: /run/dietpi/.apt_updates

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : cron
2021-03-27 12:49:49 root@VM-Buster:~# cat /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 14984 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+deb10u9_amd64.deb ...
Unpacking base-files (10.3+deb10u9) over (10.3+deb10u8) ...
Setting up base-files (10.3+deb10u9) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 14984 files and directories currently installed.)
Preparing to unpack .../libsystemd0_241-7~deb10u7_amd64.deb ...
Unpacking libsystemd0:amd64 (241-7~deb10u7) over (241-7~deb10u6) ...
Setting up libsystemd0:amd64 (241-7~deb10u7) ...
(Reading database ... 14984 files and directories currently installed.)
Preparing to unpack .../systemd_241-7~deb10u7_amd64.deb ...
Unpacking systemd (241-7~deb10u7) over (241-7~deb10u6) ...
Preparing to unpack .../udev_241-7~deb10u7_amd64.deb ...
Unpacking udev (241-7~deb10u7) over (241-7~deb10u6) ...
Preparing to unpack .../libudev1_241-7~deb10u7_amd64.deb ...
Unpacking libudev1:amd64 (241-7~deb10u7) over (241-7~deb10u6) ...
Setting up libudev1:amd64 (241-7~deb10u7) ...
Setting up systemd (241-7~deb10u7) ...
(Reading database ... 14984 files and directories currently installed.)
Preparing to unpack .../systemd-sysv_241-7~deb10u7_amd64.deb ...
Unpacking systemd-sysv (241-7~deb10u7) over (241-7~deb10u6) ...
Preparing to unpack .../debian-archive-keyring_2019.1+deb10u1_all.deb ...
Unpacking debian-archive-keyring (2019.1+deb10u1) over (2019.1) ...
Setting up debian-archive-keyring (2019.1+deb10u1) ...
Removing obsolete conffile /etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg ...
Removing obsolete conffile /etc/apt/trusted.gpg.d/debian-archive-jessie-security-automatic.gpg ...
Removing obsolete conffile /etc/apt/trusted.gpg.d/debian-archive-jessie-stable.gpg ...
(Reading database ... 14984 files and directories currently installed.)
Preparing to unpack .../iputils-ping_3%3a20180629-2+deb10u2_amd64.deb ...
Unpacking iputils-ping (3:20180629-2+deb10u2) over (3:20180629-2+deb10u1) ...
Setting up systemd-sysv (241-7~deb10u7) ...
Setting up udev (241-7~deb10u7) ...
Setting up iputils-ping (3:20180629-2+deb10u2) ...
Processing triggers for libc-bin (2.28-10) ...
```
Could you try the following and show the output:
```
sed -i '/APT_UPDATE_MODE=/a\echo "APT_UPDATE_MODE: $APT_UPDATE_MODE"' /etc/cron.daily/dietpi
sed -i 's| > /dev/null||' /etc/cron.daily/dietpi
cat /run/dietpi/.apt_updates /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
```

pitsi, 2021-03-27:
Assuming that the sed commands did work, the cat one returns no file or folder found for both of the files (the greek text means no file or folder found/exists).
```
# cat /run/dietpi/.apt_updates /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
cat: /run/dietpi/.apt_updates: Δεν υπάρχει τέτοιο αρχείο ή κατάλογος
cat: /var/tmp/dietpi/logs/dietpi-upgrade_apt.log: Δεν υπάρχει τέτοιο αρχείο ή κατάλογος
```
Could it be related to my forementined changes for apt's cache?

MichaIng, 2021-03-27:
Ah, I forgot to add executing the script itself:
```
/etc/cron.daily/dietpi
```
But actually, when /run/dietpi/.apt_updates does not exist, then no updates are available and the banner should not show anything related:
```
dietpi-banner 1
```

pitsi, 2021-03-27:
So what shall I change at the above?

MichaIng, 2021-03-27:
Please run the two above commands and paste the output here, so we can see if they do what they should. Also the `sed` command you ran earlier added a small debugging output to the cron job.

pitsi, 2021-03-28:
Full output of the above commands
```
root@dietpi:~ # sed -i '/APT_UPDATE_MODE=/a\echo "APT_UPDATE_MODE: $APT_UPDATE_MODE"' /etc/cron.daily/dietpi
root@dietpi:~ # sed -i 's| > /dev/null||' /etc/cron.daily/dietpi
root@dietpi:~ # cat /run/dietpi/.apt_updates /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
20
cat: /var/tmp/dietpi/logs/dietpi-upgrade_apt.log: Δεν υπάρχει τέτοιο αρχείο ή κατάλογος
```
That 20 there in .apt_updates is obviously the number that is used here.
`apt upgrade     : Run now to apply 20 available APT package upgrades.`
I will delay the upgrades in case it helps troubleshoot this.

MichaIng, 2021-03-28:
Please run those two commands and paste the output here:
```
/etc/cron.daily/dietpi
dietpi-banner 1
```

pitsi, 2021-03-28:
There you go
```
# /etc/cron.daily/dietpi
APT_UPDATE_MODE: 2
APT_UPDATE_MODE: 2
APT_UPDATE_MODE: 2

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v7.0.2
[ INFO ] DietPi-Update | Latest version  : v7.0.2
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease
Hit:5 https://deb.debian.org/debian buster-backports InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | Storing number of available APT upgrades to file: /run/dietpi/.apt_updates

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : avahi-daemon
[  OK  ] DietPi-Services | start : lighttpd
[  OK  ] DietPi-Services | start : cron
```
```
# dietpi-banner 1
 ─────────────────────────────────────────────────────
 DietPi v7.0.2 : 17:42 - Κυρ 28/03/2021
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (aarch64)
 - CPU temp : 31'C : 87'F (Cool runnings)
 - LAN IP : 192.168.1.107 (wlan0)
 - MOTD : Enjoy our new DietPi v7.0 <3
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspberry Pi OS Lite 64-bit)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.
```
I assume that the first command did the upgrade on those 20 packages.

MichaIng, 2021-03-28:
> I assume that the first command did the upgrade on those 20 packages

Yes exactly. You do not see that part of the output as it's redirected to the log file, which should now exist:
```
cat /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
```

So all worked as expected. I'm wondering why it didn't work the last daily cron execution 🤔. But the cron service is running, isn't it?
```
systemctl status cron
journalctl -u cron | grep daily
```

pitsi, 2021-03-28:
Cron does run
```
# systemctl status cron
● cron.service - Regular background program processing daemon
   Loaded: loaded (/lib/systemd/system/cron.service; disabled; vendor preset: en
abled)
   Active: active (running) since Sun 2021-03-28 19:33:45 EEST; 1min
7s ago
     Docs: man:cron(8)
 Main PID: 758 (cron)
    Tasks: 1 (limit: 4487)
   CGroup: /system.slice/cron.service
           └─758 /usr/sbin/cron -f

Μαρ 28 19:33:45 dietpi systemd[1]: Started Regular background program processing
 daemon.
Μαρ 28 19:33:45 dietpi cron[758]: (CRON) INFO (pidfile fd = 3)
Μαρ 28 19:33:45 dietpi cron[758]: (CRON) INFO (Running @reboot jobs)
```
But the other command returns no output. That is the whole journalctl -u cron
```
# journalctl -u cron
-- Logs begin at Thu 2019-02-14 12:11:58 EET, end at Sun 2021-03-28 19:35:41 EES
T. --
Μαρ 28 19:33:45 dietpi systemd[1]: Started Regular background program processing
 daemon.
Μαρ 28 19:33:45 dietpi cron[758]: (CRON) INFO (pidfile fd = 3)
Μαρ 28 19:33:45 dietpi cron[758]: (CRON) INFO (Running @reboot jobs)
```

MichaIng, 2021-03-28:
> But the other command returns no output.

Looks like the daemon did start only just now at `2021-03-28 19:33:45 EEST`. Try again the command tomorrow.

> Look what I just found.

Yes, that's the native daily APT systemd timer, which becomes effective together with [unattended-upgrades](https://packages.debian.org/buster/unattended-upgrades). But it's not a light solution and locks with a certain randomness (intentionally) APT, so that own installs/upgrades fail. So we implement an own solution that is a bit lighter and allows us to combine it with DietPi update checks and the banner.

pitsi, 2021-03-29:
Well, I do not keep the pi running all day long since it is not yet connected on the tv it will work on. Until then, all I can do is connect via ssh to it. And even there, the pi will be shut down every night. I have also considered making a cron job to shut it down, in case the owner forgets it.
The timestamp above probably is from its last boot, which was made in order to post what you requested.

Is there a specific time in the day that cron runs the update? If so, I will have it open at that time. If not, how about using anacron for the job?

pitsi, 2021-03-31:
Is this of any help? I noticed cockpit mentioned that some services were not running, and dietpi-firstboot was one of them 
```
# systemctl status dietpi-firstboot.service
● dietpi-firstboot.service - DietPi-FirstBoot setup
   Loaded: loaded (/etc/systemd/system/dietpi-firstboot.service; disabled; vendo
r preset: enabled)
   Active: inactive (dead)
```

ravenclaw900, 2021-03-31:
Yes, `cron` by default runs it at 1:25(AM) and immediately after boot, but you can see your specific times using `dietpi-cron` and looking for `cron.daily`.  The inactive `dietpi-firstboot` isn't important, as it is used to do the first boot setup and is then disabled.

pitsi, 2021-03-31:
Well... I enabled it and it seems it was my worst decision ever! Now it runs the first run whing again! I will let it finish and report back.

pitsi, 2021-03-31:
So, the hostname was reset to DietPi, the wifi was disabled, the timezone was set back to Europe/London, the keyboard layout to en_gb and the autostart option was reverted to manual login.
At least my stuff (xfce, firefox esr, avahi, lighttpd, cockpit etc) remained installed and the deamons enabled. 

I will probably revert CONFIG_CHECK_APT_UPDATES to 1 and have the user do the updates via cockpit-packagekit.

MichaIng, 2021-03-31:
`dietpi-firstboot.service` applies settings from `dietpi.txt`, so that basic settings, like localisations and network can be automatically set and as such intended for first boot, as its name states 😉. When you use `dietpi-config` to change those, the related entries in `dietpi.txt` are changed as well, so that `dietpi-firstboot.service` would only re-apply what is already applied. Otherwise, jep it does what it expects to do 😄. But it should be a thing of 5 minutes to revert what has changed, isn't it?CONFIG_CHECK_APT_UPDATES has nothing to do with that one, btw.

pitsi, 2021-04-01:
Yea, I noticed all that :) Any ideas on how a simple user could update the system via gui?

Joulinar, 2021-04-02:
`apt update && apt upgrade`

will update your packages. By default, there is no GUI in DietPi you can click on to update `apt` managed packages.

MichaIng, 2021-04-02:
What do you mean by "gui"? It's the following two commands for APT package updates and DietPi updates (which includes APT updates as well):
```
apt upgrade
dietpi-update
```
with apt being a console tool and `dietpi-update` at least a whiptail dialogue.
The by default daily checks + the banner notifications that show these exact two commands to users, are what addresses "simple" users that might not know about or do these checks and regular upgrades by themselves, so it's the simplest and fastest way to keep your system updated.

When you install a desktop, `apt install synaptic` installs [Synaptic](https://packages.debian.org/buster/synaptic), the most common GUI frontend for APT, and a `dietpi-update` desktop shortcut is created (or at least for the `dietpi-launcher`), while it stays a console script, opening in the terminal emulator. Synaptic is nice when you want to browse through installed packages and see details about them, but for keeping the system updated it's an overkill and slows down the process compared to running `apt upgrade` from console, which is naturally true for all GUI frontends, when one knows the console command 😉.

pitsi, 2021-04-02:
In case I forgot to mention it above, I am on debian for the last 13 years, so I know the apt commands and synaptic already.
I update both my systems with apt-get update/upgrade/dist-upgrade from the terminal every day, but a simple user who has never seen a terminal, just won't use it. As for synaptic, it is very good as a gui for apt, but its countless options are too much for that simple user.

Thus the need for a gui or another simple way of updating. In cockpit, you just connect to it, click updates on the left side of the page and packagekit does the rest. I can't think of anything simpler than that.
Major desktop enviroments include an app that sits on the tray and informs the user for package upgrades.

Joulinar, 2021-04-02:
You can create a cron job to shutdown the system. 

If you like  a simple to click gui, you can have a look to Webmin. It allows to install apt packages via a web page.

MichaIng, 2021-04-02:
> As for synaptic, it is very good as a gui for apt, but its countless options are too much for that simple user.

I agree. Especially for inexperienced/new users a simple clear notice/instruction about what is important is best.

> In cockpit, you just connect to it, click updates on the left side of the page and packagekit does the rest. I can't think of anything simpler than that.

I never used it, but since it is not GUI application but provides an API, you need to use a client, right? And a client surely has limited ability to handle upgrade errors, show interactive dialogues and questions shipped with the package upgrade. So I'm not convinced that it is the best option for inexperienced users. When running `apt upgrade` from the console, you are faced with all interactive choices and see actual errors you can then search online force, like expired APT key, suite changes (e.g. when Bullseye is released, the Buster repo is changed to "oldstable", breaking non-interactive APT upgrades) and alike issues. So IMO web/client interface frontends for APT are good only for experienced users, to simplify/speed up their remove maintenance, but who know what to do when an APT upgrade fails in the middle.

> since the CONFIG_CHECK_APT_UPDATES=2 does not work

But it does work, as we have verified above? But again, this is something that inexperienced users should not use, as a direct feedback for any failure is important especially for inexperienced users. You need to be able to at least associate it with the APT package upgrades, where to find APT/dpkg logs/status, that you should better enable persistent APT/dpkg logs then etc, to know where to start.

> You can create a cron job to shutdown the system.

Agreed. A systemd timer would do as well, but for such exact times, cron is the simplest solution.

> If you like a simple to click gui, you can have a look to Webmin. It allows to install apt packages via a web page.

Jep, an alternative to cockpit, if I understood that right. But again, such non-native frontends are food for convenience, but not to simplify something for inexperienced users. Nothing is safer and simpler "to setup"/do and debug then running "apt upgrade" from console. When one runs a server, using the console, via SSH when required remotely, is the very first and most important thing to learn and become convenient with, at least IMO 😉.

pitsi, 2021-04-02:
To be honest, I had tried webmin once, back in the days where systemd was not the default init system for debian, and it messed up some of my samba configurations, so I removed it. I had also tried a similar web administration tool, but I did not like it too.

As for the dietpi auto update, it does work for me, when it (= all its scripts seperately) is run manually. When it is run automatically, it does not seem to. Of course, I can provide all sorts of help if you want to debug it, but this will be until Sunday.
I would do the upgrades myself, remotely, from the convenience of my own pc, if it was possible. But it is not.
I can set a static ip for the pi and register a domain name for dynamic dns, but the fking router the isp provides does not provide the options to set port forwarding with the default credentials!

Last but not least, all the repos in sources.list say "buster", so I am certain nothing serious will happen once bullseye becomes the next stable. If it all works well, I will do the upgrade to it sometime next year. Dietpi does support dist-upgrades from version to version, right?

Joulinar, 2021-04-02:
> Dietpi does support dist-upgrades from version to version, right?

Not really. @MichaIng provided some steps how it could work from Stretch to Buster, but nothing we support officially

Joulinar, 2021-04-02:
Sorry for closing. To small button on my mobile phone 🥺

pitsi, 2021-04-02:
Then I will have to reinstall it fresh next year :P

Joulinar, 2021-04-02:
I guess Bullseye will be released this summer. At least this was the plan if I'm not mistaken. Not sure if this changed due to Corona.

MichaIng, 2021-04-02:
> When it is run automatically, it does not seem to.

I cannot think of a way how it is possible to succeed when run manually while it fails when called by cron, so it would be great if you could switch/leave it at `=2` and check back `/var/tmp/dietpi/logs/dietpi-upgrade_apt.log` by times.

> @MichaIng provided some steps how it could work from Stretch to Buster, but nothing we support officially. Usually it is done on own risk.

Actually I aim to support it for Buster => Bullseye. Users will do it anyway and pro-actively it's better to provide good instructions and document which software titles probably need to be reinstalled, or which configs adjusted to keep them compatible with the Bullseye packages.

> Last but not least, all the repos in sources.list say "buster", so I am certain nothing serious will happen once bullseye becomes the next stable.

When the suite code name (testing, stable, oldstable, oldoldstable, ...) changes, like the "buster" suite code name changes from "stable" to "oldstable", `apt-get update` fails, without a manual confirmation on STDIN, at least when not using the `--allow-releaseinfo-change` or `--allow-releaseinfo-change-suite` option. It's just one example where any non-native apt wrapper or UI would fail, while a manual call would easily succeed and provides information that might be interesting for the user. It reminds me that we should either add this to our `G_AGUP` function or use a pre-patch to have it called once on first DietPi update after the release.

pitsi, 2021-04-04:
"Deployment day" was moved once more for 1 week (from Mar 29th to Apr 5th to Apr 12th now), so I have one week to test autoupdate. I set the parameter back to 2 and I hope debian stable gets at least one update this week.

pitsi, 2021-04-12:
Last comment because the rpi is now in place.
First of all, there were no updates in debian stable this week, so I could not test if the parameter works.
Second, now that it is connected wirelessly. I noticed that as the os boots, it still fails to contact the ntp server to get the current time. If i disable it (next month that I will see it up close), will the system have the right time after boot?

Feel free to close the issue if you want.

MichaIng, 2021-04-12:
> I noticed that as the os boots, it still fails to contact the ntp server to get the current time.

Do you mean the attempt times out?
```
journalctl -u dietpi-boot
```
Does it work when calling it manually after boot?
```
/boot/dietpi/func/run_ntpd 1
```
Else try to switch to daemon mode and watch the service if it succeeds after that 60 seconds:
```
/boot/dietpi/func/dietpi-set_software ntpd-mode 4
systemctl status systemd-timesyncd
journalctl -fu systemd-timesyncd
# Wait until you see something like "Initial synchronization to time server 185.162.250.228:123 (0.debian.pool.ntp.org)."
# Or if that never happens and you see only timeouts, then there is an NTP connectivity from that device indeed.
```
Keeping network time sync disabled is not an option, every system that uses the internet must have an NTP daemon. If `systemd-timesyncd` does not work, then try `ntp`:
```
/boot/dietpi/func/dietpi-set_software ntpd-mode 0 # disables our NTP attempts at boot
apt install ntp
```
Or course you can also try to use different NTP mirrors to sync time with, e.g. our router/gateway or a specific pool mirror closer to your location, via `dietpi-config` > `Network Option: Misc` > `NTP Mirror`.

pitsi, 2021-04-12:
Well, I can't test all that now, but I will next time I check the pi.
What I mean is a message that comes out exactly 10 times at boot and has to do with ntp. Every attempt times out because it is not connected to the internet, although it should be... Internet is available once it reaches the desktop and works with no issues.

Joulinar, 2021-04-12:
Usually time sync time out is 60 during boot (default). Means if it's working after 10 attempts,  it succeeds. 🤔

pitsi, 2021-04-12:
Ok then, I leave it as it is.

MichaIng, 2021-04-12:
The network check loops 10 times, but that is not about NTP but raw network route only.

MichaIng, 2021-05-02:
I mark this as closed. If any issues remain, please open an individual issue, which is easier to track.

closed 2021-05-02 by MichaIng

pitsi, 2021-05-03:
Ok. I have not seen the rpi4 since I put it in place, but I will contact the owner some day this week and ask him to check if the updates were made, That is my main concern :/