MichaIng/DietPi #4214: GUI startup fails with dietpi account.

SiegmundSchreiber, 2021-03-20:
### Creating a bug report/issue

#### Required Information
- DietPi version: v7.0.2
- Distro version: Buster
- Kernel version: Linux DietPi 5.10.17-v7+
- SBC model: RPi 2 Model B (armv7l)
- Power supply used: 5V 2.4A
- SDcard used: SanDisk ultra

#### Additional Information (if applicable)
- Software title: LXDE
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce

1. Start LXDE by Autostart or by executing startx using the account of dietpi (1000)

#### Expected behaviour

- Desktop should appear

#### Actual behaviour

Fatal server error:
xf86OpenConsole: Cannot open virtual console 1 (Permission denied)

#### Extra details

- When changing to root startx is working correctly
- Same behavior with xfce

Joulinar, 2021-03-20:
Hi,

This is expected at the moment. It is not possible to run a desktop as non root user. A rework is in progress and might be available on next version

MichaIng, 2021-03-20:
It's possible, but not with all GPUs/drivers OOTB. In your particular case it seems like only TTY permissions are missing. Could you try: `usermod -aG video,render,tty dietpi`

SiegmundSchreiber, 2021-03-20:
Oh, sorry I was not aware.

Thanks for your fast response!

closed 2021-03-20 by SiegmundSchreiber