MichaIng/DietPi #4215: WiFi : Not Found | [On] | Disconnected Rpi4

fosron, 2021-03-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | ```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'```
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Official
- SDcard used | Official

#### Report

WiFi just stopped working on RPi4.

[image]

Tried enabling, disabling, reinstalling packages etc.
Had static IP setup. Switched to Eth, but that's not ideal with Wifi available on Rpi.

```
dietpi@DietPi:~$ cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.1.101
netmask 255.255.255.0
gateway 192.168.1.254
#dns-nameservers 1.1.1.1 1.0.0.1

# WiFi
allow-hotplug wlan0
iface wlan0 inet dhcp
address 0.0.0.0
netmask 0.0.0.0
gateway 0.0.0.0
#dns-nameservers 0.0.0.0
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

```
dietpi@DietPi:~$ sudo cat /etc/wpa_supplicant/wpa_supplicant.conf
# Grant all members of group "netdev" permissions to configure WiFi, e.g. via wpa_cli or wpa_gui
ctrl_interface=DIR=/run/wpa_supplicant GROUP=netdev
# Allow wpa_cli/wpa_gui to overwrite this config file
update_config=1
network={
	ssid="Tautwifi 5GHz"
	scan_ssid=1
	key_mgmt=WPA-PSK
	psk="******"
}
```

```
dietpi@DietPi:~$ cat /etc/resolv.conf
nameserver 1.1.1.1
nameserver 1.0.0.1
```
[... 240 lines cut ...]
SOFTWARE_XORG_DPI=96

# Chromium
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com/

# Home Assistant
# - Optional Python build dependencies and modules, possibly required for certain HA components
#	Space separated list (no quotation!), will be installed together with Home Assistant automatically, if present
SOFTWARE_HOMEASSISTANT_APT_DEPS=
#	Add Python modules with version string at best, e.g.: firstModule==1.2.3 secondModule==4.5.6
SOFTWARE_HOMEASSISTANT_PIP_DEPS=

#------------------------------------------------------------------------------------------------------
##### Dev settings #####
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=beta
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
##### Settings, automatically added by dietpi-update #####
#------------------------------------------------------------------------------------------------------
AUTO_SETUP_BACKUP_RESTORE=0
```

Joulinar, 2021-03-20:
Hi,

Many thanks for your message. Did you reboot after enabling WiFi on `diepi-config`

fosron, 2021-03-22:
@Joulinar yes of course, several times, to no avail

Joulinar, 2021-03-22:
maybe the adapter is blocked. can you check `rfkill list`

fosron, 2021-03-22:
@Joulinar ```
dietpi@DietPi:~$ rfkill list
-bash: rfkill: command not found
dietpi@DietPi:~$ sudo rfkill list
sudo: rfkill: command not found
```

Joulinar, 2021-03-22:
did you tried to go `dietpi-config` > `7  : Network Options: Adapters` > `WiFi`

Usually you should be get following `WiFi must be enabled before settings can be changed.` > `ok`

This will drop your network connection and WiFi should be available afterwards

fosron, 2021-03-22:
@Joulinar as i've mentioned in the issue - `Tried enabling, disabling, reinstalling packages etc.` - tried various things from `dietpi-config` with no success.

MichaIng, 2021-03-23:
Can you verify:
- `grep 'disable-wifi' /boot/config.txt` does not output anything.
- You rebooted meanwhile.
- `dietpi-config` still shows "WiFi : Not found"?
- If so, can you paste: `ip l; ls -l /sys/class/net`

fosron, 2021-04-06:
@MichaIng 

```
dietpi@DietPi:~$ grep 'disable-wifi' /boot/config.txt
dietpi@DietPi:~$
```

```
dietpi@DietPi:~$ ip l; ls -l /sys/class/net
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:28:3f:41 brd ff:ff:ff:ff:ff:ff
3: br-252301c027c7: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default
    link/ether 02:42:b1:bc:b5:c9 brd ff:ff:ff:ff:ff:ff
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default
    link/ether 02:42:ee:0d:83:8b brd ff:ff:ff:ff:ff:ff
10: veth015e464@if9: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP mode DEFAULT group default
    link/ether 66:6e:ef:ad:ac:c6 brd ff:ff:ff:ff:ff:ff link-netnsid 0
12: veth81c017d@if11: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP mode DEFAULT group default
    link/ether b6:60:8a:e9:fe:27 brd ff:ff:ff:ff:ff:ff link-netnsid 1
total 0
lrwxrwxrwx 1 root root 0 Apr  6 18:01 br-252301c027c7 -> ../../devices/virtual/net/br-252301c027c7
lrwxrwxrwx 1 root root 0 Apr  6 18:01 docker0 -> ../../devices/virtual/net/docker0
lrwxrwxrwx 1 root root 0 Apr  6 18:01 eth0 -> ../../devices/platform/scb/fd580000.ethernet/net/eth0
lrwxrwxrwx 1 root root 0 Apr  6 18:01 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx 1 root root 0 Apr  6 18:01 veth015e464 -> ../../devices/virtual/net/veth015e464
lrwxrwxrwx 1 root root 0 Apr  6 18:01 veth81c017d -> ../../devices/virtual/net/veth81c017d
```

fosron, 2021-04-06:
Also, i got these errors while trying to enable wifi via `dietpi-config`

[image]
[image]

MichaIng, 2021-04-06:
Hmm, the `wlan0` interface is not present, i.e. the device is not detected by the kernel. I remember that there was a case where Bluetooth was disabled when WiFi was disabled on a specific RPi revision.

Just to rule it out, please check whether Bluetooth is disabled (the device tree overlay) and in case enable it + reboot:
```
grep disable-bt /boot/config.txt && sed -i '/disable-bt/d' /boot/config.txt
reboot
```

If `ip l` still does not list it, it's a kernel or hardware issue that we should forward to RPi devs 🤔. Probably `dmesg` shows some related messages, e.g. errors about WiFi detection or such.

Joulinar, 2021-04-06:
Quick test could be using plain Raspberry OS on a spare SD card to check if WiFi is detected

closed 2021-05-07 by MichaIng