MichaIng/DietPi #4216: Support for OLED-Display

Gill-Bates, 2021-03-21:
For the Raspberry Pi I built an external status display.

[image]

**Ingredients:**
1x 128x64 OLED display from Alibaba ($5).
4x GIPO connectors (50 cents)
1x double sided adhesive tape

However, the software installation was complex. How about including this solution in the software catalog? Then everyone who wants to could build a status monitor in no time.

GPIO
1 = 3,3V (Grau)
6 = GND (Blau)
5 = SCL (Orange)
3 = SDA (Violett)

Manual: https://indibit.de/raspberry-pi-oled-display-128x64-mit-python-ansteuern-i2c/

Install Pyhton3 Pip from `dietpi-software` catalouge
`dietpi-software`

Enable i2C-Support
`dietpi-config ` -> `Advanced Options` -> `I2C State ON`

Get I2C Bus-Address of Display. Normally 0x3c
`i2cdetect -y 1`

Install Stuff
`sudo apt-get install python-smbus i2c-tools git python-pil build-essential libfreetype6-dev`

Install Pillow & psutil
```
sudo python3 -m pip install --upgrade pip
sudo python3 -m pip install --upgrade Pillow
```

`sudo pip install psutil`

Clone Python Library
`git clone https://github.com/BLavery/lib_oled96`

`touch display.py; chmod 777 display.py; nano display.py`

`python3 ./display.py`

Run after Reboot
`crontab -e`

```
# Display
/usr/bin/python3.7 /root/lib_oled96/display.py
@reboot sudo /usr/bin/python3 /root/lib_oled96/display.py &
```

MichaIng, 2021-03-23:
Many thanks for your request.

A few questions:
- You install a Python 2 Pillow fork package first (`python-pil`) and the actual Python 3 Pillow via pip afterwards. The example scripts are fully Python 3 compatible, so `python-pil` is not required, right?
- Similarly instead of `python-smbus` (Python 2), `python3-smbus` should be installed. But that's done together with `i2c-tools` anyway when enabling I2C from `dietpi-config`, so those can be skipped as extra steps as well.
- The repository is basically the Python library, some fonts and a logo and some example output scripts which use them, right? So we could place them basically anywhere and create a systemd unit to start them at boot, which is a bit more common compared to the cron reboot feature.

Since it's not practical (not sure if/how possible?) to show the regular console on that little display, I think it's not practical to have it in the diepti-config Display Options LCD list, but as a separate dietpi-software (?) option as additional status monitor. I'll check if dietpi-cloudshell could possibly run on it with a certain very small font. Otherwise we can create an own simple default script so that after install something is shown on the screen OOTB.

What I don't like about this particular library is that it's not maintained anymore since three years. The original one, which that fork was based on, still is: https://github.com/rm-hull/luma.oled
It has a full documentation: https://luma-oled.readthedocs.io/en/latest/intro.html
And can be installed right from PyPI: https://luma-oled.readthedocs.io/en/latest/software.html
That looks like the better solution to me, if you could verify it works as well, I'd go with that one 🙂.

Trunkzeh, 2021-04-03:
I have a similar setup on my pizero; using the adafruit pioled screen. Not sure how dietpi will address every oled / lcd screen out there so this might not be workable. Wasnt difficult to install in fairness; would have been less than 5minutes if it wasnt for the crappy processor on the zero :)