MichaIng/DietPi #4217: Kernel version

Maaago, 2021-03-21:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Yes, see #4129 and #4127

#### Describe the solution you'd like:
A menu entry in dietpi-config that let you select the kernel's version

#### Describe alternatives you've considered:
The kernel's version select menu entry as in armbian-config. 

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/
Vote here: https://feathub.com/MichaIng/DietPi/+220

Joulinar, 2021-03-21:
Well, DietPi is supporting quite a large amount of SBC and other devices, running different base images like Armbian, Meveric, plain Debian or Raspberry OS. And on top DietPi support different Debian version like Stretch and Buster. This could be challenging to support all type of combination.

Maaago, 2021-03-21:
Maybe we can start from some SBCs and give just few options like stable, testing and maybe old stable. I'm not an expert about these things, but I can help if it's needed.

MichaIng, 2021-03-23:
Well, Armbian ships two kernel version: "current" and "legacy"
The "legacy" kernel is sometimes from the manufacturers kernel sources, sometimes, AFAIK older upstream kernel sources. The "current" kernel is the latest stable kernel that Armbian provides, and while it ships bugs sometimes, it as well usually fixes them, as in the issue you linked. So basically when one faces issues, the first thing is to upgrade APT packages, which includes the kernel automatically, else we need to report to Armbian, if it has not yet been done. Providing and officially supporting a switch between those two kernels breaks some feature and software implementation, and I personally cannot effort the time to implement and maintain this.

If for some reason, the manufacturers kernel sources are better, we use them a different way than via Armbian, e.g. like we do for Odroids and ROCK Pi S, where more hardware features are available.

What I plan with Debian Bullseye is basically switching with all boards that use the Armbian kernel to the Debian kernel. Users can then control the kernel version a but by using backports or not. That is another argument why I'll personally not start to implement any additional features for Armbian-based images.

Maaago, 2021-03-23:
Ok, so the effort put into this isn't worth for the results we would get... Thank you anyway!

closed 2021-03-23 by Maaago

MichaIng, 2021-03-23:
Yes I would say so. With enough man power, of course multiple versions could be maintained and features for them, but currently I'd ship the latest stable version only and, in case of issues, help with forwarding and fixing issues with the kernel maintainers quickly, like we do with Raspberry Pi.