MichaIng/DietPi #4218: unbound servfails and "anchor could not be fixed"

coldbootinitheart, 2021-03-22:
Tried running unbound on raspbian os but got the error mentioned in the title, synced time, made sure to regenerate the root.key file, double checked the ownership and permission to no avail. Decided to nuke my sdcard to install the preconfigured dietpi thinking it might be a mistake on my part, still the same issue. I do not have a good grasp of networking and am pretty sure it has nothing to do with diet-pi but decided to ask help here anyways, in case. Here are some log files:
[dietpi.confg](http://paste.debian.net/1190404)
[systemctl status unbound](http://paste.debian.net/1190406)
[dig unbound output](http://paste.debian.net/1190407/)
[normal dig output](http://paste.debian.net/1190408/)
[unbound logs when trying to dig google.com](http://paste.debian.net/1190410/)

I am using openwrt on my router if that matters.
Any help would be greatly appreciated!

coldbootinitheart, 2021-03-22:
tried with `use-caps-for-id: no` too
but, seems to not fix the issue: [unbound logs](http://paste.debian.net/1190445/)
Increased verbosity to 5: [logs](http://paste.debian.net/1190446/)

ravenclaw900, 2021-03-22:
Hello,
Could you make sure "Override DNS Settings for All Clients" is turned off on your router.

coldbootinitheart, 2021-03-22:
Oh god I am pretty sure I sent you the wrong logs I am extremely sorry
correct logs: 
[unbound log](http://paste.debian.net/1190476/)
[dig log](http://paste.debian.net/1190477/)

I could not find such setting in openwrt even after looking through some docs. 
I am pretty sure openwrt is not overriding dns requests as my android phones connected to the same netwok use the google dns (8.8.8.8) instead of the quad9 (9.9.9.9) dns specified in openwrt.

Trunkzeh, 2021-03-23:
I had similar issues as yourself; unbound would not work and running systemctl status unbound returned anchor could not be fixed. I'm going to re-image DietPi on my SD and start again. For reference; unbound was installed as part of a pihole installation on the latest stable dietpi on a RPI4 (7.0.x)

Joulinar, 2021-03-23:
Are you running 64bit or 32bit image on the RPi4?

Trunkzeh, 2021-03-23:
Standard 32bit image.

Joulinar, 2021-03-23:
hmm I did a fresh test installation of PiHole + Unbound together on RPi4B 32bit. Both Are working withput issues and Unbound is starting correctly after a reboot

```
root@DietPi4:~# systemctl status unbound
● unbound.service - Unbound DNS server
   Loaded: loaded (/lib/systemd/system/unbound.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/unbound.service.d
           └─dietpi.conf
   Active: active (running) since Tue 2021-03-23 15:42:15 CET; 1min 26s ago
     Docs: man:unbound(8)
 Main PID: 487 (unbound)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/unbound.service
           └─487 /usr/sbin/unbound -d

Mar 23 15:42:15 DietPi4 systemd[1]: Starting Unbound DNS server...
Mar 23 15:42:15 DietPi4 package-helper[473]: /var/lib/unbound/root.key has content
Mar 23 15:42:15 DietPi4 package-helper[473]: success: the anchor is ok
Mar 23 15:42:15 DietPi4 unbound[487]: [487:0] info: start of service (unbound 1.9.0).
Mar 23 15:42:15 DietPi4 systemd[1]: Started Unbound DNS server.
root@DietPi4:~#
```

Can you check if the following file is existing `cat /etc/systemd/system/ifup@.service.d/dietpi.conf`

As well how does it behave if you restart unbound
```
systemctl daemon-reload
systemctl restart unbound
systemctl status unbound
```

Trunkzeh, 2021-03-23:
> Can you check if the following file is existing cat ***@***.***/dietpi.conf

OK - I’m going to reinstall unbound via dietpi-software and report back shortly.

Trunkzeh, 2021-03-23:
OK. 

cat /etc/systemd/system/ifup@.service.d/dietpi.conf

```
# Override service type to give network(-online).target more meaning and remove obsolete ifquery call: https://github.com/MichaIng/DietPi/issues/4121#issuecomment-782057374
[Service]
Type=oneshot
ExecStart=ifup --allow=hotplug %I
```

I've installed unbound; systemctl status unbound reports:

```
Mar 23 15:53:06 DietPi systemd[1]: Starting Unbound DNS server...
Mar 23 15:53:06 DietPi package-helper[11197]: /var/lib/unbound/root.key does not exist, copying from /usr/share/dns/root.key
Mar 23 15:53:57 DietPi package-helper[11197]: /var/lib/unbound/root.key has content
Mar 23 15:53:57 DietPi package-helper[11197]: fail: the anchor is NOT ok and could not be fixed
Mar 23 15:53:57 DietPi unbound[11203]: [11203:0] info: start of service (unbound 1.9.0).
Mar 23 15:53:57 DietPi systemd[1]: Started Unbound DNS server.
```
After daemon-reload and unbound restart:

```
Mar 23 15:57:12 DietPi systemd[1]: Starting Unbound DNS server...
Mar 23 15:58:03 DietPi package-helper[11770]: /var/lib/unbound/root.key has content
Mar 23 15:58:03 DietPi package-helper[11770]: fail: the anchor is NOT ok and could not be fixed
Mar 23 15:58:03 DietPi unbound[11778]: [11778:0] info: start of service (unbound 1.9.0).
Mar 23 15:58:03 DietPi systemd[1]: Started Unbound DNS server.
```

MichaIng, 2021-03-23:
@coldbootinitheart 
You used the wrong port for your `dig` command. Could you retry with `127.0.0.1#5335`? We switched with DietPi v7.0.2 from 5353 to 5335 (for new installs) since 5353 is the default for mDNS already.

The anchor error btw does not necessarily mean that Unbound is not usable afterwards. @Trunkzeh how did you recognise that Unbound does not work?

Here a little background on DNSSEC and the anchor: https://nlnetlabs.nl/documentation/unbound/howto-anchor/
This can be used to test updating the anchor manually: `sudo -u unbound unbound-anchor -v -a /var/lib/unbound/root.key`
DNSSEC can e.g. fail if the system clock is out of sync. Can you both verify that `date` shows the correct time?

Trunkzeh, 2021-03-23:
@MichaIng time & date correct on my RPI4. I was not able to resolve DNS when pihole was set to the local unbound server (127.0.0.1:5335) however works fine when set to cloudflare for example; thats how i tested to see if unbound was working.

MichaIng, 2021-03-23:
Could you try it with dig?
```
apt install dnsutils
dig google.com @127.0.0.1 -p5335
```

Trunkzeh, 2021-03-23:
Tried that earlier:

```
; <<>> DiG 9.11.5-P4-5.1+deb10u3-Raspbian <<>> google.com @127.0.0.1 -p5335
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 44652
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;google.com.			IN	A

;; Query time: 4941 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1)
;; WHEN: Tue Mar 23 22:47:17 GMT 2021
;; MSG SIZE  rcvd: 39
```

MichaIng, 2021-03-23:
Manually updating the anchor fails as well?
```
sudo -u unbound unbound-anchor -v -a /var/lib/unbound/root.key
```
Ah and after resetting the key?
```
mv /usr/share/dns/root.key /var/lib/unbound/root.key
sudo -u unbound unbound-anchor -v -a /var/lib/unbound/root.key
```

Trunkzeh, 2021-03-23:
I ran the first command earlier and it simply hanged; had to kill it with a cmd + c. I'll do the reset and feedback :-)

MichaIng, 2021-03-23:
Okay, I'll go through the [settings](https://github.com/MichaIng/DietPi/blob/master/.conf/dps_182/unbound.conf) and see if we might need to change a hardening related to DNSSEC or how this error can happen.

Trunkzeh, 2021-03-23:
No problem; I wont wipe my system just yet then incase there's some further checks you want done :) Nothing important on there either.

Joulinar, 2021-03-23:
out of curiosity, can you try test following: 

1. open 1st SSH session
2. install `tcpdump` > apt install `tcpdump`
2. run `tcpdump -i any -c500 -nn port 5335 or 53`
3. open a 2nd SSH session
4. run `dig dietpi.com @127.0.0.1 -p5335`
5. check the output on 1st SSH session, it should looks like this if all goes well
```
root@DietPi3:~# tcpdump -i any -c500 -nn port 5335 or 53
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
00:16:45.158596 IP 127.0.0.1.57163 > 127.0.0.1.5335: UDP, length 51
00:16:45.159111 IP 192.168.0.12.65183 > 192.42.93.30.53: 51470% [1au] A? dIeTpi.com. (39)
00:16:45.289132 IP 192.42.93.30.53 > 192.168.0.12.65183: 51470- 0/4/13 (599)
00:16:45.289655 IP 192.168.0.12.54020 > 172.64.33.176.53: 16654% [1au] A? DiETpi.COm. (39)
00:16:45.289862 IP 192.168.0.12.34902 > 108.162.192.167.53: 3102% [1au] DNSKEY? dIEtPi.cOm. (39)
00:16:45.325313 IP 172.64.33.176.53 > 192.168.0.12.54020: 16654*- 3/0/1 A 172.67.173.4, A 104.21.96.47, RRSIG (177)
00:16:45.326628 IP 108.162.192.167.53 > 192.168.0.12.34902: 3102*- 3/0/1 DNSKEY, DNSKEY, RRSIG (305)
00:16:45.329969 IP 127.0.0.1.5335 > 127.0.0.1.57163: UDP, length 71
```

Trunkzeh, 2021-03-23:
What I got:

```
root@DietPi:/etc/unbound/unbound.conf.d# tcpdump -i any -c500 -nn port 5335
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
23:21:58.263314 IP 127.0.0.1.55236 > 127.0.0.1.5335: UDP, length 51
23:22:03.263001 IP 127.0.0.1.55236 > 127.0.0.1.5335: UDP, length 51
23:22:05.120822 IP 127.0.0.1.5335 > 127.0.0.1.55236: UDP, length 39
23:22:05.121141 IP 127.0.0.1.5335 > 127.0.0.1.55236: UDP, length 39
```

Joulinar, 2021-03-23:
pls can you try a domain you never checked before. right now it looks Unbound is using it's own cache and did not ask upstream DNS server.

Trunkzeh, 2021-03-23:
Port 53 gives significantly more results:

```
23:26:15.928045 IP 127.0.0.1.45978 > 127.0.0.1.5335: UDP, length 50
23:26:16.251073 IP 10.10.24.15.59556 > 192.203.230.10.53: 42082% [1au] NS? . (28)
23:26:17.005942 IP 127.0.0.1.5335 > 127.0.0.1.45978: UDP, length 38
23:26:17.006530 IP 127.0.0.1.5335 > 127.0.0.1.45978: UDP, length 38
23:26:17.789828 IP 10.10.24.25.59200 > 10.10.24.15.53: 47849+ A? tp.bfbdc3ca1-frontier.amazon.co.uk. (52)
23:26:17.790975 IP 10.10.24.15.6660 > 1.0.0.1.53: 25144+ A? tp.bfbdc3ca1-frontier.amazon.co.uk. (52)
23:26:17.802736 IP 1.0.0.1.53 > 10.10.24.15.6660: 25144 2/0/0 CNAME dmv2chczz9u6u.cloudfront.net., A 13.224.131.182 (110)
23:26:17.803876 IP 10.10.24.15.53 > 10.10.24.25.59200: 47849 2/0/0 CNAME dmv2chczz9u6u.cloudfront.net., A 13.224.131.182 (110)
23:26:21.013663 IP 10.10.24.25.54843 > 10.10.24.15.53: 21980+ Type65? fls-eu.amazon.co.uk. (37)
23:26:21.013681 IP 10.10.24.25.57891 > 10.10.24.15.53: 16217+ A? fls-eu.amazon.co.uk. (37)
23:26:21.014339 IP 10.10.24.15.53 > 10.10.24.25.54843: 21980* 1/0/0 A 0.0.0.0 (53)
23:26:21.014765 IP 10.10.24.15.53 > 10.10.24.25.57891: 16217* 1/0/0 A 0.0.0.0 (53)
23:26:33.789775 IP 10.10.24.22.56013 > 10.10.24.15.53: 7309+ A? time-osx.g.aaplimg.com. (40)
23:26:33.790894 IP 10.10.24.15.53 > 10.10.24.22.56013: 7309 5/0/0 A 17.253.108.253, A 17.253.108.125, A 17.253.34.251, A 17.253.34.125, A 17.253.34.253 (120)
```

Trunkzeh, 2021-03-23:
Found the problem.. One of the fw rules responsible for DNS amplication attacks is catching something; when disabled dig fetches the correct dns response. Sorry guys!

coldbootinitheart, 2021-03-24:
I don't see any RRSIGs in the responses. I have no idea why but DNSSEC seems to get stripped from upstream network
Also unbound reports that the upstream servers are not authoritative but instead resolvers themselves (REC_LAME). 

[dig log](https://paste.debian.net/1190781/)
[unbound log](https://paste.debian.net/1190783/) btw the unbound log was ~3000 lines long but I just provided th last 1000 lines
[tcpdum log](https://paste.debian.net/1190784/)
[unbound config](https://paste.debian.net/1190785/)

All the above erros seem to vanish if I forward traffic to custom dns resolvers using the `forward-zone` unbound option.
I am extremely lost here.

coldbootinitheart, 2021-03-24:
So I used my mobile data on my pi with the unbound setup (with out `forward-zone`) and everything started working again. This can only mean that my ISP is doing something to prevent dnssec from working? I am extremely concerned and would like to know if something can be done (besides `forward-zone` option) as I won't be able to change ISPs easily.

MichaIng, 2021-03-24:
It's the SERVFAIL answer, which can have several reasons, including temporary network issues and such. Since DNS is natively on port 53, there is no way to resolve via DNS root servers on a different port.

There is a similar issue reported on the Pi-hole forum: https://discourse.pi-hole.net/t/new-pi-hole-with-unbound-not-functional/42256
Probably you can go through the debugging steps suggested by jfb as well.

Another idea, just to rule it out: Could you try to disable the hardening settings in your `/etc/unbound/unbound.conf.d/dietpi.conf`: https://github.com/MichaIng/DietPi/blob/master/.conf/dps_182/unbound.conf#L68-L73

coldbootinitheart, 2021-03-25:
Even after disabling the hardening settings and troubleshooting a lot unbound seems non functional. I am pretty sure that my ISP is intercepting my dns queries and resolving them via some proxy. DNS-over-tls works but I do not want to rely on a third party ISP resolver. I know this has nothing to do with dietpi but is there some method of redirecting all traffic on port 53 on the router to my unbound running on pihole? Or is it impossible to circumvent this without using DNS-over-tls?

Trunkzeh, 2021-03-25:
Yes but thats dependant on your router. I have NAT rules on my mikrotik to intercept and forward all DNS queries not originating from my RPI4 to said RPI4.

MichaIng, 2021-03-25:
Most routers have a simple setting to change the upstream DNS server, which can then be set to the local Unbound/Pi-hole server. Clients receive that via DHCP along with their IP.

coldbootinitheart, 2021-03-26:
sorry for being late to message. No matter what upstream dns server I set my router (even if the dns server in question is bogus mind you) the dns queries get intercepted and go through their proxy.

[dig log with upstream dns](https://paste.debian.net/1191088/)
[dig log with quad9 dns](https://paste.debian.net/1191089/)
[dig log with bogus dns](https://paste.debian.net/1191090/)

Trunkzeh, 2021-03-26:
Unbound supports both DoH and DoT; DoH should get around your ISP proxying efforts.

MichaIng, 2021-03-26:
At least to test DoT, which can be done easily:
```
cat << '_EOF_' > /etc/unbound/unbound.conf.d/dietpi-dot.conf
# Adding DNS-over-TLS support 
server:
tls-cert-bundle: /etc/ssl/certs/ca-certificates.crt
forward-zone:
name: "."
forward-tls-upstream: yes
## Cloudflare
forward-addr: 1.1.1.1@853#cloudflare-dns.com
forward-addr: 1.0.0.1@853#cloudflare-dns.com
## Quad9
forward-addr: 9.9.9.9@853#dns.quad9.net
forward-addr: 149.112.112.112@853#dns.quad9.net
## Cloudflare IPv6
forward-addr: 2606:4700:4700::1111@853#cloudflare-dns.com
forward-addr: 2606:4700:4700::1001@853#cloudflare-dns.com
_EOF_
systemctl restart unbound
```
Whether you use Cloudflare or Quad9, IPv4 or IPv6, multiple or just one provider does not matter, so adjust to use the provider(s) you trust most.

If this does not work, then indeed DoH will, as it uses regular HTTPS on port 443. But the Unbound version that is shipped with Debian Buster does not support it yet:
- Debian ships v1.9.0: https://packages.debian.org/buster/unbound
- DoH support was added with v1.12.0: https://nlnetlabs.nl/projects/unbound/download/#unbound-1-12-0

Trunkzeh, 2021-03-26:
This might not be the best way of doing this; but 1.12 (and certainly 1.13.1 which is the latest unbound) can be build from source for RPI. I doubt building from source is what DietPi will end up doing going forward; but you can certainly build it yourself.

coldbootinitheart, 2021-03-28:
Dns-over-tls seems to be working. I apologize as this had nothing to do with dietpi but rather a problem on my end. As such I will be closing this issue.

closed 2021-03-28 by coldbootinitheart

MichaIng, 2021-03-28:
Yes, building from source is of course an alternative, if DoH is required. But I think the cases are rare, so IMO not worth to provide own builds for it from our side.

Not applicable on RPi 32-bit, but on Debian (RPi 64-bit and all other SBCs/hardware) Unbound v1.13.1 can be installed from backports: https://packages.debian.org/buster-backports/unbound
```
apt install -t buster-backports unbound
```