MichaIng/DietPi #4219: Update fails on Odroid C2

StormCh, 2021-03-24:
#### Details:
- Date           | Wed Mar 24 18:52:32 GMT 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_81+0~20210223.34+debian9~1.gbpf52eb0_all.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-apcu/php-apcu_5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57_arm64.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-apcu/php7.3-apcu_5.1.19+4.0.11-7+0~20210222.23+debian9~1.gbp08eb57_arm64.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

StormCh, 2021-03-24:
Solved by

`rm /etc/apt/trusted.gpg.d/php.gpg
apt-key del 95BD4743
wget -O /etc/apt/trusted.gpg.d/php.gpg https://packages.sury.org/php/apt.gpg
apt-key list
`

MichaIng, 2021-03-24:
Many thanks for reporting. I recognised the same on our server, Ondrejs key expired and needs to be updated. I'll implement a pre-patch to have this done automatically on next DietPi update and the present key is older than from 16-Feb-2021.

MichaIng, 2021-03-24:
Done: https://github.com/MichaIng/DietPi/commit/912606ed46c8961dd20f260db281aac0858c95ea
Good idea to no install it via deprecated apt-key anymore: https://github.com/MichaIng/DietPi/commit/1356370cd8c98adcfd66171f76c829f70f1b37b2

MichaIng, 2021-04-08:
Opening the issue and pinning it, with solution added to first post.
Ref: https://github.com/MichaIng/DietPi/issues/4254#issuecomment-815855943

Joulinar, 2021-04-08:
Should we add the cache clean to the first post as well?

MichaIng, 2021-04-08:
It should be not required as the list files definitely changed in the meantime, so lists are updated and the cache itself is always renewed anyway on `apt update`.

closed 2021-04-29 by MichaIng