MichaIng/DietPi #4222: Dietpi install fails when selecting no-webgui option

Trunkzeh, 2021-03-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.0.2
- SBC model | RPIZero-W
- Power supply used | Anker Multi-5port unit
- SDcard used | Sandisk Ultra

#### Additional Information (if applicable)
- Software title | (PiHole)
- Fresh software install
- Can this issue be replicated on a fresh installation of DietPi? Yes.
- Bug report ID | `7b1d1bf6-6730-423f-9ae8-951befe289f0`

#### Steps to reproduce
1. Select PiHole install within dietpi-software; when prompted to install unbound select No / cancel.
2. Part of the PiHole install will ask if you want to enable web-gui; select No.
3. DietPi will install PHP & webserver files nonetheless in this instance; incorrect behaviour?
4. Continue installation process; past if you want to enable / disable public page.
5. Installation will fail on this step:

[FAILED] DietPi-Software | cd /var/www/html/admin

#### Expected behaviour
1. PiHole should install without PHP and webserver components.
- ...

#### Actual behaviour

- See above.

#### Extra details

Looks like the script is looking for the webgui html files in /var/www/html/admin which naturally are not there as pihole has installed without a webgui.

Trunkzeh, 2021-03-26:
I should note; PiHole itself works (as it has already been installed) and I confirmed this via pihole -c after CTRL+C out of the install process.

MichaIng, 2021-03-26:
Many thanks for your report.

Indeed we expect the web UI to be installed, and I guess >99% if users do it anyway. But I agree it would be great to allow a lightweight install without web UI.

It's a larger task since it requires a different way of how dependencies are resolved. Currently the webserver and PHP are installed before the Pi-hole installer is called. When switching it, we need to test if the Pi-hole installed runs through successfully when installing the web interface without having a webserver installed already (and skipping the included Lighttpd install).

Trunkzeh, 2021-03-26:
You are right - normally i would install the webgui; however im trying to run as lightweight as possible on the rpi zero as its not exactly fast; the zero acts as my backup dns instance incase the rpi4 goes down for whatever reason; that naturally has pihole webgui. I'm going to re-image the zero again and then install via the default pihole install script (rather than dietpi) and feedback :)

MichaIng, 2021-03-26:
Jep, I see the use cases.

> re-image the zero again and then install via the default pihole install script

No need to re-image. `pihole -r` > `Reconfigure` and uninstalling webserver + PHP via `dietpi-software` would do it as well 😉.

Trunkzeh, 2021-03-26:
I thought of doing that; however the webserver aspects dont show up on the uninstall menu of dietpi-software in the above scenario. I'm not sure because the script hasnt registered the installation when it was initially running the original pihole job if that makes sense? Only having a logical guess.. I havent the faintest clue on the mechanics of dietpi's optimized software scripts.

MichaIng, 2021-03-26:
Ah yes that makes sense when the install didn't went through. We need add a pending state for installs which allow another install as well as an uninstall 💡.

Trunkzeh, 2021-04-03:
Personally; this ticket can be closed. I was able to install pihole via the developer's standard installation procedure (their install script) and have it operate without a webgui with no extra steps. 99% of everybody installing pihole would utilize a webgui and dietpi works out of the box :-) 

TLDR; pihole works great on dietpi. If you want to run it without a webgui; install using the official pihole installation script rather than from dietpi-software.

MichaIng, 2021-04-03:
Let me keep it open, as I find it reasonable to allow an install without web UI via dietpi-software as well.

thanhminhmr, 2022-05-26:
Any update on this? I just try to install pi-hole thru dietpi-software, and it force me to pre-install a web server and PHP.
Of course the pi-hole install script still allow to skip web interface, so I have to uninstall the web server and PHP manually.

Joulinar, 2022-05-26:
currently there is no plan to offer such an install option without web interface.

thanhminhmr, 2022-05-26:
I don't think there is a need to support that use-case. Just show the user that: dietpi-software doesn't support no web interface install, and if the user want that then please use the install script directly from pi-hole. Something to inform user would be nice.

MichaIng, 2025-09-16:
This is obsolete since Pi-hole v6. Closing.

closed 2025-09-16 by MichaIng