MichaIng/DietPi #4224: Portainer Error - can’t remove Portainer

redschumi, 2021-03-26:
### Creating a bug report/issueMode: Uninstalling Portainer: Simplifies container management in Docker (standalone host)

[  OK  ] DietPi-Software | systemctl restart docker
[FAILED] DietPi-Software | docker rm  --force
[FAILED] DietPi-Software | docker rm  --force
[FAILED] DietPi-Software | docker rm  --force

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Mar 26 17:44:30 GMT 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian Buster
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | Linux DietPi 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | docker rm  --force
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Container name cannot be empty
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

redschumi, 2021-03-26:
Sorry, forgot to fill our everething:

What happened?:

Installed docker incl. Portainer a few days ago.
Today i was not able to reach my bitwarden container nor the portainer webif.
checked everything, rebooted but no success.
Cli „docker ps“ doesn’t show any container.
decided first to uninstall portainer.
and as you can see...no luck

MichaIng, 2021-03-26:
Many thanks for your report.

It seems you removed the container manually already, right? As a workaround to finish the uninstall, choose "Change command" from the error prompt and replace it with `:` or `true`.

@Joulinar 
On install we check at least for the data volume `/mnt/dietpi_userdata/docker-data/volumes/portainer_data`, but on uninstall we only check for `/mnt/dietpi_userdata/docker-data`. But also the data volume is no guarantee that the image or the container exists.
Best is to check for everything separately, I think:
```
local container=$(docker ps -a | mawk '/portainer\/portainer(-ce)?( |$)/{print $1;exit}')
[[ $container ]] && G_EXEC docker rm -f "$container"
local image=$(docker images -a | mawk '/portainer\/portainer(-ce)?( |$)/{print $3;exit}')
[[ $image ]] && G_EXEC docker rmi "$image"
docker volume ls -q | grep -q '^portainer_data$' && G_EXEC docker volume rm portainer_data
```
And similarly on install:
```
local container=$(docker ps -a | mawk '/portainer\/portainer(-ce)?( |$)/{print $1;exit}')
[[ $container ]] && G_EXEC docker rm -f "$container"
local image=$(docker images -a | mawk '/portainer\/portainer(-ce)?( |$)/{print $3;exit}')
[[ $image ]] && G_EXEC docker rmi "$image"
docker volume ls -q | grep -q '^portainer_data$' || G_EXEC docker volume create portainer_data
```

redschumi, 2021-03-26:
Yes, that’s right. they disappeared. and i don’t know why

MichaIng, 2021-03-26:
Okay, thanks for adding the additional info. So while what I wrote above is generally good prevent those steps failing when things have been removed or changed manually. In your case we need to find out why containers disappeared. Portainer should not have something to do with it, as it's a frontend for Docker only and a container itself.

Since the Docker data are stored to `/mnt/dietpi_userdata/docker-data` which can be easily moved to an external drive: Did you move the DietPi userdata to an external drive, and might the connection/mount of that drive be list? E.g. to check:
```
ls -l /mnt/dietpi_userdata{,/}
df
```

redschumi, 2021-03-26:
─────────────────────────────────────────────────────(2021-01-30) x86_64
 DietPi v7.0.2 : 17:41 - Fri 03/26/21
 ───────────────────────────────────────────────────── are free software;
 - Device model : Native PC (x86_64)h program are des - CPU temp : 54'C : 129'F (Running warm, but safe)
 - LAN IP : 10.1.1.10 (eth0)re/doc/*/copyright.
 - MOTD : Enjoy our new DietPi v7.0 <3
 ───────────────────────────────────────────────────── the extent
permitted by applicable law.
 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Debian Buster)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
root@DietPi:~# dietpi.software
-bash: dietpi.software: command not found
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Uninstalling Portainer: Simplifies container management in Docker (standalone host)

[  OK  ] DietPi-Software | systemctl restart docker
[FAILED] DietPi-Software | docker rm  --force
[FAILED] DietPi-Software | docker rm  --force
[FAILED] DietPi-Software | docker rm  --force

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Mar 26 17:44:30 GMT 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian Buster
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | Linux DietPi 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | docker rm  --force
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
[... 143 lines cut ...]
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 downloads

/mnt/dietpi_userdata/:
total 20
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Music
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Pictures
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Video
drwx--x--x 13 root   root   4096 Mar 26 18:44 docker-data
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 downloads
root@DietPi:~# df
Filesystem         1K-blocks      Used Available Use% Mounted on
udev                 3988600         0   3988600   0% /dev
tmpfs                 801128     11688    789440   2% /run
/dev/sda2          230881040   2639632 216511780   2% /
tmpfs                4005632      6816   3998816   1% /dev/shm
tmpfs                   5120         4      5116   1% /run/lock
tmpfs                4005632         0   4005632   0% /sys/fs/cgroup
tmpfs                  51200       464     50736   1% /var/log
tmpfs                4004864         4   4004860   1% /tmp
/dev/sda1              94759     14406     80353  16% /boot/efi
tmpfs                 801124         0    801124   0% /run/user/998
tmpfs                 801124         0    801124   0% /run/user/0
/dev/sdb1          229657424   3095908 226545132   2% /mnt/256GB
//10.1.1.51/backup 960380628 273424364 686956264  29% /mnt/omv5_backup
root@DietPi:~#

redschumi, 2021-03-26:
Update:

root@DietPi:~# cd /mnt/dietpi_userdata/docker-data/
root@DietPi:/mnt/dietpi_userdata/docker-data# ls
buildkit  containers  image  network  overlay2  plugins  runtimes  swarm  tmp  trust  volumes
root@DietPi:/mnt/dietpi_userdata/docker-data# cd containers/
root@DietPi:/mnt/dietpi_userdata/docker-data/containers# ls
root@DietPi:/mnt/dietpi_userdata/docker-data/containers# cd ../volumes
root@DietPi:/mnt/dietpi_userdata/docker-data/volumes# ls
backingFsBlockDev  metadata.db
root@DietPi:/mnt/dietpi_userdata/docker-data/volumes#

redschumi, 2021-03-26:
-bash: cd: /mnt/256GB/Die: No such file or directory
root@DietPi:~# cd /mnt/256GB/dietpi_userdata/docker-data/containers/
root@DietPi:/mnt/256GB/dietpi_userdata/docker-data/containers# ls
0a8fb39a6c55783adc65c827c52f4ed56a718079be3f5e8d0aface0bf3bf3256  8c8dec834be5c5b0da59c818480ff7b57fb85f6fd9c9f79ad8a9bcd99c04b823
root@DietPi:/mnt/256GB/dietpi_userdata/docker-data/containers# cd ../volumes/
root@DietPi:/mnt/256GB/dietpi_userdata/docker-data/volumes# ls
backingFsBlockDev  metadata.db  portainer_data
root@DietPi:/mnt/256GB/dietpi_userdata/docker-data/volumes#

redschumi, 2021-03-26:
Update: fixed it!

checked drive_manager: i don’t know why but Userdata was back on cd /mnt/dietpi_userdata/.
Changed it again via drive_manager and everything worked fine.
will this investigating the near few days and will inform you if it changes again back to standard.
thank you so much for your quick reply.

MichaIng, 2021-03-26:
Okay, so `/mnt/256GB` is the actual userdata location which contains the `portainer_data` and hence the container as well. If `/mnt/dietpi_userdata` was a symlink (what is should be after moving dietpi_userdata to an external drive), the first ls output would have shown the link + target instead of the content, so I guess for some reason the symlink was lost.

Please assure that `/mnt/dietpi_userdata/` before removing it, or just move it to a backup location for now, then re-create the symlink...

... I'm always too slow in writing 😄.

redschumi, 2021-03-26:
symlink has to be created via drive_manager, right? 
I choosed the /mint/256GB/... and enabled „move userdata...). also right?

MichaIng, 2021-03-26:
Great. I'll mark this as closed, but please reopen if the symlink disappears again, so we'd need to check why.

closed 2021-03-26 by MichaIng