MichaIng/DietPi #4225: issue installing pihole / unbound

pierrepgithub, 2021-03-27:
### Creating a bug report/issue

#### Required Information
- DietPi version
      `root@DietPi:~# cat /boot/dietpi/.version
      G_DIETPI_VERSION_CORE=7
      G_DIETPI_VERSION_SUB=0
      G_DIETPI_VERSION_RC=2
      G_GITBRANCH='master'
      G_GITOWNER='MichaIng'`

- Distro version
      `root@DietPi:~# cat /etc/debian_version
      10.8`
- Kernel version
      `root@DietPi:~# uname -a
      Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux`
- SBC model
      `root@DietPi:~# echo $G_HW_MODEL_NAME
      RPi 4 Model B (armv7l)`
- Power supply used
      CanaKit
- SDcard used
      Samsung 128GB EVO+ Class 10

#### Additional Information (if applicable)
- Software title
      1) PiHole 2) Unbound 3) RPi-Monitor 4) Fail2Ban
- Was the software title installed freshly or updated/migrated?
      Trying a brand new image / install
- Can this issue be replicated on a fresh installation of DietPi?
      - Yes

- Bug report ID
      `root@DietPi:~# echo $G_HW_UUID
      95cc46e4-cdaa-43da-a078-64f2d33d22fb`
      I actually generated a few bug reports, but didn't capture the ID, didn't know how. The above ID is the latest one after installing dietpi and then running dietpi-software after dietpi installed

#### Steps to reproduce
### Installing additional software along with dietpi
1. Fresh dietpi install
2. Install the above mentioned software during the installation

### After installing fresh dietpi
1. Install dietpi with no additional software
2. Attempt to install the above mentioned software with dietpi-software
3. Still get timeout issue
[issue_installing_pihole.tar.gz](https://github.com/MichaIng/DietPi/files/6214806/issue_installing_pihole.tar.gz)
[issue_installing_pihole_03262021.tar.gz](https://github.com/MichaIng/DietPi/files/6214807/issue_installing_pihole_03262021.tar.gz)

#### Expected behaviour
1. Successful install?

#### Actual behaviour
1. Looks like there's a timeout during the pihole setup

#### Extra details

- ...

MichaIng, 2021-03-27:
Many thanks for your report.

Yes indeed the download of the Pi-hole installer timed out. Also the very first request to `https://install.pi-hole.net` took very long, `https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh` timed out.

Can you please run and show the output of the following:
```
time curl -sSf https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh
```

It was an IPv4 connection already, so disabling IPv6, a typical attempt to solve slow connections, doesn't help here.

pierrepgithub, 2021-03-27:
I'll get you that output as soon as I can. Watching my daughter at the moment.

Wanted to so mention the same happens when I try to install the software separately instead of all together. Fail2Ban installed ok, but issue happened with rpi-monitor and pihole

MichaIng, 2021-03-27:
Such long connection times should be investigated, but as a workaround, increasing the timeout based on the above result would do:
`dietpi-config` > `Network Options: Misc` > `G_CHECK_URL Timeout`

pierrepgithub, 2021-03-27:
`time curl -sSf https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh`
`real    0m5.298s`
`user    0m0.040s`
`sys     0m0.069s`
`root@DietPi:~#`

If you need the entire output, let me know, but I'm guessing you're just looking for the times here?

MichaIng, 2021-03-27:
Okay five seconds is very long. To compare:
```
# time curl -sSf https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh > /dev/null
real    0m0.197s
user    0m0.050s
sys     0m0.017s
```
But was the script contend printed to console in your case, actually? Otherwise probably DNS resolution timed out, since the default timeout for it is 5 seconds.
The better debugging command is the following, which shows connection meta info but not the file content:
```
time curl -I https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh
```

Are you connected via cable and broadband connection, or WiFi and/or mobile internet? Or the other way round, 5 seconds for an internet connection request is not normal from that network, is it?

pierrepgithub, 2021-03-27:
`# time curl -I https://raw.githubusercontent.com/pi-hole/pi-hole/master/automated%20install/basic-install.sh
HTTP/2 200
cache-control: max-age=300
content-security-policy: default-src 'none'; style-src 'unsafe-inline'; sandbox
content-type: text/plain; charset=utf-8
etag: "8f600c8dd5bf943d55547836e3dd926c9e2e1677bc5699cab159c7db5add740f"
strict-transport-security: max-age=31536000
x-content-type-options: nosniff
x-frame-options: deny
x-xss-protection: 1; mode=block
x-github-request-id: F8FA:682F:2B5E96:384FCB:605F2FE9
accept-ranges: bytes
date: Sat, 27 Mar 2021 16:16:32 GMT
via: 1.1 varnish
x-served-by: cache-mdw17358-MDW
x-cache: HIT
x-cache-hits: 1
x-timer: S1616861793.606771,VS0,VE1
vary: Authorization,Accept-Encoding
access-control-allow-origin: *
x-fastly-request-id: 903abf662ad2b2062b2918f6b678c5e63deafa92
expires: Sat, 27 Mar 2021 16:21:32 GMT
source-age: 243
content-length: 123759

real    0m5.153s
user    0m0.033s
sys     0m0.032s`

I'm hard wired with an ethernet cable, directly into the router.
No I guess 5 seconds would not be normal.

> But was the script contend printed to console in your case, actually?

 - I did not understand this question

pierrepgithub, 2021-03-27:
Think I found the problem, had a weird local IP address in my resolv.conf.

The above curl commands run in less than 0.1s now

Trying a fresh install again

pierrepgithub, 2021-03-27:
Just my luck, it's not booting now :/

pierrepgithub, 2021-03-27:
Got the pi4 to boot now finally. This time I tried flashing with Rufus instead of etcher.

Previously I was using etcher but I noticed now for whatever reason it's creating two mount points on the sd card, wonder if that had anything to do with the pi not being able to boot?

Not sure I'm comfortable using Rufus either. The software seems sketchy. A lot of stuff was greyd out for whatever reason. And I had to do some workarounds to get it to have the proper settings according to dietpi documents.

pierrepgithub, 2021-03-27:
ok looks like i got it going...

```
root@DietPi:~# dig pi-hole.net @127.0.0.1 -p 5335

; <<>> DiG 9.11.5-P4-5.1+deb10u3-Raspbian <<>> pi-hole.net @127.0.0.1 -p 5335
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 20899
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;pi-hole.net.                   IN      A

;; ANSWER SECTION:
pi-hole.net.            3600    IN      A       192.124.249.118

;; Query time: 150 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1)
;; WHEN: Sat Mar 27 19:17:15 GMT 2021
;; MSG SIZE  rcvd: 56

root@DietPi:~# dig sigfail.verteiltesysteme.net @127.0.0.1 -p 5335

; <<>> DiG 9.11.5-P4-5.1+deb10u3-Raspbian <<>> sigfail.verteiltesysteme.net @127.0.0.1 -p 5335
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 27547
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;sigfail.verteiltesysteme.net.  IN      A

;; Query time: 3971 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1)
;; WHEN: Sat Mar 27 19:17:24 GMT 2021
;; MSG SIZE  rcvd: 57

root@DietPi:~# dig sigok.verteiltesysteme.net @127.0.0.1 -p 5335

; <<>> DiG 9.11.5-P4-5.1+deb10u3-Raspbian <<>> sigok.verteiltesysteme.net @127.0.0.1 -p 5335
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 51136
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;sigok.verteiltesysteme.net.    IN      A

;; ANSWER SECTION:
sigok.verteiltesysteme.net. 300 IN      A       134.91.78.139

;; Query time: 227 msec
;; SERVER: 127.0.0.1#5335(127.0.0.1)
;; WHEN: Sat Mar 27 19:17:29 GMT 2021
;; MSG SIZE  rcvd: 71

root@DietPi:~#
```

my resolve.conf still has my router ip address, is that ok or should something be updated there?

```
root@DietPi:~# cat /etc/resolv.conf
nameserver 192.168.50.1
root@DietPi:~#
```

The official pihole doc for unbound has some steps for resolvconf, not sure if I should follow those ?

https://docs.pi-hole.net/guides/dns/unbound/

I'd like to enable DOT as well... In the dietpi document for unbound (https://dietpi.com/docs/software/dns_servers/#unbound)

It states the following:

```
If required, you can activate DoT. Simply copy/execute following section:

cat << '_EOF_' > /etc/unbound/unbound.conf.d/dietpi-dot.conf
# Adding DNS-over-TLS support
server:
tls-cert-bundle: /etc/ssl/certs/ca-certificates.crt
forward-zone:
name: "."
forward-tls-upstream: yes
## Cloudflare
forward-addr: 1.1.1.1@853#cloudflare-dns.com
forward-addr: 1.0.0.1@853#cloudflare-dns.com
## Quad9
forward-addr: 9.9.9.9@853#dns.quad9.net
forward-addr: 149.112.112.112@853#dns.quad9.net
_EOF_
```

Does this just edit the dietpi.conf file? Does it append it to the file that's already there?

ravenclaw900, 2021-03-27:
That would create a new configuration file with the DOT settings.

MichaIng, 2021-03-27:
Jep, a separate file so that it's easy to revert.

In Rufus many settings are greyed out since the image is not a single partition but a full drive image with partition table, so it's intended that there is nothing to customise 😉. Since the RPi has two partitions, one for booting (bootloader, kernel, device tree) and the root filesystem, you see the two mount points, so that is all correct.

About the Unbound DoT config: `forward-addr` is theoretically only required once, while the other ones are fallbacks. Also the ones are just examples. Checkout the full list of public DNS providers and which one supports DoT: https://en.wikipedia.org/wiki/Public_recursive_name_server

pierrepgithub, 2021-03-27:
Then I'm not sure why it didn't want to boot haha.

Should I try to get it to work with etcher? I tried several times, it wouldn't boot.

Is there a way to validate the install and config? I just feel uneasy about using Rufus I guess. Would have felt more comfortable if etcher was used.

What about the resolve.conf or the optional methods I mentioned in the pihole doc for unbound? What should my resolv.conf be when using pihole and unbound? Currently it just points to my default gateway

ravenclaw900, 2021-03-27:
If it makes you feel any better, Rufus has 0 detections on VirusTotal. https://www.virustotal.com/gui/file/ec3136b053bd1559ad7ec1ea104113898093b886bf519e6117b138ef2e691cbb/detection
If your `resolv.conf` file isn't changed, you don't need to do any of the steps, as you usually don't want it to use unbound.

pierrepgithub, 2021-03-27:
Not concerned with viruses. I've used Rufus to image linux desktops before and haven't had an issue.

Just feeling iffy about making the dietpi image using Rufus using all those workarounds to get GPT and UEFI to come up. 

Some weird voice on my head is concerned all the image being created correctly and all necessary files are there and such. I'm weird like that.

Definitely will be dropping a donation to you guys. You're all awesome.

MichaIng, 2021-03-27:
I always use Rufus to flash (all) images, never had problems with it. As said, it automatically greys out options that do not apply for the kind of image file, so not much to worry. I can't remember currently if this is true for GPT/UEFI images as well or if the partition table type (GPT/MBR) can be selected even when it's a full drive image? However, this does not apply for RPi and most other SBC images anyway, which are all simple MBR. Nothing can be missing because all required data is written bit-by-bit from the image file to the drive, like `dd` would do, not more, not less.

It's not the first time I read of issues when using Belana Etcher 🤔. It's been a while since I used it last time (successfully). Will try it for the RPi images I'm flashing tomorrow for testing reasons, let's see if I can replicate it. On Windows I don't see any argument for Etcher, being a 125 MiB giant compared to 1.1 MiB Rufus that even has more features 😄. But Etcher is nice for it's very clean interface (no additional options = no worries that something was selected wrong) and that it works on macOS and Linux as well.

pierrepgithub, 2021-03-28:
Yea I was following the dietpi doc which says to select GPT/UEFI but when you select the image it auto picks MBR and it's greyed out. So I did a series of selects/clicks to get it to be GPT/UEFI and then picked the image. But I guess the default MBR would have worked too according to you. I was just trying to follow the doc on dietpi lol.

Yea I've always used Rufus before as well. I only got etcher because that's what basically all documents say to use when I first got into the raspberry pi stuff.

Anyways I think I got it all setup now. 

Do you know if there's a testing page to check/validate the DOT/unbound?

I know of 1.1.1.1/help but that's for cloudflare.

pierrepgithub, 2021-03-28:
@MichaIng 

I only have a couple more questions and then I'll stop bugging you for now :p 

I'm getting this after I enable DOT and restart unbound via that new .conf file
```
root@DietPi:~# systemctl restart unbound
Warning: The unit file, source configuration file or drop-ins of unbound.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```
Can I ignore this or should I run the daemon-reload command it's asking?

2) Is it a good idea / do you support cronjobs for like weekly updates?
- Was thinking about setting up some weekly updates to check for distro / package / pihole / unbound hints
- If so, got any examples?

MichaIng, 2021-03-28:
> Can I ignore this or should I run the daemon-reload command it's asking?

Strange as we didn't touch the unit file. However, no harm to run `systemctl daemon-reload` to satisfy systemd.

> Is it a good idea / do you support cronjobs for like weekly updates?

You can enable daily APT upgrades already. Weekly APT upgrades are a good idea as well, having the benefit that one can move the weekly cron job to a day/time where physical access to the device is guaranteed, in case something fails. Actually we can add a general weekly dietpi cron job, that can do (optional) weekly DietPi-Sync as well, which is currently only possible on a daily basis. Reasonable due to weekly work/office/home time schedules.

- That would do the weekly APT upgrades, preserving the log the same way our (optional) daily APT upgrade cron does:
```
# Assure that no daily APT upgrades are done (the default is check only but no upgrade == 1)
sed -i 's/CONFIG_CHECK_APT_UPDATES=2/CONFIG_CHECK_APT_UPDATES=1/' /boot/dietpi.txt
cat << 'EOF' > /etc/cron.weekly/apt_upgrade
#!/bin/dash
{ apt-get -q update; apt-get -qq upgrade; } > /var/tmp/dietpi/logs/dietpi-upgrade_apt.log 2>&1
# As some APT upgrades stop their services:
/boot/dietpi/dietpi-services start
_EOF_
chmod +x /etc/cron.weekly/apt_upgrades
```
- Unbound hints are checked monthly already, which is more than sufficient.
- Pi-hole updates should never be done non-interactive (via cron job). Of course we could manually check the upstream version and compare with the local version, but `pihole -up` must not be called, which is also what every Pi-hole developer will recommend.

closed 2021-03-30 by pierrepgithub