MichaIng/DietPi #4226: Password at boot

jorgejones7711, 2021-03-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Picture a typical Fedora desktop install. When you turn the computer on, it stops at a simple password screen. It will not continue until you put in the password. Once you do, it continues booting. This way the OS is safe from tampering, as it's encrypted.

#### Describe the solution you'd like:
Can DietPi be set up to do the same thing? I assume debian can do that as well.

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-03-27:
Many thanks for your request.

Let me try to understand better: Do you want to have the system drive encrypted? Since there is always a user+password/key login required to access the device, regardless how you connect, if a desktop is installed or not, unless you explicitly enable autologin.

If so, there are various ways of encrypting a drive, i.e. not (only) a login at boot, but so that as well when plugging the drive to another system, it cannot be accessed offline. Some filesystems like Btrfs have a native encryption feature, ecryptfs, dm-crypt/LUKS etc. But having the whole root device encrypted means that it needs to be decrypted before even it's content is accessible, i.e. likely a separate boot partition is required which a special bootloader/initramfs that loads the required modules and a decryption password prompt. We'd have to read into ways how to achieve this best, e.g. check how Fedora is doing it (if indeed encryption is what you mean it does). Generally of course it can be done on any Linux distro, when installing (or compiling) the right packages and in case setting up the right partitioning. If you find a guide on how to do what on Debian, it'll work on DietPi as well.

psi5asp, 2021-07-06:
This is a great idea. Security for data on hard drives is very important, especially if attempted access via another system is done. Without a key/password/passphrase, access should be prohibited.

Here is a guide for RHEL using cryptsetup using LUKS. -  which can be adapted for Dietpi. Any drive/partition (including the OS, etc. as mentioned) should be encryptable - in the Dietpi world this should be done via DietPi-Drive_Manager while mounting or allow drive conversion for existing mounts without the loss of data.

singles, 2022-10-15:
I would love that feature to be implemented too! :)

>  We'd have to read into ways how to achieve this best,

Maybe this will help - I followed [this approach](https://hamy.io/post/0009/how-to-install-luks-encrypted-ubuntu-18.04.x-server-and-enable-remote-unlocking/) for my current non-RPi Debian based machine. 

Basically, after boot I SSH to it provide encryption password. After that, connection is closed and system boots normally from encrypted part of the drive where whole my private data is stored.  

Unfortunately, my Linux knowledge isn't good enough to implement this as PR, but maybe mentioned article will help :)

Joulinar, 2022-10-16:
The general problem is to setup encrypted disc/partition on an already running system. Not sure if we are able to implement this.

singles, 2022-10-16:
@Joulinar I understand, looks like it's more complicated than I thought. 

I did some digging and found: 

* https://rr-developer.github.io/LUKS-on-Raspberry-Pi/
* https://codeberg.org/keks24/raspberry-pi-luks#user-content-encrypting-the-root-partition-manually (this one includes unlocking partition via SSH too)

Basically approach is to use USB drive to create copy of existing system, create new encrypted partition and copy previously installed system onto encrypted partition with adding optional "unlock via SSH" option.  

I wonder if some there's an option to create wizard runnable from `dietpi-launcher` like `Encrypt installation`. Of course this wizard would mention that one would need empty USB drive and the process will take some time. 

Not sure that's even possible, but might above will help somehow :)