MichaIng/DietPi #4227: Image | Orange Pi R1 Plus

pmjohann, 2021-03-27:
### Creating an image request

#### Formal device information
- Device name | Orange Pi R1 Plus
- Official product URL | [http://www.orangepi.org/Orange%20Pi%20R1%20Plus](http://www.orangepi.org/Orange%20Pi%20R1%20Plus)
- GitHub resource URL (if available) | [https://github.com/orangepi-xunlong/orangepi-build](https://github.com/orangepi-xunlong/orangepi-build)
- Image download URL | [https://dietpi.com/downloads/images/DietPi_NanoPiR2S-ARMv8-Buster.7z](https://dietpi.com/downloads/images/DietPi_NanoPiR2S-ARMv8-Buster.7z)

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Yes

#### If not, are there install instructions for Debian available?

- The board actually works *mostly* with the DietPi image for the NanoPiR2S. There are shutdown / reboot issues etc. **The hardware is mostly the same.**

I know that Orange Pi builds were discontinued a while ago, the reason I am proposing this board is because it is like 99% compatible with the Nano Pi R2S which is officially supported by DietPi, so it should not be too much hassle to make this happen. Looking forward to what you think.

Thank you!

MichaIng, 2021-03-27:
Many thanks for your request.

As for now, I'm personally not going to create Orange Pi images, as long as we've not switched to completely automated builds based on at best the official Debian kernel. But you'll be able to create it pretty easily:
- Download and flash the Armbian Buster image: https://www.armbian.com/orange-pi-r1/
- Boot and login with "root:1234".
- `bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh)"`

pmjohann, 2021-03-27:
Thanks for the prompt response MichaIng,

This board is the brand new R1 **Plus**, not the standard R1, so no Armbian avail for now.
If you could point me in the right direction how to build a DietPi image for NanoPi R2S but with a different DTB, I could do it myself and contribute it back if you wish. They are really almost the same hardware, and as I said the NanoPi R2S mostly works! It's just not surviving reboots (have to power down and on again) which is not acceptable for my usecase.

Thanks once more!

MichaIng, 2021-03-27:
Ah sorry, I missed the "Plus" 😅. So indeed most likely a different DTB should be sufficient to run it.

Orange Pi itself ships a quite current Debian image: https://drive.google.com/drive/folders/1vpjLivmcpFRwkvq696Lv1Wynq-e7F6ci
That one should work, when you choose "Generic Device" from within our PREP script.

pmjohann, 2021-03-27:
Awesome, I couldn't find the official Debian image to start from... sigh.
I'll try it and report back! Thanks!

pmjohann, 2021-03-27:
Aaand it worked, thank you @MichaIng !

Anyone going down this route please note that the default user+pass combos for the Orange Pi R1 Plus official Debian do not work (NONE OF THEM!, see below). You'll have to mount the partition and edit /etc/passwd directly to remove password for the root user.

This is done by removing the "x" from this line in /etc/passwd :

```root:x:0:0:root [...]```

to look like this:

```root::0:0:root [...]```

I tried the following passwords for both the "root" and "orangepi" users:

- '' (empty)
- 1234
- password
- orangepi
- root
- none
- default

MichaIng, 2021-03-28:
See here: http://www.orangepi.org/downloadresources/orangepiR1Plus/2020-12-22/orangepiR13f827bdc79a1660c8a85870d8f.html
> user(root), password(orangepi)

Sorry, I should have linked that page instead of the GDrive download page 😅.

pmjohann, 2021-03-28:
Nope, tried that combo (root : orangepi) as well, did not work.
Only removing password for the root user as described above did.

Anyway, board is working fine running DietPi! Thanks once more @MichaIng !

MichaIng, 2021-03-28:
Okay, great and good to know about/how to enable login with that image.

MichaIng, 2021-05-07:
I'm gonna mark Orange Pi related image requests as closed for now. Referring to our preparation script guide: https://dietpi.com/docs/hardware/#make-your-own-distribution

Orange Pi is on my first priority list when image creation has been automated enough, for for now I personally cannot effort adding a new manufacturer to my creation schedule.

closed 2021-05-07 by MichaIng

mrbluecoat, 2022-08-06:
FYI, the latest image https://drive.google.com/drive/folders/1b55uN1ElzAgvAJQSW772lwkKEwqunawb works with root password `orangepi`

GrauBlitz, 2023-06-25:
> 

This image is for the LTS version with a different native Ethernet Chip, so this will most likely not work with the non-LTS Version or you will only get the USB Ethernet working.