MichaIng/DietPi #4228: Error Syncthing install - Permission denied

redschumi, 2021-03-27:
### Creating a bug report/issue
sorry for disturbing, but i got another error.
Wanted install Syncthing, but no luck. Installer reports a permission denied error. But directory was created. it is empty.

______________________________________

Mode: Configuring Syncthing: backup and sync server with web interface

20:01:47 WARNING: Failed to generate config and keys: mkdir /mnt/dietpi_userdata/syncthing: permission denied
sed: can't read /mnt/dietpi_userdata/syncthing/config.xml: No such file or directory
sed: can't read /mnt/dietpi_userdata/syncthing/config.xml: No such file or directory
sed: can't read /mnt/dietpi_userdata/syncthing/config.xml: No such file or directory
[ SUB1 ] DietPi-Services > dietpi_controlled 
[  OK  ] DietPi-Services | dietpi_controlled : nmbd
[  OK  ] DietPi-Services | dietpi_controlled : smbd
[  OK  ] DietPi-Services | dietpi_controlled : php7.3-fpm
[  OK  ] DietPi-Services | dietpi_controlled : lighttpd
[  OK  ] DietPi-Services | dietpi_controlled : syncthing
[  OK  ] DietPi-Services | dietpi_controlled : docker
[  OK  ] DietPi-Services | dietpi_controlled : cron
[ SUB1 ] DietPi-Set_hardware > headless (disable)
[ INFO ] DietPi-Set_hardware | Input name (headless) is not supported by Native PC (x86_64). Nothing has been applied.
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_HEADLESS=0
[  OK  ] headless disable | Completed
[  OK  ] DietPi-Survey | Purging survey data

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[ SUB1 ] DietPi-Services > restart 
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : php7.3-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : syncthing
[  OK  ] DietPi-Services | restart : docker
[  OK  ] DietPi-Services | restart : cron
root@DietPi:~# dietpi-config

MichaIng, 2021-03-28:
Many thanks for your report.

Can you please run and paste the output of the following command:
```
ls -al /mnt/dietpi_userdata/
```
That should be actually owned by the `dietpi` user, so that it has permissions to create directories there, which is what Syncthing tries here.

redschumi, 2021-03-28:
Thanks for you patience. Here is the output:

root@DietPi:~# ls -al /mnt/dietpi_userdata/
total 36
drwx--x--x  9 root   root   4096 Mar 27 19:32 .
drwxr-xr-x  6 root   root   4096 Mar 23 22:02 ..
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Music
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Pictures
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 Video
drwxr-xr-x  3 root   root   4096 Mar 28 09:21 bitwarden
drwx--x--x 13 root   root   4096 Mar 27 20:11 docker-data
drwxrwxr-x  2 dietpi dietpi 4096 Mar 23 10:36 downloads
drwxr-xr-x  2 dietpi dietpi 4096 Mar 27 19:32 syncthing_data
root@DietPi:~#

maybe you remind, the my userdata was moved to an external drive. but the symlinks are working.

MichaIng, 2021-03-28:
Indeed we did not re-apply the dietpi:dietpi ownership after moving userdata. Just fixed with: https://github.com/MichaIng/DietPi/commit/71ad4f49893831f5f73e44c8e1f7aa6dc4fc37cb
Changelog: https://github.com/MichaIng/DietPi/commit/3ff09d406bb216e2dc9f06277efdaa4bf07c302d
And also dietpi-software will now assure that dietpi owns the directory before that config creation command is called: https://github.com/MichaIng/DietPi/commit/ddeb95a5a4aac39d044a376a58400ae3e4e27f67

redschumi, 2021-03-28:
WTF? 😲 You’re as quick as hell!👏 👏 👏

Means for me? Just make an update? Or remove/install Syncthing

ravenclaw900, 2021-03-28:
`chown dietpi:dietpi /mnt/256GB/dietpi_userdata/` should fix it.

closed 2021-03-28 by redschumi