MichaIng/DietPi #4230: Error install Python

redschumi, 2021-03-29:
### Creating a bug report/issue
Wanted to install Python, but the following error occurred 

#### Details:
- Date           | Mon Mar 29 22:32:09 CEST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux nordvpnpi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
E: Could not get lock /var/lib/apt/lists/lock - open (11: Resource temporarily unavailable)
E: Unable to lock directory /var/lib/apt/lists/
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

ravenclaw900, 2021-03-29:
That usually means that something else was using APT at the time you ran that command.  If everything's working now, can this be closed?

MichaIng, 2021-03-29:
Probably the daily APT update check. I was already thinking that we need to detect it, print a meaningful message and allow some minutes for it to finish.

@redschumi 
What time does your daily cron job run?
```
grep daily /etc/crontab
```
`22:32:09` you timestamp above doesn't look like a usual time though.

redschumi, 2021-03-30:
25 1 * * * root test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.daily; }

MichaIng, 2021-03-31:
Okay, default `01:25`. Not sure what blocked APT then. Use e.g. `htop` to see running processes and identify what has called APT, in case you run into the same issue again.

redschumi, 2021-03-31:
> Okay, default `01:25`. Not sure what blocked APT then. Use e.g. `htop` to see running processes and identify what has called APT, in case you run into the same issue again.

Ok, thanks, will do that.

closed 2021-03-31 by redschumi