MichaIng/DietPi #4231: No home directory of `dietpi` user on RPi 64bit image

grzesuav, 2021-03-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v7.0.2`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Official Rpi4 power supply, 5v 3A
- SDcard used | Class 10 good ram

#### Steps to reproduce

1. login as root, then `su dietpi`
2. `cd ~`
3. The message is `bash: cd: /home/dietpi:` no such directory

#### Expected behaviour

- should move to this directory

#### Actual behaviour

- directory not exists

ravenclaw900, 2021-03-31:
Linking 2 other posts with the same problem:
https://dietpi.com/phpbb/viewtopic.php?f=11&t=8862
https://dietpi.com/phpbb/viewtopic.php?f=9&t=8863

MichaIng, 2021-03-31:
I see, I changed the PREP script to have a more complete cleanup and default system config file restoring (e.g. Armbian and others ship with some custom files, while we want the Debian package defaults). The "dietpi" user needs to be created afterwards, so that it's home dir + skeleton shell configs remain present. I'll repack the recently updated images.

MichaIng, 2021-03-31:
Most images have been repacked, the last affected ones will be done tomorrow.

idealwebsolutions, 2021-04-21:
Any idea when this will be packed in?

grzesuav, 2021-04-28:
Thanks @MichaIng , closing issue then

closed 2021-04-28 by grzesuav