MichaIng/DietPi #4232: DietPi-Software | IPFS Node

ravenclaw900, 2021-03-30:
### Creating a software request

#### Formal software information
- Software title | 
IPFS Node
- Short description | 
Add a node to the IPFS network
- Official URL (if available) | 
https://ipfs.io
- GitHub URL (if available) | 
https://github.com/ipfs/go-ipfs
- (Official) install documentation | 
- https://docs.ipfs.io/install/command-line

#### Are there similar/alternative software titles available with DietPi-Software?

- No

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

-  Some initialization and configuration required, could be done automatically.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Download tarball
2. Run install script
3. Initialize IPFS repository.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

MichaIng, 2021-04-11:
We should think about enabling some web UI protection or limit access to local network, if there is a way. Otherwise it should be made clear in the docs that it is unprotected by default and hence in case of NAT that port should not be forwarded or otherwise blocked by firewall or via HTTP authentication-protected proxy.

ravenclaw900, 2021-04-11:
Yeah, we broke its main security feature by exposing it to LAN.  See https://github.com/ipfs/ipfs-webui/issues/836, there is currently no way to secure it, as using the read-only web UI doesn't work.

closed 2021-04-24 by MichaIng