MichaIng/DietPi #4233: Gitea is outdated

GitHubGeek, 2021-03-30:
#### Required Information

- DietPi version | 7.0.2
- Distro version | stretch
- Kernel version | `4.14.180+ #1 SMP PREEMPT Tue Oct 13 13:28:54 UTC 2020 armv7l GNU/Linux`
- SBC model | Odroid HC1

#### Additional Information (if applicable)

- Software title | Gitea

#### Steps to reproduce

Gitea is a few versions behind. Latest is 1.14; DietPI includes 1.11

MichaIng, 2021-03-31:
Correct, it's always the most recent version installed, but since it's not an APT package or so, it needs to be reinstalled to have it updated. I'll add this info to our documentation: https://github.com/MichaIng/DietPi-Docs/pull/411

The latest stable version is btw 1.13.6 according to GitHub release tags: https://github.com/go-gitea/gitea/releases
I updated our fallback URL: https://github.com/MichaIng/DietPi/commit/a5ef6bb072c0f44211e1f82d893cf2213c313967
But as ravenclaw900 correctly stated, this is only relevant when the download file naming schemes change, so that the auto-detection of the most recent version fails.

Joulinar, 2021-04-02:
v1.14.0 is still in development and has a RC flag. Might be available soon. 

Anyway, can this topic be closed? I guess questions was answered.

closed 2021-04-02 by MichaIng

GitHubGeek, 2021-04-13:
Thanks for the help. I didn't understand how the update work for Gitea.