MichaIng/DietPi #4234: What happened to the compilation guide for qBittorrent on DietPC?

crankynet, 2021-03-30:
I used to follow the guide at
https://github.com/qbittorrent/qBittorrent/wiki/Compilation:-DietPi-and-Raspbian-(Debian-9.0)
but it seems to be gone. Has it been moved?

crankynet, 2021-04-01:
Thank you for the links. I guess I can't compile it manually anymore since 4.3.4.1 requires Qt 5.12 and Raspbian 10 has Qt 5.11.
Thank you again.

closed 2021-04-01 by crankynet

MichaIng, 2021-04-02:
Ah true. Of course you could use the source code from the last version that compiled with Qt 5.11, but not sure if it's worth the effort compared to using v4.1.5 shipped with the Debian/Raspbian Buster repo.

crankynet, 2021-04-10:
I downloaded and installed a Qt5.15.2 binary from https://github.com/tranter/ raspberry-pi-qt-builds but qBittorrent 4.3.4.1 configure still comains.

`
checking for Qt5 qmake >= 5.12... not found
configure: error: Could not find qmake
x@y:~/build/qBittorrent-release-4.3.4.1# qmake -v
QMake version 3.1
Using Qt version 5.15.2 in /usr/local/Qt-5.15.2/lib
`

Joulinar, 2021-04-11:
Hi,

you need to specify the location of `qmake` 😉 

`./configure QT_QMAKE="/usr/local/Qt-5.15.2/bin" CXXFLAGS="-std=c++14" --disable-gui`

At least on my test this check is passed successfully 

```
...
checking whether to enable the GUI... no
checking whether to install the systemd service file... no
checking whether to enable the WebUI... yes
checking for Qt5 qmake >= 5.12... /opt/Qt/5.15.2/bin/qmake
checking whether QtDBus should be enabled... no
checking for boostlib >= 1.65 (106500)... yes
configure: Boost CXXFLAGS: "-I/usr/include"
...
```

crankynet, 2021-04-11:
@Joulinar , Thanks but is my deleted post still visible? I deleted the post after I found out adding QT_QMAKE to configure.

Joulinar, 2021-04-11:
No it is not visible but I hat that page open for some days and just replied 🤣

crankynet, 2021-04-11:
Now I stuck at
`
checking for Qt5 qmake >= 5.12... /usr/local/Qt-5.15.2/bin/qmake
checking whether QtDBus should be enabled... no
checking for boostlib >= 1.65 (106500)... yes
configure: Boost CXXFLAGS: "-I/usr/include"
configure: Boost LDFLAGS: "-L/usr/lib"
checking whether the Boost::System library is available... yes
configure: error: Could not find a version of the Boost::System library!
`

Joulinar, 2021-04-11:
yes basically you can specify all needed libs. For me it was quite challenging to find all possible parameter 🤣

crankynet, 2021-04-11:
Did you compile Qt yourself?

Joulinar, 2021-04-11:
yes I did, as I was not sure if there is a version to download somewhere for 64bit. 

I followed this guide. https://www.tal.org/tutorials/building-qt-512-raspberry-pi

A little bit of try and error but at the end it was working somehow. (running over night to complete. 🙄 )

crankynet, 2021-04-11:
I use the binary from https://github.com/tranter/raspberry-pi-qt-builds and compilation ends up with error.

`make -j$(nproc)
cd src/ && ( test -e Makefile || /usr/local/Qt-5.15.2/bin/qmake -o Makefile /root/build/qBittorrent-release-4.3.4.1/src/src.pro QMAKE_LRELEASE= ) && make -f Makefile 
make[1]: Entering directory '/root/build/qBittorrent-release-4.3.4.1/src'
compiling base/bittorrent/tracker.cpp
compiling base/http/responsebuilder.cpp
compiling base/http/responsegenerator.cpp
compiling base/http/server.cpp
In file included from base/http/server.cpp:31:
base/http/server.h:66:17: error: field m_key has incomplete type QSslKey
         QSslKey m_key;
                 ^~~~~
In file included from /usr/local/Qt-5.15.2/include/QtNetwork/QSslCertificate:1,
                 from base/http/server.h:34,
                 from base/http/server.cpp:31:
/usr/local/Qt-5.15.2/include/QtNetwork/qsslcertificate.h:63:7: note: forward declaration of class QSslKey
 class QSslKey;
       ^~~~~~~
base/http/server.cpp:52:11: error: QSslCipher was not declared in this scope
     QList<QSslCipher> safeCipherList()
           ^~~~~~~~~~
base/http/server.cpp:52:11: note: suggested alternative: QSslKey
     QList<QSslCipher> safeCipherList()
           ^~~~~~~~~~
           QSslKey
base/http/server.cpp:52:21: error: template argument 1 is invalid
     QList<QSslCipher> safeCipherList()
                     ^
base/http/server.cpp: In function int {anonymous}::safeCipherList():
base/http/server.cpp:55:21: error: QSslCipher was not declared in this scope
         const QList<QSslCipher> allCiphers {QSslConfiguration::supportedCiphers()};
                     ^~~~~~~~~~
base/http/server.cpp:55:21: note: suggested alternative: badCiphers
         const QList<QSslCipher> allCiphers {QSslConfiguration::supportedCiphers()};
                     ^~~~~~~~~~
                     badCiphers
base/http/server.cpp:55:31: error: template argument 1 is invalid
         const QList<QSslCipher> allCiphers {QSslConfiguration::supportedCiphers()};
                               ^
base/http/server.cpp:55:64: error: incomplete type QSslConfiguration used in nested name specifier
         const QList<QSslCipher> allCiphers {QSslConfiguration::supportedCiphers()};
                                                                ^~~~~~~~~~~~~~~~
base/http/server.cpp:56:25: error: type/value mismatch at argument 1 in template parameter list for template<class T> class QList
         QList<QSslCipher> safeCiphers;
                         ^
base/http/server.cpp:56:25: note:   expected a type, got QSslCipher
base/http/server.cpp:57:33: error: request for member cbegin in allCiphers, which is of non-class type const int
         std::copy_if(allCiphers.cbegin(), allCiphers.cend(), std::back_inserter(safeCiphers), [&badCiphers](const QSslCipher &cipher)
                                 ^~~~~~
base/http/server.cpp:57:54: error: request for member cend in allCiphers, which is of non-class type const int
         std::copy_if(allCiphers.cbegin(), allCiphers.cend(), std::back_inserter(safeCiphers), [&badCiphers](const QSslCipher &cipher)
                                                      ^~~~
In file included from /usr/include/c++/8/bits/stl_algobase.h:67,
                 from /usr/include/c++/8/algorithm:61,
                 from /usr/local/Qt-5.15.2/include/QtCore/qglobal.h:142,
                 from /usr/local/Qt-5.15.2/include/QtCore/qchar.h:43,
                 from /usr/local/Qt-5.15.2/include/QtCore/qhash.h:44,
                 from /usr/local/Qt-5.15.2/include/QtCore/qset.h:43,
                 from /usr/local/Qt-5.15.2/include/QtCore/QSet:1,
                 from base/http/server.h:33,
                 from base/http/server.cpp:31:
/usr/include/c++/8/bits/stl_iterator.h: In instantiation of class std::back_insert_iterator<int>:
base/http/server.cpp:57:92:   required from here
/usr/include/c++/8/bits/stl_iterator.h:490:7: error: int is not a class, struct, or union type
       operator=(const typename _Container::value_type& __value)
       ^~~~~~~~
/usr/include/c++/8/bits/stl_iterator.h:497:7: error: int is not a class, struct, or union type
       operator=(typename _Container::value_type&& __value)
       ^~~~~~~~
[... 269 lines cut ...]
                   ^~~
base/http/server.cpp:137:57: error: invalid use of incomplete type class QSslKey
     const QSslKey key {Utils::Net::loadSSLKey(privateKey)};
                                                         ^
In file included from /usr/local/Qt-5.15.2/include/QtNetwork/QSslCertificate:1,
                 from base/http/server.h:34,
                 from base/http/server.cpp:31:
/usr/local/Qt-5.15.2/include/QtNetwork/qsslcertificate.h:63:7: note: forward declaration of class QSslKey
 class QSslKey;
       ^~~~~~~
make[1]: *** [Makefile:9040: server.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from base/bittorrent/tracker.cpp:41:
./base/http/server.h:66:17: error: field m_key has incomplete type QSslKey
         QSslKey m_key;
                 ^~~~~
In file included from /usr/local/Qt-5.15.2/include/QtNetwork/QSslCertificate:1,
                 from ./base/http/server.h:34,
                 from base/bittorrent/tracker.cpp:41:
/usr/local/Qt-5.15.2/include/QtNetwork/qsslcertificate.h:63:7: note: forward declaration of class QSslKey
 class QSslKey;
       ^~~~~~~
make[1]: *** [Makefile:8372: tracker.o] Error 1
make[1]: Leaving directory '/root/build/qBittorrent-release-4.3.4.1/src'
make: *** [Makefile:47: sub-src-make_first] Error 2`

crankynet, 2021-07-25:
@Joulinar, how did you build libtorrent under Raspbian 64-bit? When I ran configure, I got 
"configure: error: Could not find a version of the Boost::System library!" 

https://github.com/qbittorrent/qBittorrent/discussions/15222

Joulinar, 2021-07-25:
I did it once for testing and basically followed this guide. https://www.tal.org/tutorials/building-qt-512-raspberry-pi

A little bit of try and error but at the end it was working somehow.