MichaIng/DietPi #4235: Unable to update Allo Sparky to 6.35

b-chandar, 2021-03-31:
#### Details:
- Date           | Wed Mar 31 13:34:28 BST 2021
- Bug report     | 00a3ec72-cae0-4382-9d0d-ff197c6bff6c
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Updating my Allo Sparky from 6.34.3 to 6.35
2. Ran the dietpi-update
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_81+0~20210223.34+debian9~1.gbpf52eb0_all.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

MichaIng, 2021-03-31:
Many thanks for your report.

Jep the PHP repo key as expired. Please run the following two commands to have it updated:
```
apt-key del '95BD4743' # removes old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # applies new key without deprecated apt-key as separate file
```

b-chandar, 2021-03-31:
Thank you for the quick update. I followed your instructions and re-attempted the update process. Haven't been successful :-(

```
#### Details:
- Date           | Wed Mar 31 16:41:06 BST 2021
- Bug report     | 00a3ec72-cae0-4382-9d0d-ff197c6bff6c
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi1 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:

E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_81+0~20210223.34+debian9~1.gbpf52eb0_all.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

```

MichaIng, 2021-03-31:
Please try to clean the apt cache first and run the APT calls manually first:
```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

b-chandar, 2021-03-31:
Thank you, works now 👍

Joulinar, 2021-04-02:
Guys I'm going to close the topic as issue seems to be solved. Feel free to reopen if needed.

closed 2021-04-02 by Joulinar