MichaIng/DietPi #4237: Jackett config empty after update

trinsic, 2021-04-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.0.2
- Distro version | Buster
- Kernel version | 4.19.0-16-amd64
- SBC model | x86 Intel
- Power supply used | PSU 450w
- SDcard used | SSD

#### Additional Information (if applicable)
- Software title | Jackett
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Maybe

#### Steps to reproduce
1. Allow Jackett to auto-update

#### Expected behaviour
1. Work as normal

#### Actual behaviour
1. Config and indexers are gone on restart

#### Extra details
Jackett has fixed a bug and now decided to move the config to /root/.config/Jackett with their latest updates, however, this doesn't seem to be working and now I end up with an empty config and all my indexers removed from /opt/jackett/Jackett.

Reading their thread on it [#11313](https://github.com/Jackett/Jackett/issues/11313), I tried pointing to the Jackett config but it still doesn't work. I moved my config to .config and it doesn't get touched.

Zeuskk, 2021-04-01:
Confirm this problem, already also mentioned about it on DietPi forum.

Arghh, 2021-04-01:
Can confirm.

DietPi has set jackett home to /opt/jacket/Jackett and the config-files get saved there, but after restarting the service they are gone. The problem started after a jackett update. I think it's a jackett issue.

josh3003, 2021-04-02:
Also having the issue today and reported it to /r/dietpi

ololizoz, 2021-04-02:
Also having this issue. The Jackett config seems to reset after a reboot.

josh3003, 2021-04-02:
Anyway to escalate the problem??

Joulinar, 2021-04-02:
What do you want to escalate?

josh3003, 2021-04-02:
To get a workaround or fix going?

Joulinar, 2021-04-02:
This seems more an issue of Jackett itself if I read the entry on Jackett GitHub correctly. At least I would expect a software to respect actual configuration and not to delete it.

josh3003, 2021-04-02:
Did you have a prev install of jackett that wont auto update perhaps? Will keep an ear out on main thread as well. Thanks.

Joulinar, 2021-04-02:
That's something you would need to configure in Jackett I guess. DietPi is going to install official software always and not an own version

PrzemekSkw, 2021-04-02:
I also have that issue. Config is gone and API changes. I use mini PC.

Regards.

jspoonamore, 2021-04-02:
Here's my fix/work-around. I modified /etc/systemd/system/jacket.service.
I pointed the config setting to a new folder (/opt/jackett/jackett_config). I restarted Jackett and settings did not get wiped.

Joulinar, 2021-04-02:
For sure DietPi install script could be adjusted but would be good to know what the final folder structure should be. As well, this would impact new installations + re-install only. Already running systems would not be touched and will end up on same situation of the update will be applied. 😉

MichaIng, 2021-04-02:
Having the configs in a separate dir is likely the right solution for this. If we know the files, we can move them on DietPi update as well. I'd go with `/mnt/dietpi_userdata/jackett`, which aligns with how do it in other cases as well:
- Install dir: `/opt/<software_name>`
- Data/config dir: `/mnt/dietpi_userdata/<software_name>`

So the install dir can be removed without loosing any data or config, and the data can be easily moved to an external drive (e.g. to reduce disk I/O when a constantly written database or log file is contained) and easily can be migrated to new systems, backed up etc.

Just faced the same with Medusa, where both, install+config/data is both in `/mnt/dietpi_userdata/medusa` currently. Good to have that separated consequently, where possible.

ololizoz, 2021-04-02:
Would someone care to give a hand as to how one would go about applying this workaround (newbie friendly)? I've tried creating /mnt/dietpi_userdata/jackett, copying the contents of /opt/jackett to there and changing the working directory in jackett.config, but that just broke Jackett.

MichaIng, 2021-04-02:
I would also need to check how to do it. When Jackett now fails, checking it's logs would be a start:
```
journalctl -u jackett
```
I guess the service file needs to be adjusted as well.

ololizoz, 2021-04-02:
> I would also need to check how to do it. When Jackett now fails, checking it's logs would be a start:
> 
> ```
> journalctl -u jackett
> ```
> 
> I guess the service file needs to be adjusted as well.

Log tells me that jackett fails to start because of read-only filesystem, which I assume has something to do with the permissions of the new folder I created? However, this is way above my technical level. I'm amazed I even got dietpi up and running.

MichaIng, 2021-04-02:
Ah, the jackett user needs to own the directory of course:
```
chown -R jackett:jackett /mnt/dietpi_userdata/jackett
```

ololizoz, 2021-04-02:
This is what I've tried, however the settings are still removed after reboot. Please advise on where I went wrong. I'd love to get this up and running.
> mkdir /mnt/dietpi_userdata/jackett
> cp -r /opt/jackett . /mnt/dietpi_userdata
> chown -R jackett:jackett /mnt/dietpi_userdata/jackett
> nano /etc/systemd/system/jackett.service
> -> edit all folder mentions to /mnt/dietpi_userdata/jackett (make sure that the one that has an extra /jacket also has an extra after you have edited the line)
> systemctl daemon-reload
> systemctl restart jackett.service
> sudo reboot
>

jspoonamore, 2021-04-02:
Hey guys....  Let's keep my solution/quick-fix simple.  You do not have to create the folder for the config files 1st.  Jackett will create the folder automatically if you change the folder destination in /etc/systemd/system/jacket.service.  I changed the config folder to /opt/jackett/jackett_config because it would be in the folder where Jackett was installed anyway.  And... Jackett would set the folder rights privileges accordingly.

wdesponts, 2021-04-02:
Thanks @jspoonamore, your fix did the trick.

wdesponts, 2021-04-02:
For clarity the new jackett.service file contents should look like this:
`[Unit]
Description=Jackett (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
SyslogIdentifier=jackett
User=jackett
WorkingDirectory=/opt/jackett
Environment=XDG_CONFIG_HOME=/opt/jackett/jackett_config
ExecStart=/bin/dash -c '/opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater > /dev/null; do sleep 1; done'
Restart=always
RestartSec=5
TimeoutStopSec=30

# Hardening
ProtectSystem=strict
ProtectHome=true
PrivateDevices=true
PrivateTmp=true
ReadWritePaths=/opt/jackett

[Install]
WantedBy=multi-user.target
`

ololizoz, 2021-04-02:
> For clarity the new jackett.service file - Service Section should look like this:
> 
> [Service]
> SyslogIdentifier=jackett
> User=jackett
> WorkingDirectory=/opt/jackett
> Environment=XDG_CONFIG_HOME=/opt/jackett/jackett_config
> ExecStart=/bin/dash -c '/opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater > /dev/null; do sleep 1; done'
> Restart=always
> RestartSec=5
> TimeoutStopSec=30

I'm honestly not sure what I'm doing wrong. On a fresh install of Jackett, this is the only modification I'm doing however it results in Jackett not showing

> For clarity the new jackett.service file - Service Section should look like this:
> 
> [Service]
> SyslogIdentifier=jackett
> User=jackett
> WorkingDirectory=/opt/jackett
> Environment=XDG_CONFIG_HOME=/opt/jackett/jackett_config
> ExecStart=/bin/dash -c '/opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater > /dev/null; do sleep 1; done'
> Restart=always
> RestartSec=5
> TimeoutStopSec=30

Thanks! This seems to do the trick. The config file still doensn't seem to survive changing the settings in Jackett, however it does now survive a reboot.

Joulinar, 2021-04-03:
There might be a fix available soon https://github.com/Jackett/Jackett/pull/11465

MichaIng, 2021-04-03:
Great! We should still separate install and config dir.

paliasp, 2021-04-03:
It worked fine for me with a few tweaks in the systemd file.
https://dietpi.com/phpbb/viewtopic.php?p=33340#p33340

Arghh, 2021-04-04:
Fixed in Jackett v0.17.841 with https://github.com/Jackett/Jackett/pull/11465

Joulinar, 2021-04-04:
Yep new version has been released by Jackett today.

closed 2021-04-24 by MichaIng