MichaIng/DietPi #4238: MedUSA won't upgrade

xanderphillips, 2021-04-01:
I've tried updating dietpi to latest version available, and tried upgrading software by doing the dietpi-sofware reinstall 116 as well as through the medusa upgrade option on it's web console.

Running on a PRi 4B, dietpi 7.0.
Medusa is stuck at 0.4.3 no matter what I do.
Not sure what the process is to capture logs / versions / error messages needed to help troubleshoot.
When I do the reinstall 116 it LOOKS like it sucessfully removed and re-installed medusa, but the version never goes up.
Any help greatly appreciated!

xanderphillips, 2021-04-01:
And today after a thousand tries it 'just worked'.  Please ignore this ticket.  No clue what changed.  Thanks anyway!

xanderphillips, 2021-04-01:
Problem resolved itself.

closed 2021-04-01 by xanderphillips

MichaIng, 2021-04-01:
Medusa has an internal updater, hasn't it? At least on a reinstall we skip downloading the new application code an print an info to use the internal updater from the web interface.

xanderphillips, 2021-04-01:
the internal updater wasn't working, got a failure.  But for some unknown
reason it worked today?!?!  I must have tried updating that app at least 50
times over the past months... Oh well, it's working now!
Thanks!

MichaIng, 2021-04-02:
Currently the info about using the internal updater is only an info message, while we could make it an interactive dialogue that requires to hit return to continue. It also contains the info that the install dir `/mnt/dietpi_userdata/medusa` can be removed to force a fresh install. But the issue in case of Medusa is that it's config/database is stored in this very same directory, right?

Probably we can have both separated:
- Install dir in `/opt/medusa`, which can be removed without loosing any config/data.
- Data/config in `/mnt/dietpi_userdata/medusa`, so that it can be easily moved to external drive, migrated to new systems etc.

Then we could even make the dialogue a yes/no question to either do or skip the removal and fresh install of the program files 🤔.