MichaIng/DietPi #4241: Sonarr - Repository is not signed

aarongeorge, 2021-04-04:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
buster 10.9

- Kernel version
Linux DietPi 5.10.17-v71+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv71 GNU/Linux

- SBC model
RPi 4 Model B (armv71)

- Power supply used
Official RPi 4 Power Supply

- SDcard used
256GB SanDisk Ultra

#### Additional Information (if applicable)
- Software title
Sonarr

- Was the software title installed freshly or updated/migrated?
Fresh Install

- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Either a fresh install of DietPi headless with Sonarr as an application to install or;
2. Fresh install of DietPi through the interactive prompt and install Sonarr

#### Expected behaviour
- Sonarr should install

#### Actual behaviour
Error is thrown
```
Err:3 https://apt.sonarr.tv master InRelease
The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Reading package lists...
W: GPG error: https://apt.sonarr.tv master InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repositiory 'https://apt.sonarr.tv master InRelease' is not signed.
```

#### Extra details
It seems no matter how you try and get DietPi to install Sonarr on a fresh installation it throws this signature error :(

MichaIng, 2021-04-04:
Many thanks for your report.

Probably now that Sonarr v3 has been released, the key for Sonarr v2 is not valid anymore.

What does the following show?
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
```

aarongeorge, 2021-04-04:
It returns
```
gpg: keyserver receive failed: Server indicated a failure
```

aarongeorge, 2021-04-04:
It seems like going into the subshell and running the following:
```
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 2009837CBFFD68F45BC180471F4F90DE2A9B4BF8
echo "deb https://apt.sonarr.tv/debian buster main" | sudo tee /etc/apt/sources.list.d/sonarr.list
exit
```
And then retrying the last failed command works just fine

MichaIng, 2021-04-04:
Ah, so the new Sonarr v3 key works for Sonarr v2 as well, interesting. We'll migrate to Sonarr v3 with next release anyway, but I'll add the key change as quick fix now.

MichaIng, 2021-04-04:
Since we don't have one, I turned this issue into the v3 migration request. It's strange that they invalidated key for the still existing v2 repository, but that only makes the anyway reasonable migration a necessity.

aarongeorge, 2021-04-06:
@MichaIng seems I was wrong. Adding the key still fails the install :( Is there any workaround whilst we wait for the 7.1 update to drop? At the moment it's not possible to install Sonarr via regular DietPi means.

MichaIng, 2021-04-06:
> At the moment it's not possible to install Sonarr via regular `dietpi-software` means.

If the Sonarr v2 key was invalidated and removed from the key server, then there is not chance to use the current `dietpi-software` implementation, yes. Sonarr v2 and v3 differ too much, regarding name and directory structure etc. It's quite a radical step to drop the key, especially since the v2 repository is still up, so I'm not sure if this was intended or an accident.

`dietpi-software` uses/creates the "sonarr" user and makes "dietpi" it's primary group, so that it can import media data from downloaders and media players/servers can R/W that imported media.

Joulinar, 2021-04-06:
I did a test installation of Sonarr on a RPi3B+ 32bit without issues. Key got imported and is still valid

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Sonarr: automatically download TV shows

Executing: /tmp/apt-key-gpghome.5Cnfm19zCa/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
gpg: key EBFF6B99D9B78493: public key "NzbDrone <contact@nzbdrone.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Get:3 https://apt.sonarr.tv master InRelease [13.3 kB]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Get:5 https://apt.sonarr.tv master/main armhf Packages [67.9 kB]
Fetched 81.2 kB in 2s (36.9 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update
[ INFO ] DietPi-Software | APT install for: nzbdrone, please wait...
```

MichaIng, 2021-04-07:
Probably it has been fixed meanwhile.

closed 2021-04-24 by MichaIng