MichaIng/DietPi #4242: DietPi-Update terminates

feuerwasser85, 2021-04-04:
#### Details:
- Date           | Sun Apr  4 19:54:04 CEST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux pihole 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
1. login as root
2. do a "dietpi-update
3. 
#### Expected behaviour:
The update should succeed

#### Actual behaviour:
DietPi-Update terminates

#### Extra details:
-

#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/shadow/login_4.4-4.1+deb9u1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/shadow/passwd_4.4-4.1+deb9u1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/ca-certificates/ca-certificates_20200601~deb9u2_all.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-opcache_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-json_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-xml_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-readline_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-sqlite3_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-fpm_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-curl_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-mbstring_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-gd_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-zip_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-cli_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.4/php7.4-common_7.4.15-7+0~20210227.41+debian9~1.gbpb91400_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

ravenclaw900, 2021-04-04:
Hello,
It looks like you have 2 different problems here.  One is with your DNS, and the other is the fact that the `packages.sury.org` repo key has expired.  The latter can be fixed by running:
```
apt-key del '95BD4743' # Removes old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # Applies new key without deprecated apt-key as separate file
```
And, could you also post the output of `ping -c 1 raspbian.raspberrypi.org`?

Joulinar, 2021-04-04:
Indeed main problem seems to be DNS related as you are not able to resolve host name for respbian repository. Are you using something like PiHole or any other DNS / AdBlock on your network?

closed 2021-04-29 by MichaIng