MichaIng/DietPi #4243: Dietpi Update error

ZiggyPi, 2021-04-04:
#### Details:
- Date           | Sun  4 Apr 19:54:48 CEST 2021
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi-HTPC 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Run dietpi-update
2. ...

#### Expected behaviour:

-  i except it to install dietpi 6.35 over my 6.28

#### Actual behaviour:

- error with aptpkg cache and apt-show-versions

#### Extra details:

- i wantes to update because jackett wasnt working anymore. All indexers were gone and sonarr and radarr also stopped working.

#### Additional logs:
```
Hit:1 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease
Hit:2 https://downloads.plex.tv/repo/deb public InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Hit:4 https://download.docker.com/linux/raspbian buster InRelease
Hit:5 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Hit:6 https://apt.sonarr.tv develop InRelease
Hit:7 https://mirror.netcologne.de/raspbian/raspbian buster InRelease
Can't locate AptPkg/Cache.pm in @INC (you may need to install the AptPkg::Cache module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/bin/apt-show-versions line 40.
BEGIN failed--compilation aborted at /usr/bin/apt-show-versions line 40.
Reading package lists...
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code

Any idea how i can resolve this.
I dont really need the update. But i was hoping it would fix jackett. But now i can get the terminal up but nothing else is working. No access to sonarr, radarr or other webservices and no acceas to samba.

Joulinar, 2021-04-04:
Hi, 

Yes there was an issue on Jackett and a new version of Jackett has been released today that should adress this issue.  Best to my knowledge. However this not related to the issue running `apt` package update. Pls can you try following 

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

ZiggyPi, 2021-04-04:
@Joulinar

No luck

 
DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi-HTPC:~# apt update
Get:1 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease [17.5 kB]
Get:2 https://downloads.plex.tv/repo/deb public InRelease [6,685 B]
Get:3 https://download.docker.com/linux/raspbian buster InRelease [29.7 kB]Get:4 https://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:5 https://apt.sonarr.tv develop InRelease [13.3 kB]
Get:6 https://download.mono-project.com/repo/debian raspbianbuster InRelease [2,238 B]
Get:7 https://mirror.netcologne.de/raspbian/raspbian buster InRelease [15.0 kB]
Get:8 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial/main armhf Packages [3,128 B]
Get:9 https://downloads.plex.tv/repo/deb public/main armhf Packages [1,644 B]
Get:10 https://download.docker.com/linux/raspbian buster/stable armhf Packages [12.0 kB]
Get:11 https://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Get:12 https://download.mono-project.com/repo/debian raspbianbuster/main armhf Packages [47.5 kB]
Get:13 https://apt.sonarr.tv develop/main armhf Packages [109 kB]
Get:14 https://mirror.netcologne.de/raspbian/raspbian buster/rpi armhf Packages [1,360 B]
Get:15 https://mirror.netcologne.de/raspbian/raspbian buster/contrib armhf Packages [58.7 kB]
Get:16 https://mirror.netcologne.de/raspbian/raspbian buster/non-free armhf Packages [104 kB]
Get:17 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.8 MB in 13s (1,090 kB/s)
Can't locate AptPkg/Cache.pm in @INC (you may need to install the AptPkg::Cache module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/bin/apt-show-versions line 40.
BEGIN failed--compilation aborted at /usr/bin/apt-show-versions line 40.
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code

Joulinar, 2021-04-04:
hmm there seems to be an issue with `apt-show-versions`. Let's try following without hitting YES 😉 to see if there are any dependency before removing.

`apt remove apt-show-versions`

ZiggyPi, 2021-04-04:
@Joulinar 

Okay so i tried it ans got the following

root@DietPi-HTPC:~# apt-show-versions
Can't locate AptPkg/Cache.pm in @INC (you may need to install the AptPkg::Cache module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/bin/apt-show-versions line 40.
BEGIN failed--compilation aborted at /usr/bin/apt-show-versions line 40.
root@DietPi-HTPC:~# apt remove apt-show-versions
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'apt-show-versions' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

So i tried installing it and got the following error

root@DietPi-HTPC:~# aptitude install apt-show-versions
The following NEW packages will be installed:
  apt-show-versions libapt-pkg-perl{a}
0 packages upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 98.5 kB of archives. After unpacking 299 kB will be used.
Do you want to continue? [Y/n/?] y
Get: 1 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf libapt-pkg-perl armhf 0.1.34+b1 [65.9 kB]
Get: 2 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf apt-show-versions all 0.22.11 [32.6 kB]
Fetched 98.5 kB in 0s (262 kB/s)
dpkg: unrecoverable fatal error, aborting:
 unknown system group 'ssl-cert' in statoverride file; the system group got removed
before the override, which is most probably a packaging bug, to recover youcan remove the override manually with dpkg-statoverride
E: Sub-process /usr/bin/dpkg returned an error code (2)

Joulinar, 2021-04-04:
@MichaIng 
can you have a look. There seems to be something broken on `apt`

wilsonmsalberto, 2021-04-05:
Similar problem with fetch with `apt` preventing update from v6.34.3 to v6.35.0, submitted the error:

Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/phpcommon_81+0~20210223.34+debian10~1.gbpf52eb0_all.deb

Joulinar, 2021-04-05:
@wilsonmsalberto 
well your issue seems to be different. Can you share full output of `apt update && apt upgrade` pls

wilsonmsalberto, 2021-04-05:
@Joulinar https://pastebin.com/M53mkDrM

Thanks for checking in!

Joulinar, 2021-04-05:
ok the PGP key for PHP repository got invalid

```
Err:4 https://packages.sury.org/php buster InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Reading package lists... Done
```

similar to #4235

You can try following to fix

```
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

wilsonmsalberto, 2021-04-05:
Thank you! Seems to be updating everything properly, will confirm after the fact.

Joulinar, 2021-04-05:
wow you have 213 packages that can be upgraded. Are you going to perform a full system upgrade from Stretch to Buster? I hope you created a backup before 😉

wilsonmsalberto, 2021-04-05:
I did backup, yes. I've only been using the dietpi-update command, don't usually run anything else.

But, everything is working after the update, all is well that ends well. Thanks for all the help!

ZiggyPi, 2021-04-05:
@Joulinar 

Tried it, but still same result

root@DietPi-HTPC:~# apt-key del '95BD4743'
OK
root@DietPi-HTPC:~# curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
root@DietPi-HTPC:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi-HTPC:~# apt update
Get:1 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease [17.5 kB]
Get:2 https://downloads.plex.tv/repo/deb public InRelease [6,685 B]
Get:3 https://download.docker.com/linux/raspbian buster InRelease [29.7 kB]Get:4 https://apt.sonarr.tv develop InRelease [13.3 kB]
Get:5 https://download.mono-project.com/repo/debian raspbianbuster InRelease [2,238 B]
Get:6 https://mirror.netcologne.de/raspbian/raspbian buster InRelease [15.0 kB]
Get:7 https://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:8 https://downloads.plex.tv/repo/deb public/main armhf Packages [1,644 B]
Get:9 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial/main armhf Packages [3,128 B]
Get:10 https://download.docker.com/linux/raspbian buster/stable armhf Packages [12.0 kB]
Get:11 https://download.mono-project.com/repo/debian raspbianbuster/main armhf Packages [47.5 kB]
Get:12 https://apt.sonarr.tv develop/main armhf Packages [109 kB]
Get:13 https://mirror.netcologne.de/raspbian/raspbian buster/contrib armhf Packages [58.7 kB]
Get:14 https://mirror.netcologne.de/raspbian/raspbian buster/rpi armhf Packages [1,360 B]
Get:15 https://mirror.netcologne.de/raspbian/raspbian buster/non-free armhf Packages [104 kB]
Get:16 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf Packages [13.0 MB]
Get:17 https://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Fetched 13.8 MB in 13s (1,086 kB/s)
Can't locate AptPkg/Cache.pm in @INC (you may need to install the AptPkg::Cache module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/bin/apt-show-versions line 40.
BEGIN failed--compilation aborted at /usr/bin/apt-show-versions line 40.
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code
root@DietPi-HTPC:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

MichaIng, 2021-04-05:
Is there a specific reason for using `apt-show-versions`? I reviewed it a while back where it was a dependency of Webmin, but it it has only downsides and can easily completely be replaced by native `apt-cache` and `apt list` calls, which are much faster as well. As a result Webmin has removed it from their dependencies.

I see you installed it above, but actually it obviously was somehow installed before already, at least the APT calls tried to invoke it, meaning there was a hook in `/etc/apt/apt.conf.d`.

Please try:
```
apt purge apt-show-versions
ls -l /etc/apt/apt.conf.d # Check if there is still a hook for apt-show-versions
apt update
```

ZiggyPi, 2021-04-05:
@MichaIng

Still no change

 root@DietPi-HTPC:~# apt purge apt-show-versions
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'apt-show-versions' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi-HTPC:~# ls -l /etc/apt/apt.conf.d # Check if there is still a hook for apt-show-versions
total 32
-r--r--r-- 1 root root  440 Dec 24  2019 01autoremove-kernels
-rw-r--r-- 1 root root  162 Feb 16  2019 20apt-show-versions
-rw-r--r-- 1 root root  182 Jul 14  2019 70debconf
-rw-r--r-- 1 root root 1370 Mar 11 17:06 97dietpi
-rw-r--r-- 1 root root   27 Dec 24  2019 98-dietpi-no_translations
-rw-r--r-- 1 root root   62 Jul 14  2019 99-dietpi-forceconf
-rw-r--r-- 1 root root   27 Jul 14  2019 99-dietpi-force-ipv4
-rw-r--r-- 1 root root  154 Jul 14  2019 99-dietpi-norecommends
root@DietPi-HTPC:~# apt update
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://mirror.netcologne.de/raspbian/raspbian buster InRelease
Hit:3 https://download.docker.com/linux/raspbian buster InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://apt.sonarr.tv develop InRelease
Hit:6 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease
Hit:7 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Can't locate AptPkg/Cache.pm in @INC (you may need to install the AptPkg::Cache module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/bin/apt-show-versions line 40.
BEGIN failed--compilation aborted at /usr/bin/apt-show-versions line 40.
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code

MichaIng, 2021-04-05:
Do the following:
```
rm 20apt-show-versions
apt update
```

ZiggyPi, 2021-04-05:
@MichaIng 

This seems to have done the trick, atleast for the most part. A couple of services still failed and cant reach sonarr via IP. Radarr en deluge are reachable via IP.
─────────────────────────────────────────────────────
 DietPi v6.28.0 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 42'C : 107'F (Optimal temperature)
 - LAN IP : 10.0.0.30 (wlan0)
 - MOTD :        \     /
                 \\   //
                  )\-/(
                  /e e\
                 ( =T= )
                 /`---'\
            ____/ /___\ \
       \   /   '''     ```~~"--.,_
    `-._\ /   Happy Easter!       `~~"--.,_
   ------>|     by all from DietPi         `~~"--.,_
    _.-'/ \                            ___,,,---""~~``'
       /   \____,,,,....----""""~~~~````
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community | oct8l
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-update   : Run now to update DietPi from v6.28.0 to v6.35.0.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi-HTPC:~# rm /etc/apt/apt.conf.d/20apt-show-versions
root@DietPi-HTPC:~# apt update
Hit:1 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease
Hit:2 https://downloads.plex.tv/repo/deb public InRelease
Hit:3 https://download.docker.com/linux/raspbian buster InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://apt.sonarr.tv develop InRelease
Hit:6 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Get:7 https://mirror.netcologne.de/raspbian/raspbian buster InRelease [15.0 kB]
RPi 4 Model B (armv7l) | IP: 10.0.0.30                                     ┌────────────────────────────┤ DietPi-Update ├────────────────────────────┐│                                                                         ││ >----------------------------------Notice------------------------------ ││ ----<                                                                   ││ - A benefit of DietPi is: We use standard Linux (Debian) configurations ││ and commands.                                                           ││ - A potential downside is: We can't possibly accommodate or predict all ││ modification to Linux configurations files by the end user, outside of  ││ DietPi programs, during updates.                                        ││                                                                         ││ Although we test the updates thoroughly, if you have made any custom    ││ changes to Linux configuration files outside of the DietPi programs, an ││ update may trigger a potential issue.                                   ││ >---------------------------------------------------------------------- ││ ----<                                                                   ││                                                                         ││ Do you wish to continue and update DietPi to v6.35.0?                   ││                                                                         ││                    <Ok>                        <Cancel>                 ││                                                                         │└─────────────────────────────────────────────────────────────────────────┘[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : rpimonitor
[  OK  ] DietPi-Services | stop : deluge-web
[  OK  ] DietPi-Services | stop : htpc-manager
[  OK  ] DietPi-Services | stop : lidarr
[  OK  ] DietPi-Services | stop : radarr
[  OK  ] DietPi-Services | stop : sonarr
[  OK  ] DietPi-Services | stop : jackett
[  OK  ] DietPi-Services | stop : ubooquity
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : deluged
[  OK  ] DietPi-Services | stop : coturn
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : php7.1-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[... 309 lines cut ...]
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : nfs-kernel-server
[FAILED] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.3-fpm
[  OK  ] DietPi-Services | restart : nginx
[FAILED] DietPi-Services | restart : lighttpd
[FAILED] DietPi-Services | restart : coturn
[  OK  ] DietPi-Services | restart : deluged
[  OK  ] DietPi-Services | restart : plexmediaserver
[  OK  ] DietPi-Services | restart : ubooquity
[  OK  ] DietPi-Services | restart : jackett
[  OK  ] DietPi-Services | restart : sonarr
[  OK  ] DietPi-Services | restart : radarr
[  OK  ] DietPi-Services | restart : lidarr
[  OK  ] DietPi-Services | restart : deluge-web
[  OK  ] DietPi-Services | restart : rpimonitor
[  OK  ] DietPi-Services | restart : htpc-manager
[  OK  ] DietPi-Services | restart : docker
[  OK  ] DietPi-Services | restart : cron
[FAILED] DietPi-Services | ncc maintenance:mode --off
[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

Terminated
root@DietPi-HTPC:~#

Joulinar, 2021-04-05:
ok the `apt` issue is gone. For the failing serviced you would need to check one by one why they are failing.

ZiggyPi, 2021-04-05:
@Joulinar 

This maybe a noob question, but how would i begin doing that?

Joulinar, 2021-04-05:
let's check what services we have in total `dietpi-services status`

ZiggyPi, 2021-04-05:
@Joulinar 

As instructed... FYI my dietpi is only connected via wifi.

root@DietPi-HTPC:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | nmbd                 active (running) since Mon 2021-04-05 21:58:04 CEST; 3h 20min ago
[  OK  ] DietPi-Services | smbd                 active (running) since Mon 2021-04-05 21:58:04 CEST; 3h 20min ago
[  OK  ] DietPi-Services | nfs-kernel-server    active (exited) since Mon 2021-04-05 21:58:04 CEST; 3h 20min ago
[FAILED] DietPi-Services | ● redis-server.service - Advanced key-value store
   Loaded: loaded (/lib/systemd/system/redis-server.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
     Docs: http://redis.io/documentation,
           man:redis-server(1)
  Process: 3160 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=127)

Apr 05 21:58:07 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 05 21:58:07 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 5.
Apr 05 21:58:07 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 05 21:58:07 DietPi-HTPC systemd[1]: redis-server.service: Start request repeated too quickly.
Apr 05 21:58:07 DietPi-HTPC systemd[1]: redis-server.service: Failed with result 'exit-code'.
Apr 05 21:58:07 DietPi-HTPC systemd[1]: Failed to start Advanced key-value store.
[  OK  ] DietPi-Services | mariadb              active (running) since Mon 2021-04-05 21:58:06 CEST; 3h 20min ago
[  OK  ] DietPi-Services | php7.3-fpm           active (running) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
[  OK  ] DietPi-Services | nginx                active (running) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
[FAILED] DietPi-Services | ● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-04-05 21:58:10 CEST; 3h 20min ago
  Process: 3398 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)

Apr 05 21:58:10 DietPi-HTPC systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Apr 05 21:58:10 DietPi-HTPC systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Apr 05 21:58:10 DietPi-HTPC systemd[1]: Stopped Lighttpd Daemon.
Apr 05 21:58:10 DietPi-HTPC systemd[1]: lighttpd.service: Start request repeated too quickly.
Apr 05 21:58:10 DietPi-HTPC systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 05 21:58:10 DietPi-HTPC systemd[1]: Failed to start Lighttpd Daemon.
[FAILED] DietPi-Services | ● coturn.service - coTURN STUN/TURN Server
   Loaded: loaded (/lib/systemd/system/coturn.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/coturn.service.d
           └─dietpi-logging.conf
   Active: failed (Result: exit-code) since Mon 2021-04-05 21:58:08 CEST; 3h 20min ago
     Docs: man:coturn(1)
           man:turnadmin(1)
           man:turnserver(1)
  Process: 3312 ExecStart=/usr/bin/turnserver --daemon -c /etc/turnserver.conf --pidfile /run/turnserver/turnserver.pid -l stdout --no-stdout-log --simple-log (code=exited, status=217/USER)

Apr 05 21:58:08 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 05 21:58:08 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 5.
Apr 05 21:58:08 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 05 21:58:08 DietPi-HTPC systemd[1]: coturn.service: Start request repeated too quickly.
Apr 05 21:58:08 DietPi-HTPC systemd[1]: coturn.service: Failed with result 'exit-code'.
Apr 05 21:58:08 DietPi-HTPC systemd[1]: Failed to start coTURN STUN/TURN Server.
[  OK  ] DietPi-Services | deluged              active (running) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
[  OK  ] DietPi-Services | plexmediaserver      active (running) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
[FAILED] DietPi-Services | ● ubooquity.service - Ubooquity (DietPi)
   Loaded: loaded (/etc/systemd/system/ubooquity.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-04-05 21:58:07 CEST; 3h 20min ago
  Process: 3279 ExecStart=/usr/bin/java -jar /mnt/dietpi_userdata/ubooquity/Ubooquity.jar --headless --remoteadmin --adminport 2038 --libraryport 2039 (code=exited, status=217/USER)
 Main PID: 3279 (code=exited, status=217/USER)

Apr 05 21:58:07 DietPi-HTPC systemd[1]: Started Ubooquity (DietPi).
Apr 05 21:58:07 DietPi-HTPC systemd[3279]: ubooquity.service: Failed to determine user credentials: No such process
Apr 05 21:58:07 DietPi-HTPC systemd[3279]: ubooquity.service: Failed at step USER spawning /usr/bin/java: No such process
Apr 05 21:58:07 DietPi-HTPC systemd[1]: ubooquity.service: Main process exited, code=exited, status=217/USER
Apr 05 21:58:07 DietPi-HTPC systemd[1]: ubooquity.service: Failed with result 'exit-code'.
[  OK  ] DietPi-Services | jackett              activating (auto-restart) since Tue 2021-04-06 01:18:43 CEST; 4s ago
[... 39 lines cut ...]
Apr 05 21:57:54 DietPi-HTPC systemd[1]: /lib/systemd/system/xrdp.service:8: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp.pid → /run/xrdp/xrdp.pid; please update the unit [... cut ...]
Apr 04 16:55:53 DietPi-HTPC dietpi-wifi-monitor.sh[780]: Checking connection for: wlan via ping to default gateway every 10 seconds
Apr 04 16:55:53 DietPi-HTPC dietpi-wifi-monitor.sh[780]: Cannot find device "wlan"
Apr 04 16:55:53 DietPi-HTPC dietpi-wifi-monitor.sh[780]: ERROR: WiFi adapter has been unplugged: wlan. Exiting...
Apr 04 16:55:53 DietPi-HTPC systemd[1]: dietpi-wifi-monitor.service: Main process exited, code=exited, status=1/FAILURE
Apr 04 16:55:53 DietPi-HTPC systemd[1]: dietpi-wifi-monitor.service: Failed with result 'exit-code'.

Joulinar, 2021-04-05:
do you use an external disk where you have moved part of the system on?

let's check `lighttpd`
`/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf`

MichaIng, 2021-04-06:
There is a lot of missing or unexpectedly changed (and the falsely existing one) files it seems. So before going through the individual services, please also check for kernel errors. Probably the file system itself is corrupted, causing everything else:
```
dmesg -l emerg,alert,crit,err
```

ZiggyPi, 2021-04-06:
@Joulinar 

I have an external hdd what was mounted as HTPC. But this mount appears to be gone. This was only used to store all the downloads for deluge and tv shows en movies for sonarr and radarr.

<details>

<summary>log click to expand</summary>

root@DietPi-HTPC:~# /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
2021-04-06 13:26:19: (configfile.c.1328) opening configfile  /etc/lighttpd/conf-enabled/50-phpmyadmin.conf failed: No such file or directory
2021-04-06 13:26:19: (configfile.c.1296) source: /etc/lighttpd/lighttpd.conf line: 47 pos: 15 parser failed somehow near here: (EOL)
root@DietPi-HTPC:~#
</details>

MichaIng, 2021-04-06:
Looks like you had phpMyAdmin installed and its Lighttpd config is still enabled while not being present anymore. We migrated from the Debian APT package to a source-based install, and possibly purging the package did not cleanly remove the Lighttpd config.

Just to assure:
```
dpkg -l phpmyadmin
dietpi-software list | grep phpmyadmin
```
To remove the invalid symlink:
```
rm /etc/lighttpd/conf-enabled/50-phpmyadmin.conf
```
Looks like we need to remove that one manually. I'll do tests with Apache2 as well. Not much pre-v6.26 users exist anymore, I guess, but removing some definitely false symlinks after the package purge is easy to add.

Please still run and paste the output of the command suggested above: [#issuecomment-814001887](#issuecomment-814001887)

ZiggyPi, 2021-04-06:
@MichaIng 

<details>

<summary>results from 814001887</summary>

root@DietPi-HTPC:~# dmesg -l emerg,alert,crit,err
[    0.246588] bcm2708_fb soc:fb: Unable to determine number of FB's. Assuming 1
[    0.246839] raspberrypi-firmware soc:firmware: Request 0x00048003 returned status 0x80000001
[    0.246862] bcm2708_fb soc:fb: Failed to allocate GPU framebuffer (-22)
[    0.246898] bcm2708_fb soc:fb: probe failed, err -22
[    3.580162] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.821207] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.832991] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b0
</details>

<details>

<summary> log from phpadmin</summary>

root@DietPi-HTPC:~# dpkg -l phpmyadmin
dpkg-query: no packages found matching phpmyadmin
root@DietPi-HTPC:~# dietpi-software list | grep phpmyadmin
id 90 | =2 | phpmyadmin: optional mysql admin tools | +webserver +php +mariadb | https://dietpi.com/docs/software/webserver_stack/#phpmyadmin

root@DietPi-HTPC:~# dpkg -l phpmyadmin
dpkg-query: no packages found matching phpmyadmin
root@DietPi-HTPC:~# dietpi-software list | grep phpmyadmin
id 90 | =2 | phpmyadmin: optional mysql admin tools | +webserver +php +mariadb | https://dietpi.com/docs/software/webserver_stack/#phpmyadmin
root@DietPi-HTPC:~# rm /etc/lighttpd/conf-enabled/50-phpmyadmin
rm: cannot remove '/etc/lighttpd/conf-enabled/50-phpmyadmin': No such file or directory

</details>

MichaIng, 2021-04-06:
Okay, no important kernel errors at least. Strange thing with the phpMyAdmin config, as the symlink does not exist. Does the lighttpd.conf load it directly?
```
cat /etc/lighttpd/lighttpd.conf
```

ZiggyPi, 2021-04-06:
@MichaIng 

<details>

<summary> content config file </summary>

root@DietPi-HTPC:~# cat /etc/lighttpd/lighttpd.conf
server.modules = (
        "mod_indexfile",
        "mod_access",
        "mod_alias",
        "mod_redirect",
)

server.document-root        = "/var/www"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

# strict parsing and normalization of URL for consistency and security
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails
# (might need to explicitly set "url-path-2f-decode" = "disable"
#  if a specific application is encoding URLs inside url-path)
server.http-parseopts = (
  "header-strict"           => "enable",# default
  "host-strict"             => "enable",# default
  "host-normalize"          => "enable",# default
  "url-normalize-unreserved"=> "enable",# recommended highly
  "url-normalize-required"  => "enable",# recommended
  "url-ctrls-reject"        => "enable",# recommended
  "url-path-2f-decode"      => "enable",# recommended highly (unless breaks app)
 #"url-path-2f-reject"      => "enable",
  "url-path-dotseg-remove"  => "enable",# recommended highly (unless breaks app)
 #"url-path-dotseg-reject"  => "enable",
 #"url-query-20-plus"       => "enable",# consistency in query string
)

index-file.names            = ( "index.php", "index.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.conf.pl"
include "/etc/lighttpd/conf-enabled/*.conf"

#server.compat-module-load   = "disable"
server.modules += (
        "mod_compress",
        "mod_dirlisting",
        "mod_staticfile",
)
</details>

MichaIng, 2021-04-06:
Mysterious. What is the content of the other drop-in configs?
```
for i in /etc/lighttpd/conf-enabled/*.conf; do echo "${i}:"; cat "$i"; done
```

ZiggyPi, 2021-04-06:
@MichaIng 

<details>

<summary> log </summary>

root@DietPi-HTPC:~# for i in /etc/lighttpd/conf-enabled/*.conf; do echo "${i}:"; cat "$i"; done
/etc/lighttpd/conf-enabled/05-auth.conf:
# /usr/share/doc/lighttpd/authentication.txt.gz

server.modules                += ( "mod_auth" )

# auth.backend                 = "plain"
# auth.backend.plain.userfile  = "lighttpd.user"
# auth.backend.plain.groupfile = "lighttpd.group"

# auth.backend.ldap.hostname   = "localhost"
# auth.backend.ldap.base-dn    = "dc=my-domain,dc=com"
# auth.backend.ldap.filter     = "(uid=$)"

# auth.require                 = ( "/server-status" =>
#                                (
#                                 "method"  => "digest",
#                                 "realm"   => "download archiv",
#                                 "require" => "group=www|user=jan|host=192.168.2.10"
#                               ),
#                               "/server-info" =>
#                                (
#                                 "method"  => "digest",
#                                 "realm"   => "download archiv",
#                                 "require" => "group=www|user=jan|host=192.168.2.10"
#                               )
#                              )
/etc/lighttpd/conf-enabled/10-fastcgi.conf:
# /usr/share/doc/lighttpd/fastcgi.txt.gz
# http://redmine.lighttpd.net/projects/lighttpd/wiki/Docs:ConfigurationOptions#mod_fastcgi-fastcgi

server.modules += ( "mod_fastcgi" )
/etc/lighttpd/conf-enabled/10-rewrite.conf:
# /usr/share/doc/lighttpd/rewrite.txt
# http://redmine.lighttpd.net/projects/lighttpd/wiki/Docs_ConfigurationOptions#mod_rewrite-rewriting

server.modules += ( "mod_rewrite" )
/etc/lighttpd/conf-enabled/15-fastcgi-php.conf:
# -*- depends: fastcgi -*-
# /usr/share/doc/lighttpd/fastcgi.txt.gz
# https://redmine.lighttpd.net/projects/lighttpd/wiki/Docs:ConfigurationOptions#mod_fastcgi-fastcgi

## Start an FastCGI server using php-fpm
fastcgi.server += ( ".php" =>
        ((
                "socket" => "/run/php/php7.3-fpm.sock",
                "broken-scriptfilename" => "enable"
        ))
)
/etc/lighttpd/conf-enabled/50-phpmyadmin.conf:
cat: /etc/lighttpd/conf-enabled/50-phpmyadmin.conf: No such file or directory
/etc/lighttpd/conf-enabled/98-dietpi-phpmyadmin.conf:
# https://docs.phpmyadmin.net/en/latest/setup.html#securing-your-phpmyadmin-installation
$HTTP["url"] =~ "^/phpmyadmin/(libraries|templates|tmp)($|/)" {
        url.access-deny = ("")
}
/etc/lighttpd/conf-enabled/99-dietpi-dav_redirect.conf:
# Redirect Cal/CardDAV requests to Nextcloud endpoint:
url.redirect += (
        "^/.well-known/caldav"  => "/nextcloud/remote.php/dav",
        "^/.well-known/carddav" => "/nextcloud/remote.php/dav"
)
/etc/lighttpd/conf-enabled/99-dietpi-nextcloud.conf:
# Derived from:
# - Apache: https://github.com/nextcloud/server/blob/master/.htaccess
# - Nginx:  https://docs.nextcloud.com/server/stable/admin_manual/installation/nginx.html

# Redirect OCM/OCS provider requests to Nextcloud endpoint:
url.redirect += (
        "^/ocm-provider" => "/nextcloud/ocm-provider",
        "^/ocs-provider" => "/nextcloud/ocs-provider"
)

$HTTP["url"] =~ "^/nextcloud($|/)" {

        # Hardening
        # - Directories
        $HTTP["url"] =~ "^/nextcloud/(build|tests|config|lib|3rdparty|templates|data)($|/)" { url.access-deny = ("") }
        # - Files
        $HTTP["url"] =~ "^/nextcloud/(\.|autotest|occ|issue|indie|db_|console)" { url.access-deny = ("") }
        # - Directory listing
        dir-listing.activate = "disable"
        # - Cache control and security headers for static resources
        $HTTP["url"] =~ "^/nextcloud/.+[^/]\.(css|js|woff2?|svg|gif|map)$" {
                setenv.add-response-header += (
                        "Cache-Control" => "public, max-age=15778463",
                        "Referrer-Policy" => "no-referrer",
                        "X-Content-Type-Options" => "nosniff",
                        "X-Download-Options" => "noopen",
                        "X-Frame-Options" => "SAMEORIGIN",
                        "X-Permitted-Cross-Domain-Policies" => "none",
                        "X-Robots-Tag" => "none",
                        "X-XSS-Protection" => "1; mode=block",
                )
        }

}
/etc/lighttpd/conf-enabled/99-unconfigured.conf:
# override prior index-file.name directive
# to fall back to default index.lighttpd.html
index-file.names := ( "index.php", "index.html", "index.lighttpd.html" )

</details>

MichaIng, 2021-04-06:
```
/etc/lighttpd/conf-enabled/50-phpmyadmin.conf:
cat: /etc/lighttpd/conf-enabled/50-phpmyadmin.conf: No such file or directory
```
There is something, but it cannot be removed? What is this?
```
ls -l /etc/lighttpd/conf-enabled/
ls -l /etc/lighttpd/conf-enabled/50-phpmyadmin.conf
rm -R /etc/lighttpd/conf-enabled/50-phpmyadmin.conf
```

ZiggyPi, 2021-04-06:
@MichaIng 

<details>

<summary> log </summary>

```
root@DietPi-HTPC:~# ls -l /etc/lighttpd/conf-enabled/
total 0
lrwxrwxrwx 1 root root 30 Dec 24  2019 05-auth.conf -> ../conf-available/05-auth.conf
lrwxrwxrwx 1 root root 33 Dec 24  2019 10-fastcgi.conf -> ../conf-available/10-fastcgi.conf
lrwxrwxrwx 1 root root 33 Feb 16  2020 10-rewrite.conf -> ../conf-available/10-rewrite.conf
lrwxrwxrwx 1 root root 37 Dec 24  2019 15-fastcgi-php.conf -> ../conf-available/15-fastcgi-php.conf
lrwxrwxrwx 1 root root 36 Dec 24  2019 50-phpmyadmin.conf -> ../conf-available/50-phpmyadmin.conf
lrwxrwxrwx 1 root root 43 Jan  3  2020 98-dietpi-phpmyadmin.conf -> ../conf-available/98-dietpi-phpmyadmin.conf
lrwxrwxrwx 1 root root 45 Feb 16  2020 99-dietpi-dav_redirect.conf -> ../conf-available/99-dietpi-dav_redirect.conf
lrwxrwxrwx 1 root root 42 Feb 16  2020 99-dietpi-nextcloud.conf -> ../conf-available/99-dietpi-nextcloud.conf
lrwxrwxrwx 1 root root 38 Dec 24  2019 99-unconfigured.conf -> ../conf-available/99-unconfigured.conf
root@DietPi-HTPC:~# ls -l /etc/lighttpd/conf-enabled/50-phpmyadmin.conf
lrwxrwxrwx 1 root root 36 Dec 24  2019 /etc/lighttpd/conf-enabled/50-phpmyadmin.conf -> ../conf-available/50-phpmyadmin.conf
root@DietPi-HTPC:~# rm -R /etc/lighttpd/conf-enabled/50-phpmyadmin.conf

```
</details>

MichaIng, 2021-04-06:
So now it was removed it seems. The original file can be removed as well:
```
rm /etc/lighttpd/conf-available/50-phpmyadmin.conf
```
Now try again to restart Lighttpd:
```
systemctl restart lighttpd
sleep 1
systemctl status lighttpd
```

ZiggyPi, 2021-04-06:
@MichaIng 

<details>

<summary>log</summary>

```
root@DietPi-HTPC:~# rm /etc/lighttpd/conf-available/50-phpmyadmin.conf
rm: cannot remove '/etc/lighttpd/conf-available/50-phpmyadmin.conf': No such file or directory
root@DietPi-HTPC:~# systemctl restart lighttpd
root@DietPi-HTPC:~# sleep 1
root@DietPi-HTPC:~# systemctl status lighttpd
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-04-06 22:43:39 CEST;
224ms ago
  Process: 9187 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
 Main PID: 9192 (lighttpd)
    Tasks: 3 (limit: 4915)
   Memory: 1.6M
   CGroup: /system.slice/lighttpd.service
           ├─9192 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
           ├─9195 /bin/sh -c /usr/share/lighttpd/create-mime.conf.pl
           └─9196 /usr/bin/perl -w /usr/share/lighttpd/create-mime.conf.pl

Apr 06 22:43:39 DietPi-HTPC systemd[1]: Starting Lighttpd Daemon...
Apr 06 22:43:39 DietPi-HTPC lighttpd[9187]: 2021-04-06 22:43:39: (server.c.1493) WARNING: unknown config-key: setenv.add-response-header (ignored)
Apr 06 22:43:39 DietPi-HTPC systemd[1]: Started Lighttpd Daemon.
```
</details>

Joulinar, 2021-04-06:
looks like web server Lighttpd is running again

MichaIng, 2021-04-07:
One thing to have some security headers effective for Nextcloud, please run:
```
G_CONFIG_INJECT '"mod_setenv",' '	"mod_setenv",' /etc/lighttpd/lighttpd.conf '"mod_.+",'
```
Not sure why it's missing, since we add this actively during install. Sadly Lighttpd has problems when mods are added multiple times, else we could add it to the Nextcloud drop-in config directly.
___________
Let's check the Redis server, which is required to run Nextcloud:
```
journalctl -u redis-server
```

Joulinar, 2021-04-07:
If you like to use Revers Proxy it might be better to go with Nginx instead of Lighttpd. But this would require a reinstallation of Nextcloud along with Nginx

ZiggyPi, 2021-04-07:
I almost would think that een complete reinstall would be better. But it has been a lot of work already to get radarr sonarr jackett deluge and samba runninf with an external mounted hdd. And on a headless rpi4

MichaIng, 2021-04-07:
> Nothing happens with the config injection

Jep that is fine. It is nothing visible, just some HTTP response headers for hardening. Although it seems you copied only half of the line?

> -- No entries --

Try to restart the Redis server to produce fresh logs:
```
systemctl restart redis-server
sleep 1
journalctl -u redis-server
```

> I almost would think that een complete reinstall would be better.

Possible. Hard to say what would take longer. It also depends a bit on the personal learning preference 😉. Debugging and fixing each service implies learning some basics around debugging, the services and their config files. A fresh install could be used as encouragement to create text files or even better scripts with the commands/steps you did to setup the system. Ideally these scripts can stored on a USB stick or on a Git repository, from where they can be executed even automatically on first boot of the fresh DietPi system via `/boot/dietpi.txt` `AUTO_SETUP_CUSTOM_SCRIPT_EXEC` option 🙂.

ZiggyPi, 2021-04-07:
@MichaIng 

Yeah i rather not do a fresh install.

Reinstalling would mean doing all the tutorials again because it has been such a long time okay.

So lets see what this debugging gets us.
<details>

<summary>log</summary>

```
root@DietPi-HTPC:~# systemctl restart redis-server
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
root@DietPi-HTPC:~# sleep 1
root@DietPi-HTPC:~# journalctl -u redis-server
-- Logs begin at Tue 2021-04-06 23:11:55 CEST, end at Wed 2021-04-07 22:21:10 CEST. --
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:04 DietPi-HTPC redis-server[24403]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 1.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:04 DietPi-HTPC redis-server[24408]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 2.
```
</details>

I already learned the propper way to post in the comments so that's a good start 😁
```

MichaIng, 2021-04-07:
It cannot find a certain required library. Strangely the related DEB package is a hard dependency of the [redis-tools](https://packages.debian.org/buster/redis-tools) package and should hence be installed in any case. But let's see if it is really missing and as a generic fix reinstall the package.
```
ls -l /usr/lib/arm-linux-gnueabihf/libjemalloc.so.2
apt install --reinstall libjemalloc2
systemctl restart redis-server
sleep 1
journalctl -u redis-server
```

ZiggyPi, 2021-04-07:
@MichaIng 

<details>

<summary>log</summary>

```
root@DietPi-HTPC:~# ls -l /usr/lib/arm-linux-gnueabihf/libjemalloc.so.2
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libjemalloc.so.2': No such file or directory
root@DietPi-HTPC:~# apt install --reinstall libjemalloc2
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  libjemalloc2
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 169 kB of archives.
After this operation, 472 kB of additional disk space will be used.
Get:1 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf libjemalloc2 armhf 5.1.0-3 [169 kB]
Fetched 169 kB in 0s (370 kB/s)
dpkg: unrecoverable fatal error, aborting:
 unknown system group 'ssl-cert' in statoverride file; the system group got removed
before the override, which is most probably a packaging bug, to recover youcan remove the override manually with dpkg-statoverride
E: Sub-process /usr/bin/dpkg returned an error code (2)
```
</details>

ZiggyPi, 2021-04-08:
@MichaIng 

I have absolutely no idea what version i installed. I bought rpi4 about 3 or 4 years ago and downloaded dietpi from the website and did een fresh install and installed the software via dietpi-software. 

And run apt-get update && apt-get upgrade once in a while.

```
root@DietPi-HTPC:~# dpkg-statoverride --list
root crontab 2755 /usr/bin/crontab
root root 1733 /var/lib/php/sessions
root ssl-cert 710 /etc/ssl/private
debian-deluged debian-deluged 750 /var/lib/deluged/config
debian-deluged adm 2750 /var/log/deluged
root messagebus 4754 /usr/lib/dbus-1.0/dbus-daemon-launch-helper
debian-deluged debian-deluged 750 /var/lib/deluged
redis redis 640 /etc/redis/redis.conf
root@DietPi-HTPC:~# getent group ssl-cert
```

Joulinar, 2021-04-08:
3-4 years could be Stretch that time. Maybe thinks get messed up during Buster upgrade?

MichaIng, 2021-04-08:
Probably. Is the RPi 4 already available for 3-4 years? However, lets fix the statoverride:
```
dpkg-statoverride --remove /etc/ssl/private
apt install --reinstall libjemalloc2
systemctl restart redis-server
sleep 1
journalctl -u redis-server
```

ZiggyPi, 2021-04-08:
@MichaIng 

I checked my receipt and i bought it december 2019.

<details>

<summary>log</summary>

```
root@DietPi-HTPC:~# dpkg-statoverride --remove /etc/ssl/private
root@DietPi-HTPC:~# apt install --reinstall libjemalloc2
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  libjemalloc2
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 169 kB of archives.
After this operation, 472 kB of additional disk space will be used.
Get:1 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf libjemalloc2 armhf 5.1.0-3 [169 kB]
Fetched 169 kB in 0s (405 kB/s)
Selecting previously unselected package libjemalloc2:armhf.
(Reading database ... 84119 files and directories currently installed.)
Preparing to unpack .../libjemalloc2_5.1.0-3_armhf.deb ...
Unpacking libjemalloc2:armhf (5.1.0-3) ...
Setting up libjemalloc2:armhf (5.1.0-3) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi-HTPC:~# systemctl restart redis-server
root@DietPi-HTPC:~# sleep 1
root@DietPi-HTPC:~# journalctl -u redis-server
-- Logs begin at Wed 2021-04-07 20:36:08 CEST, end at Thu 2021-04-08 19:40:39 CEST. --
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:04 DietPi-HTPC redis-server[24403]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 1.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:04 DietPi-HTPC redis-server[24408]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:04 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:04 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 2.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:05 DietPi-HTPC redis-server[24415]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 3.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:05 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:05 DietPi-HTPC redis-server[24419]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:05 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
[... 2 lines cut ...]
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 4.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 07 22:21:06 DietPi-HTPC redis-server[24422]: /usr/bin/redis-server: error while loading shared libraries: libjemalloc.so.2: cannot open shared object file: No such file or directory
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-serve
r.service: Control process exited, code=exited, status=127/n/a
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 5.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: Stopped Advanced key-value store.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-serve
[... cut ...]
Apr 07 22:21:06 DietPi-HTPC systemd[1]: redis-serve
r.service: Failed with result 'exit-code'.
Apr 07 22:21:06 DietPi-HTPC systemd[1]: Failed to s
tart Advanced key-value store.
Apr 08 19:40:35 DietPi-HTPC systemd[1]: Starting Advanced key-value store...
Apr 08 19:40:35 DietPi-HTPC systemd[1]: redis-server.service: Can't open PID file /run/redis/redis-server.pid (yet?) after start: No such file or directory
Apr 08 19:40:35 DietPi-HTPC systemd[1]: Started Advanced key-value store.
```
</details>

MichaIng, 2021-04-08:
And I just remember that Raspbian Stretch never worked on RPi 4. The release of RPi 4 was the reason for the early release of Raspberry Pi OS Buster, prior to Debian Buster.

Okay, Redis is running now. Let's go on with Coturn:
```
systemctl restart coturn
sleep 1
journalctl -u coturn
```

ZiggyPi, 2021-04-08:
@MichaIng 

<details>

<summary> log </summary>

```
root@DietPi-HTPC:~# systemctl restart coturn
Job for coturn.service failed because the control process exited with error code.
See "systemctl status coturn.service" and "journalctl -xe" for details.
root@DietPi-HTPC:~# sleep 1
root@DietPi-HTPC:~# journalctl -u coturn
-- Logs begin at Wed 2021-04-07 23:15:01 CEST, end at Thu 2021-04-08 20:34:34 CEST. --
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:30 DietPi-HTPC systemd[24050]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:30 DietPi-HTPC systemd[24050]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process

Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 1.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:30 DietPi-HTPC systemd[24055]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:30 DietPi-HTPC systemd[24055]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process

Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 2.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:31 DietPi-HTPC systemd[24065]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:31 DietPi-HTPC systemd[24065]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process
0m
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 3.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:31 DietPi-HTPC systemd[24068]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:31 DietPi-HTPC systemd[24068]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process
0m
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 4.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:31 DietPi-HTPC systemd[24071]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:31 DietPi-HTPC systemd[24071]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process
0m
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 5.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Start request repeated too quickly.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
```
</details>

MichaIng, 2021-04-08:
> coturn.service: Failed to determine user credentials: No such process

This seems to happen when the user does not exist. Let's see if a reinstall restores it:
```
apt install --reinstall coturn
systemctl restart coturn
sleep 1
journalctl -u coturn
```

ZiggyPi, 2021-04-08:
@MichaIng 

I am still seeing the same error

<details>

<summary>Log</summary>

```

root@DietPi-HTPC:~# apt install --reinstall coturn
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 303 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf coturn armhf 4.5.1.1-1.1+deb10u2 [303 kB]
Fetched 303 kB in 0s (692 kB/s)
(Reading database ... 84125 files and directories currently installed.)
Preparing to unpack .../coturn_4.5.1.1-1.1+deb10u2_armhf.deb ...
Unpacking coturn (4.5.1.1-1.1+deb10u2) over (4.5.1.1-1.1+deb10u2) ...
Setting up coturn (4.5.1.1-1.1+deb10u2) ...
Adding group `turnserver' (GID 117) ...
Done.
Adding system user `turnserver' (UID 111) ...
Adding new user `turnserver' (UID 111) with group `turnserver' ...
Not creating home directory `/'.
Processing triggers for systemd (241-7~deb10u6+rpi1) ...
root@DietPi-HTPC:~# systemctl restart coturn
root@DietPi-HTPC:~# sleep 1
root@DietPi-HTPC:~# journalctl -u coturn
-- Logs begin at Wed 2021-04-07 23:15:01 CEST, end at Thu 2021-04-08 21:21:01 CEST. --
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:30 DietPi-HTPC systemd[24050]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:30 DietPi-HTPC systemd[24050]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process

Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 1.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:30 DietPi-HTPC systemd[24055]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:30 DietPi-HTPC systemd[24055]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process

Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:30 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
Apr 08 20:34:30 DietPi-HTPC systemd[1]: Failed to s
tart coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Service RestartSec=100ms expired, scheduling restart.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.service: Scheduled restart job, restart counter is at 2.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Stopped coTURN STUN/TURN Server.
Apr 08 20:34:31 DietPi-HTPC systemd[1]: Starting coTURN STUN/TURN Server...Apr 08 20:34:31 DietPi-HTPC systemd[24065]: coturn.
service: Failed to determine user credentials: No such process
Apr 08 20:34:31 DietPi-HTPC systemd[24065]: coturn.
service: Failed at step USER spawning /usr/bin/turnserver: No such process
0m
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Control process exited, code=exited, status=217/USER
Apr 08 20:34:31 DietPi-HTPC systemd[1]: coturn.serv
ice: Failed with result 'exit-code'.
[... 105 lines cut ...]
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (main listener thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: Wait for relay ports initialization...
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay 10.0.0.30 initialization...
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay 10.0.0.30 initialization done
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay 172.17.0.1 initialization...
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay 172.17.0.1 initialization done
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay ::1 initialization...
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0:   relay ::1 initialization done
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: Relay ports initialization done
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (general relay thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (general relay thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (general relay thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: turn server id=1 created
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: turn server id=0 created
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: turn server id=2 created
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (general relay thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: turn server id=3 created
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: Total General servers: 4
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (auth thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (auth thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: IO method (admin thread): epoll (with changelist)
Apr 08 21:20:57 DietPi-HTPC turnserver[824]: 0: SQLite DB connection success: /var/lib/turn/turndb
Apr 08 21:20:59 DietPi-HTPC systemd[1]: Started coTURN STUN/TURN Server.
```
</details>

ZiggyPi, 2021-04-08:
@Joulinar 

Okay nice.
So 3 down 5 to go? 🙈

ZiggyPi, 2021-04-08:
@MichaIng 

Will i loose my sonarr and radarr data if i do a reinstall?

ZiggyPi, 2021-04-08:
@MichaIng 

<details>

<summary>log of reinstalls</summary>

```
root@DietPi-HTPC:~# systemctl restart xrdp
Job for xrdp.service failed because the control process exited with error cRPi 4 Model B (armv7l) | IP: 10.0.0.30                                     ┌───────────────────────────┤ DietPi-Software ├───────────────────────────┐│                                                                         ││ Would you like to create (or update) a "DietPi-Backup" of the system,   ││ before proceeding?                                                      ││                                                                         ││ "DietPi-Backup" creates a system restore point, which can be recovered  ││ if unexpected issues occur.                                             ││                                                                         ││ For more information on "DietPi-Backup", please use the link below:     ││  - https://dietpi.com/docs/dietpi_tools/#dietpi-backup-backuprestore    ││                                                                         ││                    <Ok>                        <Cancel>                 ││                                                                         │└─────────────────────────────────────────────────────────────────────────┘[  OK  ] DietPi-Software | Free space check: path=/ | available=31904 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : nmbd
[  OK  ] DietPi-Services | unmask : smbd
[  OK  ] DietPi-Services | unmask : nfs-kernel-server
[  OK  ] DietPi-Services | unmask : redis-server
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.3-fpm
[  OK  ] DietPi-Services | unmask : nginx
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : coturn
[  OK  ] DietPi-Services | unmask : deluged
[  OK  ] DietPi-Services | unmask : plexmediaserver
[  OK  ] DietPi-Services | unmask : ubooquity
[  OK  ] DietPi-Services | unmask : jackett
[  OK  ] DietPi-Services | unmask : sonarr
[  OK  ] DietPi-Services | unmask : radarr
[  OK  ] DietPi-Services | unmask : lidarr
[  OK  ] DietPi-Services | unmask : deluge-web
[  OK  ] DietPi-Services | unmask : rpimonitor
[  OK  ] DietPi-Services | unmask : htpc-manager
[  OK  ] DietPi-Services | unmask : docker
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : htpc-manager
[  OK  ] DietPi-Services | stop : rpimonitor
[  OK  ] DietPi-Services | stop : deluge-web
[  OK  ] DietPi-Services | stop : lidarr
[  OK  ] DietPi-Services | stop : radarr
[  OK  ] DietPi-Services | stop : sonarr
[  OK  ] DietPi-Services | stop : jackett
[  OK  ] DietPi-Services | stop : ubooquity
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : deluged
[  OK  ] DietPi-Services | stop : coturn
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://apt.sonarr.tv develop InRelease
Hit:3 http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial InRelease
Get:4 https://download.docker.com/linux/raspbian buster InRelease [29.7 kB]Get:5 https://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:6 https://mirror.netcologne.de/raspbian/raspbian buster InRelease [15.0 kB]
Hit:7 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Get:8 https://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
Get:9 https://mirror.netcologne.de/raspbian/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.5 MB in 11s (1205 kB/s)
Reading package lists...
[... 168 lines cut ...]
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : nfs-kernel-server
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.3-fpm
[  OK  ] DietPi-Services | restart : nginx
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : coturn
[  OK  ] DietPi-Services | restart : deluged
[  OK  ] DietPi-Services | restart : plexmediaserver
[  OK  ] DietPi-Services | restart : ubooquity
[  OK  ] DietPi-Services | restart : jackett
[  OK  ] DietPi-Services | restart : sonarr
[  OK  ] DietPi-Services | restart : radarr
[  OK  ] DietPi-Services | restart : lidarr
[  OK  ] DietPi-Services | restart : deluge-web
[  OK  ] DietPi-Services | restart : rpimonitor
[  OK  ] DietPi-Services | restart : htpc-manager
[  OK  ] DietPi-Services | restart : docker
[  OK  ] DietPi-Services | restart : cron
[FAILED] DietPi-Services | ncc maintenance:mode --off
```
</details>

MichaIng, 2021-04-08:
So far so good. Lets see which ones are working now and how the XRDP logs say:
```
dietpi-servives status
journalctl -u xrdp
```

Joulinar, 2021-04-08:
somehow NextCloud is still failing and stuck in maintenance mode. Maybe a reboot would be good to see what is missing still and if everything is surviving 😃

MichaIng, 2021-04-08:
So somehow many service users have gone, same for XRDP and likely Sonarr:
```
apt install --reinstall nzbdrone xrdp
usermod -aG redis www-data # this probably fixes Nextcloud
journalctl -u lighttpd -u lidarr
```

Joulinar, 2021-04-08:
hmm there seems to be something running on port 80 (http) already

`Apr 08 23:32:34 DietPi-HTPC lighttpd[1578]: 2021-04-08 23:32:33: (network.c.313) can't bind to socket: 0.0.0.0:80 Address already in us`

pls can you have a look `ss -tulpn | grep LISTEN`

ZiggyPi, 2021-04-08:
@Joulinar 

<details>

<summary>listen log</summary>

```
root@DietPi-HTPC:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      128                      127.0.0.1:32600       0.0.0.0:*     users:(("Plex Tuner Serv",pid=2068,fd=13))                           
tcp   LISTEN 0      5                          0.0.0.0:8888        0.0.0.0:*     users:(("rpimonitord",pid=1602,fd=5))                                
tcp   LISTEN 0      128                        0.0.0.0:443         0.0.0.0:*     users:(("nginx",pid=1389,fd=8),("nginx",pid=1387,fd=8),("nginx",pid=1386,fd=8),("nginx",pid=1385,fd=8),("nginx",pid=1384,fd=8))
tcp   LISTEN 0      64                         0.0.0.0:44763       0.0.0.0:*                                                                          
tcp   LISTEN 0      128                        0.0.0.0:58939       0.0.0.0:*     users:(("rpc.mountd",pid=872,fd=17))                                 
tcp   LISTEN 0      50                         0.0.0.0:445         0.0.0.0:*     users:(("smbd",pid=810,fd=30))                                       
tcp   LISTEN 0      50                         0.0.0.0:58846       0.0.0.0:*     users:(("deluged",pid=1483,fd=16))                                   
tcp   LISTEN 0      5                          0.0.0.0:6881        0.0.0.0:*     users:(("deluged",pid=1483,fd=13))                                   
tcp   LISTEN 0      64                         0.0.0.0:2049        0.0.0.0:*                                                                          
tcp   LISTEN 0      128                      127.0.0.1:41545       0.0.0.0:*     users:(("Plex Script Hos",pid=1681,fd=7))                            
tcp   LISTEN 0      80                       127.0.0.1:3306        0.0.0.0:*     users:(("mysqld",pid=1076,fd=23))                                    
tcp   LISTEN 0      128                      127.0.0.1:6379        0.0.0.0:*     users:(("redis-server",pid=950,fd=6))                                
tcp   LISTEN 0      50                         0.0.0.0:139         0.0.0.0:*     users:(("smbd",pid=810,fd=31))                                       
tcp   LISTEN 0      5                          0.0.0.0:5900        0.0.0.0:*     users:(("vncserver-x11-c",pid=628,fd=11))                            
tcp   LISTEN 0      5                          0.0.0.0:5901        0.0.0.0:*     users:(("Xtigervnc",pid=725,fd=7))                                   
tcp   LISTEN 0      5                          0.0.0.0:5999        0.0.0.0:*     users:(("vncserver-virtu",pid=620,fd=7))                             
tcp   LISTEN 0      128                        0.0.0.0:111         0.0.0.0:*     users:(("rpcbind",pid=292,fd=4),("systemd",pid=1,fd=73))             
tcp   LISTEN 0      50                         0.0.0.0:8112        0.0.0.0:*     users:(("deluge-web",pid=1524,fd=5))                                 
tcp   LISTEN 0      128                        0.0.0.0:80          0.0.0.0:*     users:(("nginx",pid=1389,fd=5),("nginx",pid=1387,fd=5),("nginx",pid=1386,fd=5),("nginx",pid=1385,fd=5),("nginx",pid=1384,fd=5))
tcp   LISTEN 0      128                      127.0.0.1:32401       0.0.0.0:*     users:(("Plex Media Serv",pid=1500,fd=61))                           
tcp   LISTEN 0      128                        0.0.0.0:47987       0.0.0.0:*     users:(("rpc.mountd",pid=872,fd=9))                                  
tcp   LISTEN 0      10                         0.0.0.0:8085        0.0.0.0:*     users:(("python",pid=1604,fd=5))                                     
tcp   LISTEN 0      128                      10.0.0.30:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=87))                                
tcp   LISTEN 0      128                      127.0.0.1:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=85))                                
tcp   LISTEN 0      128                      10.0.0.30:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=73))                                
tcp   LISTEN 0      128                      10.0.0.30:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=72))                                
tcp   LISTEN 0      128                      10.0.0.30:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=71))                                
tcp   LISTEN 0      128                      127.0.0.1:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=65))                                
tcp   LISTEN 0      128                      127.0.0.1:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=38))                                
tcp   LISTEN 0      128                      127.0.0.1:3478        0.0.0.0:*     users:(("turnserver",pid=1423,fd=43))                                
tcp   LISTEN 0      128                        0.0.0.0:22          0.0.0.0:*     users:(("dropbear",pid=705,fd=3))                                    
tcp   LISTEN 0      128                      10.0.0.30:3479        0.0.0.0:*     users:(("turnserver",pid=1423,fd=88))                                
tcp   LISTEN 0      128                      127.0.0.1:3479        0.0.0.0:*     users:(("turnserver",pid=1423,fd=86))                                
tcp   LISTEN 0      128                      10.0.0.30:3479        0.0.0.0:*     users:(("turnserver",pid=1423,fd=77))                                
[... cut ...]
tcp   LISTEN 0      128                          [::1]:3479           [::]:*     users:(("turnserver",pid=1423,fd=84))                                
tcp   LISTEN 0      128                          [::1]:3479           [::]:*     users:(("turnserver",pid=1423,fd=83))                                
tcp   LISTEN 0      128                          [::1]:3479           [::]:*     users:(("turnserver",pid=1423,fd=41))                                
tcp   LISTEN 0      128                          [::1]:3479           [::]:*     users:(("turnserver",pid=1423,fd=90))

```
</details>

Joulinar, 2021-04-08:
you have installed Nginx along with Lighttpd. Means it's 2 web server now. This of course is not going to work.

`tcp   LISTEN 0      128                        0.0.0.0:80          0.0.0.0:*     users:(("nginx",pid=1389,fd=5),("nginx",pid=1387,fd=5),("nginx",pid=1386,fd=5),("nginx",pid=1385,fd=5),("nginx",pid=1384,fd=5))`

any particular reason for this?

ZiggyPi, 2021-04-08:
@Joulinar 
I installed ngix brcaise i founda tutorial on setting up reverse proxy for htpc manager so i could reach sonarr and radarr from outside the network.

[26 more replies]

closed 2021-04-24 by MichaIng