MichaIng/DietPi #4244: Error trying to install transmission torrent software

pioniere, 2021-04-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...
#### Details:
- Date           | Sun Apr  4 15:56:12 PDT 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install transmission-daemon
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up transmission-daemon (2.94-2+deb10u2) ...
chown: invalid group: ‘debian-transmission:debian-transmission’
dpkg: error processing package transmission-daemon (--configure):
 installed transmission-daemon package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 transmission-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2021-04-04:
Hi,

many thanks for your message. Hmm DietPi is doing nothing else than installing `transmission-daemon` via `apt package` from global repository. I did a test installation on my RPi3B+ 32bit without issues. Can you try to setup again pls. Maybe do a reboot before.

pioniere, 2021-04-05:
I did try rebooting and ran setup again, same result.

Joulinar, 2021-04-05:
strange, usually `debian-transmission` user and group should be created by the Debian package. Can you following to files if a user/group `debian-transmission` is existing 

```
cat /etc/passwd
cat /etc/group
```

Found an older issue on reddit. But strange that it is working for me
https://www.reddit.com/r/linuxquestions/comments/7zxtbm/chown_invalid_group/

MichaIng, 2021-04-29:
I mark this as closed. Feel free to reopen if required.

LeVraiRoiDHyrule, 2021-06-06:
Hi, I'm having the exact same issue

[image]

I found no way to correctly install transmission.

Joulinar, 2021-06-06:
Hi,

There is no real error message shown on your screen print. Pls can you run following

```
apt update && apt upgrade
apt install transmission-daemon
```

Plus post the full output

LeVraiRoiDHyrule, 2021-06-06:
When running the apt upgrade, here is what I get:

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
Atteint :3 https://download.docker.com/linux/raspbian buster InRelease
Atteint :4 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Atteint :5 https://apt.ombi.app/develop jessie InRelease
Atteint :6 https://archive.raspberrypi.org/debian buster InRelease
Atteint :7 https://apt.hrfee.dev trusty InRelease
Atteint :8 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Atteint :9 https://repo.jellyfin.org/debian buster InRelease
Ign :10 https://download.webmin.com/download/repository sarge InRelease
Atteint :11 https://download.webmin.com/download/repository sarge Release
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
Tous les paquets sont à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
1 partiellement installés ou enlevés.
Après cette opération, 0 o d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] O
Paramétrage de transmission-daemon (2.94-2+deb10u2) ...
chown: groupe incorrect: « debian-transmission:debian-transmission »
dpkg: erreur de traitement du paquet transmission-daemon (--configure) :
 installed transmission-daemon package post-installation script subprocess returned error exit status 1
Des erreurs ont été rencontrées pendant l'exécution :
 transmission-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~#

Apparently there is some rests of the transmission installation and the command doesn't succeed to remove them.

MichaIng, 2021-06-06:
The `postinst` script fails. Can you please run the following:
```
dash -x /var/lib/dpkg/info/transmission-daemon.postinst configure
```

LeVraiRoiDHyrule, 2021-06-06:
Thanks for your answer. Here is what I get:

`root@DietPi:~# dash -x /var/lib/dpkg/info/transmission-daemon.postinst configure
+ set -e
+ test
+ getent passwd debian-transmission
+ chperms debian-transmission debian-transmission 4750 /var/lib/transmission-daemon/.config/transmission-daemon
+ dpkg-statoverride --list /var/lib/transmission-daemon/.config/transmission-daemon
+ OVERRIDE=
+ [ -e /var/lib/transmission-daemon/.config/transmission-daemon ]
+ chown debian-transmission:debian-transmission /var/lib/transmission-daemon/.config/transmission-daemon
chown: groupe incorrect: « debian-transmission:debian-transmission »
root@DietPi:~#`

Joulinar, 2021-06-06:
In general there is no need to post the issue twice. This will not speed up things as the same people are on both platform.

MichaIng, 2021-06-06:
Hmm, do we remove that group? Let me see. We don't use it anymore, but it shouldn't be removed.

LeVraiRoiDHyrule, 2021-06-06:
> 
> 
> Hmm, do we remove that group? Let me see. We don't use it anymore, but it shouldn't be removed.

I have no group called "debian-transmission". But I have a user named like that, do you want me to remove it and try again ?

LeVraiRoiDHyrule, 2021-06-06:
> 
> 
> In general there is no need to post the issue twice. This will not speed up things as the same people are on both platform.

I'm very sorry, I won't do that again next time

LeVraiRoiDHyrule, 2021-06-06:
> getent group debian-transmission

Done ! (it didn't return anything)

MichaIng, 2021-06-06:
Nonono don't remove that user 😄. The opposite: Create the group.
```
groupadd -r debian-transmission
usermod -aG debian-transmission debian-transmission
```

LeVraiRoiDHyrule, 2021-06-06:
It's done. I try again ? With dietpi-software or just with apt install ?

MichaIng, 2021-06-06:
Just to be sure it's really solved, finish the install first:
```
apt -f install
```

LeVraiRoiDHyrule, 2021-06-06:
The install completed without issues.
But I can't access [localip]:9091, I've got a 403 forbidden error.

MichaIng, 2021-06-06:
Did you rerun `dietpi-software` as well?

LeVraiRoiDHyrule, 2021-06-06:
I didn't. But now I did, and it works ! Transmission seems to work perfectly, Thank you !

I think the problem came from my trial at creating a new group for permissions. I have my download folder where both transmission, radarr and sonarr should all be able to read and write, while others can only read. I created a new group called downloadaccess including debian-transmission, radarr and sonarr users. 
Is it the correct way ? I probably failed something

Joulinar, 2021-06-06:
Usually we use user group `dietpi` for shared access between multiple apps

MichaIng, 2021-06-06:
> Usually we use user group `dietpi` for shared access between multiple apps

Exactly. I mean there is not issue to create a new group in general to further split permissions, but as in the apparent case, some package installs have issues when their system groups are removed. When the whole user is removed, it is re-created, but it is then often assumed, that when the user exists, that the group exists as well, and then `chown` commands can fail.

LeVraiRoiDHyrule, 2021-06-06:
> Exactly. I mean there is not issue to create a new group in general to further split permissions, but as in the apparent case, some package installs have issues when their system groups are removed. [...]

Ah I see. The thing is that I would like to have some custom groups, one for videoaccess (that will contain jellyfin, sonarr, radarr, ombi but without write rights) and one for download access only for transmission, sonarr and radarr with write access

Joulinar, 2021-07-14:
another report on the forum https://dietpi.com/phpbb/viewtopic.php?f=11&t=9227

@MichaIng 
should we created the group in advance?

MichaIng, 2021-07-14:
I guess it happens on reinstalls only as we removed the group once, and the postinst script checks for the user only, and then expects the group to exist. So this is something that we should add to update patches instead. I'll verify and open a PR then.

MichaIng, 2021-07-24:
Done with: https://github.com/MichaIng/DietPi/commit/14998ff875b5e0dcd6981d4def8d73b7e1c861d1

closed 2021-07-24 by MichaIng