MichaIng/DietPi #4245: META | DietPi Mastodon Account

ravenclaw900, 2021-04-05:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Not really, but mostly personal preference, as Mastodon is open source etc.

#### Describe the solution you'd like:

- A DietPi Mastodon account (probably on https://fosstodon.org).

#### Describe alternatives you've considered:

- Same thing, but just a bot that mirrors the Twitter account.

MichaIng, 2021-04-05:
Can such a mirror bot be created easily? Major concern about joining new platforms is that one needs to maintain them. Mirroring tweets would at least enable some basic activity, if then one of us finds time to respond to users by times, that would be good.

Since it's decentralised, we could run a little node as well, right?

ravenclaw900, 2021-04-05:
https://github.com/yogthos/mastodon-bot looks like one that could work.  And yes, we could run a DietPi Mastodon node.

ravenclaw900, 2021-05-26:
#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/+229