MichaIng/DietPi #4248: "Block non-LAN access to Pi-hole admin panel" won't work with non standard local networks

zell-mbc, 2021-04-05:
I just went through a fresh install of Pihole on DietPi and was getting the Pihole Dashboard 403 Forbidden error mentioned in other posts here.
After a couple or re-installs and reconfigurations I checked the lighttpd settings and noticed that the script blocking external access to the Pihole Admin Interface expects local to be an 192.y.z style address. Well, not the case for my install. Changing the regex to fit my network solved the issue, but maybe a better way to handle this would be if the install script would check for the actual IP of the host rather than assuming it's going to be 192.168...?

` $HTTP["remoteip"] !~ "^1(27|92\.168|0|72\.(1[6-9]|2[0-9]|3[0-1]))\." {
                url.access-deny = ("")
        }
`

Not an issue for me any longer and, once one figured out what's going on, the fix is easy,  but I thought I'd make you aware.

MichaIng, 2021-04-05:
Many thanks for your report.

1. During install, you are asked whether you want to have WAN access blocked or not.
2. The regex we use matches the one you posted, if this is the one you use now? https://github.com/MichaIng/DietPi/blob/master/.conf/dps_93/lighttpd.block_public_admin.conf#L3
    So it does not expect `192.168...` but allows all existing IP ranges that are reserved for LANs, hence are not used in WAN.

Can you verify `/etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.on` containing the above regex and can you show the particular IP range you use in your LAN?

We can extend that prompt to explicitly show the IP ranges that are allowed, and add another line that makes clear that all other IP ranges are used for WAN hosts and that it is hence not recommended and potentially dangerous to use them in your LAN 😉.

zell-mbc, 2021-04-05:
1. Indeed, and I wanted WAN acces to be blocked, just didn't realise at first that the block also blocked local access because of the network mismatch
2. The one above is the original one, I changed the regex to 
`$HTTP["remoteip"] !~ "^1(27|75\.0|72\.(1[6-9]|2[0-9]|3[0-1]))\." {`
 (75 instead of 92) so it works with my 175.0.. network.

I know that my 175.0 network is not standard, and not recommended as someone out there will be the real owner of this network but hey, it's been with me for 20+ years and changing all devices would be pretty painful :-) 

> We can extend that prompt to explicitly show the IP ranges that are allowed,

Yes, I think that would have helped in my case, just not sure how much of a niche case my setup is and if it warrants a change for everyone.

MichaIng, 2021-04-05:
Okay, makes all sense then. Let's add the info. I remember at least a second case from the forum, where I already thought about adding that info.

closed 2021-04-05 by zell-mbc

MichaIng, 2021-04-06:
Done: https://github.com/MichaIng/DietPi/commit/6e36347d63f6a7cf0416da1269c48e1d25fc2030