MichaIng/DietPi #4249: Cannot play music from Dac

Balmoral86, 2021-04-06:
Date           | Tue Apr  6 11:29:19 BST 2021
- Bug report     | d1b790e4-df3a-47c0-bcab-3d58a0ed2e4a
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | DietPi
- Hardware       | Sparky SBC (armv7l) (ID=70)
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | amixer -c 1 cset Playback 100%
- Exit code      | 1
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. Open `dietpi-config`
2. Select Audio options
3. Select Sound Card Usb-dac
#### Expected behaviour:

- Dac should play music
#### Actual behaviour:

- No sound from Dac, as if the sound was muted.
#### Extra details:

- ...
#### Additional logs:
```
Wrong control identifier: Playback
```

MichaIng, 2021-04-06:
Many thanks for your report.

Can you please paste the output of the following commands:
```
aplay -l
amixer -c 1 controls
amixer -c 1 cset playback 100%
```

Balmoral86, 2021-04-06:
Thank you for your answer.

[image]

MichaIng, 2021-04-06:
Okay, makes sense that our selector is broken, it was since its implementation with DietPi v6.8 but now creates a script termination the first time as we implemented the G_EXEC error handler wrapper more and more 😄.

Please do the following and re-select the DAC in dietpi-config:
```
sed -i '/cset/c\true' /boot/dietpi/func/dietpi-set_hardware
```

MichaIng, 2021-04-06:
Fix: https://github.com/MichaIng/DietPi/commit/744bd8ee62ca136570a954c9e511ad5c2a98cfd0
But could you run the following test:
```
while read -r line
do
G_EXEC amixer -c 1 cset "$line" 100%
done < <(amixer -c 1 controls | mawk -F, '/layback/{print $1}')
```
Other than numid 4 and 5 are actually no volume controls, so they might still fail when the value "100%" is seen invalid. I'll also test this with a few VMs. If so, I'll remove that part completely from the code, as I'm not sure how to reliably identify volume control that accept percentage values then. The code was btw originating from one of the Allo/Sparky SBC devs, to have DAC volumes maxed automatically, which otherwise might be low or even muted by default, i.e. one needs to change it manually via e.g. `alsamixer`.

Balmoral86, 2021-04-06:
> Okay, makes sense that our selector is broken, it was since its implementation with DietPi v6.8 but now creates a script termination the first time as we implemented the G_EXEC error handler wrapper [...]

Thanks a lot Michalng. It worked! My dac started playing music again 👍

Balmoral86, 2021-04-06:
> Fix: [744bd8e](https://github.com/MichaIng/DietPi/commit/744bd8ee62ca136570a954c9e511ad5c2a98cfd0)
> But could you run the following test:
> 
> ```
> while read -r line
> do
> G_EXEC amixer -c 1 cset "$line" 100%
> done < <(amixer -c 1 controls | mawk -F, '/layback/{print $1}')
> ```
> 
> Other than numid 4 and 5 are actually no volume controls, so they might still fail when the value "100%" is seen invalid. I'll also test this with a few VMs. If so, I'll remove that part completely from the code, as I'm not sure how to reliably identify volume control that accept percentage values then. The code was btw originating from one of the Allo/Sparky SBC devs, to have DAC volumes maxed automatically, which otherwise might be low or even muted by default, i.e. one needs to change it manually via e.g. `alsamixer`.

Ok. I ran the test you asked for. Here's what I got:

[image]

Joulinar, 2021-04-06:
Hi, there is no need to do screenshots, you can simply copy/ paste the output from your SSH client 😉

MichaIng, 2021-04-06:
Thanks. So while the loop works as intended now, it's indeed wrong to assume that all controls with the word "playback" on their name accept percentage values or are even volume controls. While we could set the volume on a per-driver basis, we cannot do it in a genetic way for all USB DAC. Also good to know that in your case it's not muted by default when it plays sound already.

MichaIng, 2021-04-06:
I removed these amixer calls now: https://github.com/MichaIng/DietPi/commit/87aa2ade222215fe16fa17c2526c6aecfcab7a16
If we find a generic way to detect volume controls and if they accept percentage values, we could re-add it correctly, but obviously it was not required as it never worked before.

closed 2021-04-24 by MichaIng