MichaIng/DietPi #4251: Single File PHP Gallery,

Alexgolshtein, 2021-04-07:
### Single File PHP Gallery doesn't work after install. 
HTTP://[host]/gallery/index.php  returns 403

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster  10.9
- Kernel version | Linux DietPi 5.10.21-sunxi64 #21.02.3 SMP Mon Mar 8 00:45:13 UTC 2021 aarch64 GNU/Linux
- SBC model | Orange Pi Prime
- Power supply used | 5V 2A
- Sony uhs3

#### Additional Information (if applicable)
- Software title | Single File PHP Gallery
- Fresh install
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | dfc8ca53-b4fe-4784-abfc-bea6b945e2dc

#### Steps to reproduce
Install WebMin,
Install Nextcloud
Install Syncthing
Uninstall Nextcloud
Install Single File PHP Gallery
/var/www/gallery   exists and contains data

#### Expected behaviour
HTTP://[host]/gallery Should work

#### Actual behaviour
HTTP://[host]/gallery   returns 403
HTTP://[host]/gallery/index.php  returns 403

Joulinar, 2021-04-07:
Hi,

many thanks for your message. Looks like file system permission are incorrect. Pls try following as workaround

`chown -r www-data:www-data /var/www/gallery/`

Alexgolshtein, 2021-04-07:
> Hi,
> 
> many thanks for your message. Looks like file system permission are incorrect. Pls try following as workaround
> 
> `chown -R www-data:www-data /var/www/gallery/`
This fixed the issue. Thank you!

MichaIng, 2021-04-07:
Reopening it until we merged the fix. I want to find out why it worked fine in the past with the permissions we had, but now not anymore. AFAIK the version was not updated since then.

Joulinar, 2021-04-07:
Gallery folder has r/w permission for `root` user only. No other permission for `groups` or `other`. Means, web server is not able to access the folder at all 😉

MichaIng, 2021-04-07:
All dirs are created via `mkdir -p` which should lead to default 755 modes, i.e. read access for the webserver. Only the `_sfpg_data` sub dir is `chown`ed to permit write access, which should be sufficient.

... lol found the issue: `7zr` creates the target parent directory with 700 mode, ignoring the default umask... needs to be reported to the Debian bug tracker. As a quick workaround, i'll pre-create the target dir for 7zr extraction, if it does not exist yet.

To verify, `chmod 755 /var/www/gallery` is sufficient to get the intended access and permissions.

Joulinar, 2021-04-07:
> To verify,` chmod 755 /var/www/gallery` is sufficient to get the intended access and permissions.

yep is working as 👍

MichaIng, 2021-04-08:
Reported to Debian bug tracker: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=986634

I tested the new official 7-Zip Linux binaries, which work fine and do not suffer from the issue: #4195

As I do not expect it to be fixed anymore, since p7zip is basically abandoned, creating own 7-Zip DEB packages seems to be a nice alternative. But not sure if this is a good idea as long as they're marked as alpha state.

closed 2021-04-24 by MichaIng