MichaIng/DietPi #4253: Failing to install Kodi on Diet-Pi (fresh install)

ashishsantikari, 2021-04-07:
#### Details:
- Date           | Wed Apr  7 21:00:29 CEST 2021
- Bug report     | 98e007cf-7b8e-4af0-a132-22ea8df558cb
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install alsa-utils
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. Search software kodi
2. Select 31 Kodi: the media center for Linux
2. Proceed with installation.
#### Expected behaviour:
Kodi should be installed
- ...
#### Actual behaviour:
Installation failed. Some required packages are missing.
- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://mirror.chaoticum.net/rpi/raspbian/pool/main/g/gcc-8/libgomp1_8.3.0-6+rpi1_armhf.deb  404  Not Found [IP: 176.9.51.222 80]
E: Failed to fetch http://mirror.chaoticum.net/rpi/raspbian/pool/main/f/fftw3/libfftw3-single3_3.3.8-2_armhf.deb  404  Not Found [IP: 176.9.51.222 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```
---------------------------------------------------------------------

Joulinar, 2021-04-07:
Hi, 

Many thanks for your message. Looks some files are missing on the Rasbian `apz`

ashishsantikari, 2021-04-14:
Sorry, I forgot to update. It worked after a few hours :)

closed 2021-04-14 by ashishsantikari