MichaIng/DietPi #4254: Error while updating DietPi

rg-sg, 2021-04-08:
#### Details:
- Date           | Thu Apr  8 14:13:21 BST 2021
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_79+0~20201210.30+debian9~1.gbpedaf15_all.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-zip_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-intl_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-json_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-opcache_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-fpm_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-readline_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-gd_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-xml_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-curl_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-sqlite3_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-mbstring_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-cli_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-common_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2021-04-08:
Hi,

Many thanks for your message. Pls can you post full output of `apt update && apt upgrade`

rg-sg, 2021-04-08:
`dietpi@DietPi:~$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  php-apcu
The following packages will be upgraded:
  apt apt-transport-https apt-utils bind9-host ca-certificates curl dnsutils git git-man
  libapt-inst2.0 libapt-pkg5.0 libbind9-140 libbsd0 libcurl3 libcurl3-gnutls libdns-export162
  libdns162 libfreetype6 libgssapi-krb5-2 libisc-export160 libisc160 libisccc140 libisccfg140
  libk5crypto3 libkrb5-3 libkrb5support0 libldap-2.4-2 libldap-common liblwres141 libp11-kit0
  libpython3.5-minimal libpython3.5-stdlib libsqlite3-0 libssl1.0.2 libssl1.1 login openssl passwd
  php-common php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-intl php7.3-json
  php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  python-apt-common python3-apt python3.5 python3.5-minimal screen sqlite3 sudo tzdata
60 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Need to get 3,846 kB/29.6 MB of archives.
After this operation, 50.2 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://packages.sury.org/php stretch/main armhf php-common all 2:79+0~20201210.30+debian9~1.gbpedaf15
  404  Not Found
Err:2 https://packages.sury.org/php stretch/main armhf php7.3-zip armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:3 https://packages.sury.org/php stretch/main armhf php7.3-intl armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:4 https://packages.sury.org/php stretch/main armhf php7.3-json armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:5 https://packages.sury.org/php stretch/main armhf php7.3-opcache armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:6 https://packages.sury.org/php stretch/main armhf php7.3-fpm armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:7 https://packages.sury.org/php stretch/main armhf php7.3-readline armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:8 https://packages.sury.org/php stretch/main armhf php7.3-gd armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:9 https://packages.sury.org/php stretch/main armhf php7.3-xml armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:10 https://packages.sury.org/php stretch/main armhf php7.3-curl armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:11 https://packages.sury.org/php stretch/main armhf php7.3-sqlite3 armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:12 https://packages.sury.org/php stretch/main armhf php7.3-mbstring armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:13 https://packages.sury.org/php stretch/main armhf php7.3-cli armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
Err:14 https://packages.sury.org/php stretch/main armhf php7.3-common armhf 7.3.25-1+0~20201130.73+debian9~1.gbp042074
  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php-defaults/php-common_79+0~20201210.30+debian9~1.gbpedaf15_all.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-zip_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-intl_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-json_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-opcache_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-fpm_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-readline_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-gd_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-xml_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-curl_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-sqlite3_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-mbstring_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-cli_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found
E: Failed to fetch https://packages.sury.org/php/pool/main/p/php7.3/php7.3-common_7.3.25-1+0~20201130.73+debian9~1.gbp042074_armhf.deb  404  Not Found`

Here is the output for apt-get upgrade

MichaIng, 2021-04-08:
The output of the first command (apt update) is probably more important. Known is the expiry of the sury.org repository key, which can be resolved with the following commands:
```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
```

Joulinar, 2021-04-08:
I guess cache would need to be cleaned as well. 

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

rg-sg, 2021-04-08:
Thank you, guys, it worked, the expiry of the sury.org repository key was the issue. 

Closing the issue.

closed 2021-04-08 by rg-sg

ravenclaw900, 2021-04-08:
@MichaIng,
Should we pin an issue about this with the fix?

MichaIng, 2021-04-08:
Good idea, just done with the first reported issue about it.