MichaIng/DietPi #4256: Failing to install mympd

sofad, 2021-04-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- SBC model | RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: Samsung Evo)

#### Additional Information (if applicable)
- Software title | mympd
- fresh install
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | 
 353491c0-9d58-4347-937a-fe58aacebfd8

#### Steps to reproduce

1. install mympd
2. type 'mympd' in terminal
 WARN     mympd     Unkown config option: webserver - webport
ERROR    mympd     Can't bind to http://0.0.0.0:80

#### Expected behaviour

- mympd should install

#### Actual behaviour

- install fails

#### Extra details

Joulinar, 2021-04-09:
Hi,

Many thanks for your report. Are you using any other web server?

sofad, 2021-04-09:
Hi Michalng, thank you for your support. 

I installed a fresh dietpi image, additional software: pihole, shairport-sync and mympd. i use mympd in my browser (ip-adress:1333). because i could not access mympd i was trying 'mympd' in the terminal. this is where i found the warning and error message:

root@DietPi:~# mympd
15:25:19 NOTICE   mympd     Starting myMPD 7.0.1
15:25:19 NOTICE   mympd     Libmympdclient 1.0.5 based on libmpdclient 2.20.0
15:25:19 NOTICE   mympd     Mongoose 7.3
15:25:19 NOTICE   mympd     Parsing config file: /etc/mympd.conf
15:25:19 WARN     mympd     Unkown config option: webserver - webport
15:25:19 NOTICE   mympd     Scripting disabled, disabling scripteditor
15:25:19 NOTICE   mympd     Setting loglevel to NOTICE
15:25:19 NOTICE   mympd     Localstate dir: "/var/lib/mympd"
2021-04-09 13:25:19  E mongoose.c:2902:mg_open_l Failed to listen on http://0.0.0.0:80, errno 98
15:25:19 ERROR    mympd     Can't bind to http://0.0.0.0:80
2021-04-09 13:25:19  I mongoose.c:2307:mg_mgr_fr All connections closed

MichaIng, 2021-04-09:
Joulinar is right that myMPD is started as system service. Starting it manually from the command line uses an unintended user and unintended options due to missing command arguments, like port 80 which is used by a regular webserver.

To start/stop/restart myMPD, do it via its systemd service, like:
```
systemctl restart mympd
```
If you cannot access, debug it via:
```
journalctl -u mympd
```

Joulinar, 2021-04-09:
as well you can check LISTEN ports to check if myMPD is running on correct port

`ss -tulpn | grep LISTEN`

sofad, 2021-04-09:
this is what i did:

> systemctl restart mympd

```
Apr 10 00:12:39 DietPi systemd[1]: Started myMPD server daemon.
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Starting myMPD 7.0.1
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Libmympdclient 1.0.5 based on libmpdclient 2.20.0
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Mongoose 7.3
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Parsing config file: /etc/mympd.conf
Apr 10 00:12:39 DietPi mympd[26331]: WARN     mympd     Unkown config option: webserver - webport
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Scripting disabled, disabling scripteditor
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Setting loglevel to NOTICE
Apr 10 00:12:39 DietPi mympd[26331]: NOTICE   mympd     Localstate dir: "/var/lib/mympd"
Apr 10 00:12:39 DietPi mympd[26331]: 2021-04-09 22:12:39  E mongoose.c:2902:mg_open_l Failed to listen on http://0.0.0.0:80, errno98
Apr 10 00:12:39 DietPi mympd[26331]: ERROR    mympd     Can't bind to http://0.0.0.0:80
Apr 10 00:12:39 DietPi mympd[26331]: 2021-04-09 22:12:39  I mongoose.c:2307:mg_mgr_fr All connections closed
Apr 10 00:12:39 DietPi systemd[1]: mympd.service: Main process exited, code=exited, status=1/FAILURE
Apr 10 00:12:39 DietPi systemd[1]: mympd.service: Failed with result 'exit-code'.
```

> ss -tulpn | grep LISTEN

--> mympd is not running / listening

Joulinar, 2021-04-09:
looks like config option for web server port incorrect. That's why `mympd` is trying to start with port 80, which of course is failing as port 80 is already used by the web server service PiHole. 

`Unkown config option: webserver - webport`

let me have a look

Joulinar, 2021-04-09:
ok found it, the web server port option changed. You would need to adjust `/etc/mympd.conf`

remove `webport = 1333` and adjust `httpport = 80` to `httpport = 1333`

Once done restart service `systemctl restart mympd.service`

MichaIng, 2021-04-10:
Many thanks guys! Fix has been merged. I'll also add a MOTD hack fix, as it's a trivial string replace here 😉.

MichaIng, 2021-04-10:
Also, we need to disable the option on Debian Stretch, since myMPD now requires MPD v0.20.0 at least, while Stretch ships v0.19.21:
- https://github.com/jcorporation/myMPD/releases/tag/v7.0.0
- https://packages.debian.org/stretch/mpd

... ah no wait, we ship own v0.20.23 packages for Stretch 👍: https://dietpi.com/downloads/binaries/all/

sofad, 2021-04-10:
Thank you very much!

closed 2021-05-05 by MichaIng