MichaIng/DietPi #4258: Dietpi Update Error 6.34.3 > 6.35

edo0, 2021-04-09:
#### Details:
- Date           | Fri Apr  9 17:31:22 CEST 2021
- Bug report     | c0124f77-7079-4022-be96-f1d55f89d830
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux Pi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. I realised I was late to the party, as version 7 of dietpi has been released already
2. I tried to upgrade from 6.34.3, and version 6.35 would show up as update
3. The update process fails. I have reverted to the backup I ceremoniously took just before attempting the upgrade
#### Expected behaviour:

- Update should succeed
#### Actual behaviour:

- An error pops up and the upgrade process halts (I think I may have seen something with expired PGP signatures as well in the process)
#### Extra details:

#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

ravenclaw900, 2021-04-09:
Hello,
Could you post the full output of `apt update` and `apt upgrade`?

MichaIng, 2021-04-09:
Many thanks for your report. Please check which package would be downgraded. I remember WireGuard caused this.

... ah, ravenclaw900 was faster, yes the second command would show you which package would be downgraded and will ask you to confirm. If it's WireGuard, you can accept, as it is not really a downgrade, only the version string changed when it was added to the Buster backports repository.

edo0, 2021-04-09:
Turns out the error was due to an outdated gpg for the `packages.sury.org` repository I use not being updated
<details> 
  <summary> root@Pi:~# apt update </summary>

```
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Get:5 https://packages.sury.org/php buster InRelease [6823 B]
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Err:5 https://packages.sury.org/php buster InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@
sury.org>
Reading package lists... Done
Building dependency tree
Reading state information... Done
19 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the
```
</details>

I therefore proceeded to updating the key 
and upgrading the packages   

<details> 
  <summary> root@Pi:~# apt upgrade </summary>
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  php5.6-apcu php5.6-imagick php5.6-redis php7.0-apcu php7.0-imagick php7.0-redis php7.1-apcu
  php7.1-imagick php7.1-redis php7.2-apcu php7.2-imagick php7.2-redis php7.3-apcu php7.3-imagick
  php7.3-redis php7.4-apcu php7.4-imagick php7.4-redis
Use 'apt autoremove' to remove them.
The following packages will be upgraded:
  base-files curl debian-archive-keyring iputils-ping libbsd0 libcurl4 libopenjp2-7 libpq5
  libpython3.7-minimal libpython3.7-stdlib libsystemd0 libudev1 python3.7 python3.7-minimal redis-server
  redis-tools systemd systemd-sysv udev
The following packages will be DOWNGRADED:
  libssl1.1 openssl
19 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 13.0 MB of archives.
After this operation, 138 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 base-files arm64 10.3+deb10u9 [69.9 kB]
Get:2 https://deb.debian.org/debian buster/main arm64 libsystemd0 arm64 241-7~deb10u7 [314 kB]
Get:3 https://deb.debian.org/debian buster/main arm64 systemd arm64 241-7~deb10u7 [3256 kB]
Get:4 https://deb.debian.org/debian buster/main arm64 udev arm64 241-7~deb10u7 [1245 kB]
Get:5 https://deb.debian.org/debian buster/main arm64 libudev1 arm64 241-7~deb10u7 [146 kB]
Get:6 https://deb.debian.org/debian buster/main arm64 systemd-sysv arm64 241-7~deb10u7 [100 kB]
Get:7 https://deb.debian.org/debian-security buster/updates/main arm64 libssl1.1 arm64 1.1.1d-0+deb10u6 [138
2 kB]
Get:8 https://deb.debian.org/debian buster/main arm64 python3.7 arm64 3.7.3-2+deb10u3 [330 kB]
Get:9 https://deb.debian.org/debian buster/main arm64 libpython3.7-stdlib arm64 3.7.3-2+deb10u3 [1642 kB]
Get:10 https://deb.debian.org/debian buster/main arm64 python3.7-minimal arm64 3.7.3-2+deb10u3 [1404 kB]
Get:11 https://deb.debian.org/debian buster/main arm64 libpython3.7-minimal arm64 3.7.3-2+deb10u3 [588 kB]
Get:12 https://deb.debian.org/debian buster/main arm64 redis-server arm64 5:5.0.3-4+deb10u3 [78.5 kB]
Get:13 https://deb.debian.org/debian buster/main arm64 redis-tools arm64 5:5.0.3-4+deb10u3 [499 kB]
Get:14 https://deb.debian.org/debian buster/main arm64 debian-archive-keyring all 2019.1+deb10u1 [93.5 kB]
Get:15 https://deb.debian.org/debian buster/main arm64 iputils-ping arm64 3:20180629-2+deb10u2 [43.0 kB]
Get:16 https://deb.debian.org/debian-security buster/updates/main arm64 curl arm64 7.64.0-4+deb10u2 [260 kB]
Get:17 https://deb.debian.org/debian-security buster/updates/main arm64 libcurl4 arm64 7.64.0-4+deb10u2 [312
 kB]
Get:18 https://deb.debian.org/debian buster/main arm64 libbsd0 arm64 0.9.1-2+deb10u1 [97.6 kB]
Get:19 https://deb.debian.org/debian-security buster/updates/main arm64 libopenjp2-7 arm64 2.3.0-2+deb10u2 [
148 kB]
Get:20 https://deb.debian.org/debian buster/main arm64 libpq5 arm64 11.11-0+deb10u1 [162 kB]
Get:21 https://deb.debian.org/debian-security buster/updates/main arm64 openssl arm64 1.1.1d-0+deb10u6 [823
[... 151 lines cut ...]
[ SUB3 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.3-fpm
[  OK  ] DietPi-Services | restart : nginx
[  OK  ] DietPi-Services | restart : coturn
[  OK  ] DietPi-Services | restart : cron
[FAILED] DietPi-Services | ncc maintenance:mode --off
[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

Terminated
root@Pi:~#
root@Pi:~#
root@Pi:~# reboot now
```

_However_, the update seems to have broken the Nextcloud instance I was hosting. My website reports

```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```

Joulinar, 2021-04-09:
Can you share status for all services

`dietpi-services status`

edo0, 2021-04-09:
Sure

```
root@Pi:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[  OK  ] DietPi-Services | redis-server		active (running) since Fri 2021-04-09 18:58:06 CEST; 45min ago
[  OK  ] DietPi-Services | mariadb		active (running) since Fri 2021-04-09 18:58:09 CEST; 45min ago
[  OK  ] DietPi-Services | php7.3-fpm		active (running) since Fri 2021-04-09 18:58:11 CEST; 45min ago
[  OK  ] DietPi-Services | nginx		active (running) since Fri 2021-04-09 18:58:11 CEST; 45min ago
[  OK  ] DietPi-Services | coturn		active (running) since Fri 2021-04-09 18:58:13 CEST; 45min ago
[  OK  ] DietPi-Services | cron			active (running) since Fri 2021-04-09 18:58:13 CEST; 45min ago
[  OK  ] DietPi-Services | dropbear		active (running) since Fri 2021-04-09 18:58:05 CEST; 45min ago
[  OK  ] DietPi-Services | fail2ban		active (running) since Fri 2021-04-09 18:57:46 CEST; 45min ago
[ INFO ] DietPi-Services | dietpi-ramlog	inactive (dead)
[  OK  ] DietPi-Services | dietpi-preboot	active (exited) since Fri 2021-04-09 18:57:46 CEST; 45min ago
[  OK  ] DietPi-Services | dietpi-boot		active (exited) since Fri 2021-04-09 18:58:05 CEST; 45min ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Fri 2021-04-09 18:58:05 CEST; 45min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	inactive (dead)
```

Joulinar, 2021-04-09:
Ok all services running, can you try

`ncc maintenance:mode --off`

edo0, 2021-04-09:
It gives me this error
```
root@Pi:~# ncc maintenance:mode --off
PHP Fatal error:  Allowed memory size of 536870912 bytes exhausted (tried to allocate 135168 bytes) in /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php on line 133
root@Pi:~# 

```

Joulinar, 2021-04-09:
Can you reboot your system

edo0, 2021-04-09:
Rebooted: the nextcloud "homepage" displays the same error message, and equally
```
root@Pi:~# ncc maintenance:mode --off
PHP Fatal error:  Allowed memory size of 536870912 bytes exhausted (tried to allocate 135168 bytes) in /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php on line 133
root@Pi:~# 
```

edo0, 2021-04-09:
Seems like it may be related to this issue maybe?
https://github.com/nextcloud/server/issues/25742
(I am running Nextcloud 21.0.1)

Joulinar, 2021-04-09:
Could be. This might be a workaround. https://github.com/nextcloud/server/issues/25742#issuecomment-783770628

Instead of PHP8, it should be PHP7.3 in your case

MichaIng, 2021-04-09:
Strange, we actually apply `apc.enable_cli=1` by default on Nextcloud install, so that should not happen: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L8151
Can you check:
```
cat /etc/php/7.3/mods-available/dietpi-nextcloud.ini
grep -r 'apc.enable_cli' /etc/php/7.3/fpm
```

edo0, 2021-04-09:
```
root@Pi:~# cat /etc/php/7.3/mods-available/dietpi-nextcloud.ini
; Nextcloud PHP settings
; priority=98
apc.enable_cli=1
opcache.enable=1
opcache.interned_strings_buffer=8
opcache.max_accelerated_files=10000
opcache.memory_consumption=128
opcache.save_comments=1
opcache.revalidate_freq=1
memory_limit=512M
root@Pi:~# 
```

```
root@Pi:~# grep -r 'apc.enable_cli' /etc/php/7.3/fpm
root@Pi:~# 
```
The second command does not give any output

edo0, 2021-04-09:
> Instead of PHP8, it should be PHP7.3 in your case

Nothing to do, that will not work unfortunately
```
root@Pi:~# cat /etc/php/7.3/cli/conf.d/20-apcu.ini
apc.enable_cli=1
```

MichaIng, 2021-04-09:
Ah not sure if grep -r parses symlinks:
```
phpenmod dietpi-nextcloud
ls -l /etc/php/7.3/fpm/conf.d/98-dietpi-nextcloud.ini
```

edo0, 2021-04-09:
```
root@Pi:~# phpenmod dietpi-nextcloud
root@Pi:~# ls -l /etc/php/7.3/fpm/conf.d/98-dietpi-nextcloud.ini
lrwxrwxrwx 1 root root 48 Oct 17 09:26 /etc/php/7.3/fpm/conf.d/98-dietpi-nextcloud.ini -> /etc/php/7.3/mods-available/dietpi-nextcloud.ini
root@Pi:~# 
```

MichaIng, 2021-04-09:
You broke the apcu.ini file. It's first two lines are essential to load the module and now missing. Let's see if we can restore or:
```
echo -e '; priority=20\nextension=apcu.so' > /etc/php/7.3/mods-available/apcu.ini
```

edo0, 2021-04-09:
Thank you for spotting it. I have inserted the commands you gave me, and this is the output if I try turning off maintenance mode
```
root@Pi:~# ncc maintenance:mode --off
PHP Warning:  PHP Startup: Unable to load dynamic library 'apcu.so' (tried: /usr/lib/php/20180731/apcu.so (/usr/lib/php/20180731/apcu.so: cannot open shared object file: No such file or directory), /usr/lib/php/20180731/apcu.so.so (/usr/lib/php/20180731/apcu.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
PHP Fatal error:  Allowed memory size of 536870912 bytes exhausted (tried to allocate 135168 bytes) in /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php on line 1
```

MichaIng, 2021-04-09:
It's the module not installed?
```
apt install php-apcu
```

edo0, 2021-04-09:
It figures as installed
```
root@Pi:~# apt install php-apcu
Reading package lists... Done
Building dependency tree       
Reading state information... Done
php-apcu is already the newest version (5.1.20+4.0.11-1+0~20210304.24+debian10~1.gbp1c3d4d).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@Pi:~# 

```

edo0, 2021-04-09:
I was missing indeed php7.3-acpu

edo0, 2021-04-09:
Thank you, that solved the issue!!
Apparently I had `php-apcu` (version 5), but not `php7.3-acpu`. I think it was removed during the upgrade process?

MichaIng, 2021-04-09:
Ahh you have the sury.org repository installed, so unexpected PHP versions are installed. Did you install this repository intentionally or was is a Stretch image that was dist-upgraded to Buster?

Try:
```
apt install php7.3-apcu php7.3-redis
```

edo0, 2021-04-09:
Thank you very much
I had added them to Stretch a couple of years ago, and when I moved to Buster (fresh install some months ago), I re-added them as a sort of automatism

closed 2021-04-09 by edo0

Joulinar, 2021-04-09:
That's not needed on Buster ☺️

edo0, 2021-04-09:
@Joulinar 
Thank you for the tip. Would it be enough to remove the repo?

Joulinar, 2021-04-09:
let's ask @MichaIng

MichaIng, 2021-04-10:
When you remove the repo, don't forget to remove the key as well to fully clean it up. Generally, there is nothing bad about the repo, it ships newer PHP versions (also newer minor versions for older major versions like PHP7.3), but indeed the meta package names that do not contain the version string, like `php-apcu`, always pull the newest officially released PHP version, currently PHP8.0, so it's important to install the specific versioned package names then. But to rule out that accidentally or as dependency of other packages the meta packages pulling PHP8.0 are installed, better remove it, so one thing less to worry about 😉.