MichaIng/DietPi #4260: Dietpi arm32v7 Docker containers will fail on certain Alpine 3.13 or Ubuntu Focal based images if libseccomp2 is not updated on the system

trinsic, 2021-04-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.2
- Distro version | buster
- Kernel version | Linux towncrier 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)
- Power supply used | OEM Pi brick
- SDcard used | Samsung 32gb?

#### Additional Information (if applicable)
libseccomp2 is a dependency of Docker. Docker made some changes in 19.03.9 but arm32v7 systems still require libseccomp2 2.4.2 or newer to fix it. Older versions of seccomp block certain syscalls made in containers and cause them to fail.
```
generating self-signed keys in /config/keys, you can replace these with your own keys if required
Generating a RSA private key
...............................+++++
....................................................................................+++++
writing new private key to '/config/keys/cert.key'
-----
1996473232:error:0D0D90AD:asn1 encoding routines:ASN1_TIME_adj:error getting time:crypto/asn1/a_time.c:330:
```
The latest version of seccomp2 on buster
```
root@towncrier:~# apt-get upgrade libseccomp2
Reading package lists... Done
Building dependency tree
Reading state information... Done
libseccomp2 is already the newest version (2.3.3-4).
```

Linuxserver.io recommends adding buster-backports
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 04EE7237B7D453EC 648ACFD622F3D138
echo "deb http://deb.debian.org/debian buster-backports main" | sudo tee -a /etc/apt/sources.list.d/buster-backports.list
apt update
apt install -t buster-backports libseccomp2
```
or updating the library with dpkg:
```
wget http://ftp.us.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.4.4-1~bpo10+1_armhf.deb
dpkg -i libseccomp2_2.4.4-1~bpo10+1_armhf.deb
```

Reported errors include:
> 502 errors in Jellyfin as seen in linuxserver/docker-jellyfin#71
> Error starting framework core messages in the docker log for Plex. linuxserver/docker-plex#247
> No WebUI for Radarr, even though the container is running. linuxserver/docker-radarr#118
> Images based on our Nginx base-image(Nextcloud, SWAG, Nginx, etc.) fails to generate a certificate, with a message similar to error getting time:crypto/asn1/a_time.c:330
> docker exec <container-name> date returns 1970

#### Extra details
Main discussion [#40734](https://github.com/moby/moby/issues/40734)
Linuxserver recommendations https://docs.linuxserver.io/faq#libseccomp

Joulinar, 2021-04-10:
Hi,

I did a test installation of Docker on my RPi3B+ 32bit without issues. Docker install and running fine. As well no issues to run the  Jellyfin container 

[image]

[image]

MichaIng, 2021-04-10:
On Debian we could install that library from backports, which are added to `sources.list` by default, but Raspbian has no backports, so no solution for 32-bit RPis. Adding the Debian backports to Raspbian, or dedicated package from the Debian repository opens a can of warms, e.g. when it gets dependencies from backports (currently not the case). And it doesn't work on ARMv6 models, which are likely affected the same way, so it requires a complicated logic, solving it only partly.

Interesting that it works in case of Joulinar 🤔. `libseccomp2` is pre-installed as hard dependency, even for `apt`, so probably it's more than just arm32 and that library to trigger the error?

Before adding such kind of workaround, I'm more keen to have a look into the actual issue in Docker or Alpine Linux and whether it can be fixed/worked around there. There should be a major interest since Debian Buster and Ubuntu Focal are the current stable/LTS versions, that very most people use.

For all affected containers you mentioned, DietPi offers native install options, which come lighter and better configurable 😉. I know it is not a valid option in all cases, but for completeness...

Joulinar, 2021-04-10:
Docker is an application used quite often on DietPi and I could not recall such issues.

MichaIng, 2021-09-01:
Since we cannot replicate this issue and have no other reports from our users either and are upgrading our images to Bullseye, I mark this as closed. Feel free to reopen if required.

closed 2021-09-01 by MichaIng