MichaIng/DietPi #4261: Failed to try-restart sshswitch.service: Unit sshswitch.service is masked.

Gill-Bates, 2021-04-10:
Hey there,

during `apt dist-upgrade` I encounter following error message:

```
Unpacking raspberrypi-sys-mods (20210310) over (20210208) ...
Setting up systemd-sysv (241-7~deb10u7+rpi1) ...
Setting up libssl1.1:armhf (1.1.1d-0+deb10u6+rpt1) ...
Setting up libcurl3-gnutls:armhf (7.64.0-4+deb10u2) ...
Setting up raspberrypi-sys-mods (20210310) ...
Blocking vscode repo...
Failed to try-restart sshswitch.service: Unit sshswitch.service is masked.
Failed to try-restart sshswitch.service: Unit sshswitch.service is masked.
Failed to try-restart sshswitch.service: Unit sshswitch.service is masked.
Failed to try-restart sshswitch.service: Unit sshswitch.service is masked.
Setting up libpython3.7-minimal:armhf (3.7.3-2+deb10u3) ...
```

Bug or Feature?

Joulinar, 2021-04-10:
Hi,

Many thanks for your message. Best to you knowledge, this can be ignored safely.

Gill-Bates, 2021-04-10:
Thanks Joulinar. Is there a chance, to change the console output to a less confusing message like (can be ignored).

Joulinar, 2021-04-10:
This output is provided by `apt` package installer and not really influenced by DietPi script.

MichaIng, 2021-04-10:
It's part of `raspberrypi-sys-mods` indeed, which contains some useful tweaks and some that are not relevant for DietPi. That service is to have OpenSSH enabled at boot, when `/boot/ssh` exists. Creating that file is the way to have SSH enabled on first boot of the Raspberry Pi OS image, where it is disabled by default. On DietPi, SSH is enabled by default and the Dropbear SSH server is installed instead of the OpenSSH server. That service has hence no use on DietPi, would fail and is hence masked on our images.

Good that the messages on package upgrade says already "Message can be ignored", compared to the possible case where the service is enabled and would fail at boot. So keeping it masked is still the least confusing solution.

Joulinar, 2021-04-14:
Guys, I guess we can close this topic as we are not able to influence. Do you agree?

closed 2021-04-15 by MichaIng