MichaIng/DietPi #4263: DietPi_RPi-ARMv6-Buster_Amiberry not booting in Raspberry Pi 3B rev 1.2

sromeroi, 2021-04-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | DietPi_RPi-ARMv6-Buster_Amiberry
- Distro version | DietPi_RPi-ARMv6-Buster_Amiberry
- Kernel version | DietPi_RPi-ARMv6-Buster_Amiberry
- SBC model | Raspberry Pi 3B rev 1.2 - (a02082)
- Power supply used | official Raspberry Pi Foundation power supply
- SDcard used | Several tried: Sandisk, Medion, Kingston...

#### Additional Information (if applicable)
- Software title | Amiberry version
- Freshly installed. Not booting (boot screen does not show up). Moved the same SD's to a RPi 3B+ and Rpi 4 and they boot. After the first boot, I test the working SD cards in Pi 3B and do not work. Other distributions boot perfectly in the same Pi 3B (RaspbianOS, Dosbian, Pimiga, Retropie, Retropie-crt, Amibian...).
- Can this issue be replicated on a fresh installation of DietPi? Yes.

#### Steps to reproduce
1. Download DietPi_RPi-ARMv6-Buster_Amiberry and flash it to an SD card or USB drive.
2. Try to boot. 

#### Expected behaviour
The image boots.

#### Actual behaviour
The Pi powers on. The red led switches on. The green led flashes for a second and then nothing else happens. The TV is switched on but nothing appears in the screen. No boot sequence, no green led blink to identify errors. It happens in TWO DIFFERENT Pi 3B boards.

#### Extra details
The same SD cards work if I plug them on a Pi 3B+ or in my 2 Pi 4 (2 and 4Meg). But none of my 2 Pi 3B rev 1.2 can boot the image.

Joulinar, 2021-04-12:
Hi,

can you try using the normal DietPi image on your Pi 3B rev 1.2. Would be interesting to see how it behave.

sromeroi, 2021-04-12:
I tried a 4th SD card (this one a chinese SD card) and both  the normal DietPi image and the Amiberry image booted.

Maybe is something wrong in the bootloader of Pi 3B rev 1.2 related to SD cards. A friend of mine reported the same issue with his 3B's board too, but works in his 3B+.

Probably it's not DietPi's fault then.

closed 2021-04-12 by sromeroi