MichaIng/DietPi #4264: Incorrect rpi-eeprom branch set

Trunkzeh, 2021-04-12:
Noticed that the firmware branch rpi-eeprom is set to is default; this should be set to stable. I was wondering why my RPI4 hasn't updated past 03/09/2020 and now I know why. Changes as follows:

```
root@DietPi:~# vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
```

```
root@DietPi:~# nano /etc/default/rpi-eeprom-update
FIRMWARE_RELEASE_STATUS="stable"
```

```
root@DietPi:~# sudo rpi-eeprom-update -a
*** INSTALLING EEPROM UPDATES ***

BOOTLOADER: update available
   CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
    LATEST: Thu Mar 18 08:54:11 UTC 2021 (1616057651)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up-to-date
   CURRENT: 000138a1
    LATEST: 000138a1
BOOTFS /boot

EEPROM updates pending. Please reboot to apply the update.
To cancel a pending update run "sudo rpi-eeprom-update -r".
```
:)

Joulinar, 2021-04-12:
well that's the original setting from the `rpi-eeprom` package as it is provided by Raspberry Foundation

https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
- `default` will update releases marked as `critical`

Joulinar, 2021-04-30:
@MichaIng 
I guess we can close this one as we follow RPi standard

MichaIng, 2021-05-01:
> and periodic update for new features that have been tested

So also the default = critical branch receives latest feature updates, but later than the "latest" branch on a periodic level. I think that is fine and we should follow the defaults of the RPi foundation.

closed 2021-05-01 by MichaIng